Louisvdw/dbus-serialbattery #3: Daly BMS integration

Louisvdw, 2021-02-04:
Add the Daly BMS basic communication to the driver.

[Part 4_ Daly RS485+UART Protocol.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/5926615/Part.4_.Daly.RS485%2BUART.Protocol.pdf)

neoneddy, 2021-02-26:
Up Vote

gyrex, 2021-03-04:
Great work on this project mate! I'd love to see this added please

svaeyens, 2021-03-09:
Up Vote

pau1phi11ips, 2021-03-11:
This would be awesome!

MichaelStadler91, 2021-03-13:
Would be very nice! How is the current status?

Louisvdw, 2021-03-15:
> Would be very nice! How is the current status?

I already have code that can read the SOC for the Daly. Now I need to move the code a bit so that it can easily be shared between the current battery driver and the Daly code (issue #11) and then I can give you Daly users something to test.

pau1phi11ips, 2021-04-14:
Do you have a rough estimate to when the Daly code will be available @Louisvdw?

Louisvdw, 2021-04-30:
I hope to have a first version for the the Daly's SOC at the end of May.

neoneddy, 2021-04-30:
Awesome, do you have a paypal or something to donate to?

dreadnought, 2021-05-01:
@Louisvdw you can have a look at it if you like: https://github.com/dreadnought/python-daly-bms

Louisvdw, 2021-05-03:
> Awesome, do you have a paypal or something to donate to?

Yes you can use Kofi. https://ko-fi.com/louisvdw

yannickudb, 2021-06-15:
Up vote!

Louisvdw, 2021-06-20:
There is a new [beta build v0.5beta2](https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.5beta2) for the initial Daly BMS integration for those that want to test it and give feedback.
This build should work for both the RS485 and UART connection, update the battery SOC, voltage and current in the GX device.

Download the venus-data.tar.gz from beta2 and FTP it to your GX/pi. 
Extract using `tar -zxf venus-data.tar.gz -C /data`
Reboot and let me know if it works. 
Those mini 4sell Daly BMS seem to be running a different protocol, so I don't expect them to work.

pau1phi11ips, 2021-06-21:
Awesome, I'll give it a go tonight. Excited! 😀

pau1phi11ips, 2021-06-21:
@Louisvdw I've installed Beta 2 on my Cerbo GX. It's reading the values but only for a split second.

I made a screen recording: https://photos.app.goo.gl/n8SuuhjrU6c5bxQAA

Louisvdw, 2021-06-22:
@pau1phi11ips do you have SSH access to the GX to copy the log file and attach it here? /data/log/dbus-serialbattery.ttyUSB0/current

pau1phi11ips, 2021-06-22:
Attached

pau1phi11ips, 2021-06-22:
Struggling to attach the log directly but could get a screenshot.
[image]

I think the error it due to my BMS only having 1 temperature sensor.

Louisvdw, 2021-06-22:
I think you are corect. That temp sensor seems to be the issue.

pau1phi11ips, 2021-06-22:
I'll try and get it working here and post the results. I don't really know Python (just PHP and JavaScript) but think I can handle removing the temp2 references to see if that sorts it. At least that will nail there culprit down.

Louisvdw, 2021-06-22:
@pau1phi11ips can you check if this fix your temp sensor issue
[venus-data.tar.gz](https://github.com/Louisvdw/dbus-serialbattery/files/6691664/venus-data.tar.gz)

pau1phi11ips, 2021-06-22:
@Louisvdw Yep, working now thanks! 😎

pau1phi11ips, 2021-06-23:
One word of caution. I have a 24V LiFePO4 battery and usually charge to 3.35V/cell = 26.8V.

The charge limit on the Multiplus II is set to 26.8V. The battery voltage was 26.7V as soon as I enabled DVCC the Mutliplus started putting 50A into the battery. It was at 27.1V in no time with no sign of stopping.

It looks like because the driver isn't getting Vmax from the BMS it might be using the default 55V?

[](url)
[image]

Enabling the Limited charge voltage in the DVCC settings sorted it out but this isn't the default setting so beware!

Louisvdw, 2021-06-23:
Thanks for highlighting @pau1phi11ips 
All the current features of the driver will still apply. What you are referring too is described under  point 7, but perhaps I will expand that to mention the voltages. The driver works on an upper limit of 3.45V per cell. 
If you want to lower either the charge current of voltage you can set those under DVCC.

pau1phi11ips, 2021-06-23:
OK, thanks, I was worried it was using the default voltage the **Maximum Charge voltage** is set to as soon as you enable **Limit managed battery charge voltage**.

Louisvdw, 2021-06-23:
The battery publish the discharge current, charge current and charge voltage (for the driver that is 60A,50A, 3.45V x cell_count)
You can limit the discharge current under ESS. The lowest value between that limit and the driver will be used.
You can limit the charge current and/or voltage in DVCC. The lowest value between that and the driver will be used.

This is there for you to override (lower) the defaults. It is normally used more when you set up your battery and want to give it time to balance and settle, but it works well for our needs in this case as well.

pau1phi11ips, 2021-06-23:
Thanks for the info. Much appreciated.

BTW, I've edited my daly.py to change the polarity of the current value so it's the same direction as the value from the Victron BMV 712. It's definitely wired the right way around around ;)

Changed
`self.current = (current- 30000) / 10`
to 
`self.current = -1 * (current- 30000) / 10`

This is how it looks now
[image]
[image]

Louisvdw, 2021-06-23:
Inverting should show a negative draw on the battery. I'll recheck the Daly protocol, but just from looking at you screenshot with the Miltiplus II your change does look correct.

yannickudb, 2021-06-27:
Got my small daly 4S BMS today, didn't get it working. Anyone who want to help?

Louisvdw, 2021-06-28:
@yannickudb can you upload the log file for us. See the trouble shoot section in the readme on how to do that.

yannickudb, 2021-06-28:
I Will do tonight. But i've been messing around with the bms, and found the 4s needs another pc program. I suspect i uses other communication. Any way i can help with that?

Louisvdw, 2021-06-28:
I think you are correct.
We need the communications protocol for that. If you can find the protocol documentation, open a new ticket and attach that doc.

Louisvdw, 2021-06-28:
Implemented in release V0.5

closed 2021-06-28 by Louisvdw

yannickudb, 2021-06-28:
> I think you are correct.
> We need the communications protocol for that. If you can find the protocol documentation, open a new ticket and attach that doc.

I did, have attached a PDF!

janekx, 2022-06-21:
[0612b045a09d492f5593ebf9b24742b3.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/8946050/0612b045a09d492f5593ebf9b24742b3.pdf)
Daly 1.2v protocol