Louisvdw/dbus-serialbattery #4: Read BMS charge/discharge limits

Louisvdw, 2021-02-04:
Read charge/discharge limits from the BMS settings instead of using a hard coded value.
This command is not in the LTT protocol document, but the app does set these so it should be possible to read.

neoneddy, 2021-12-09:
Does this also mean the ability to change these?

Louisvdw, 2021-12-10:
This is to read the value that is set in the BMS. You can change them in the BMS using the bluetooth software, and then the updated values will pull through. Is that what you mean by changing them?

neoneddy, 2021-12-10:
I was hoping we could somehow set the values via VenusOS.  I'm noticing the voltage limit on mine is now set at 13.8v or so, I was running 14v before.

Are you aware if it's possible to run both BT and  UART?  or split the serial signal and use both BT and UART adapters?

Louisvdw, 2021-12-10:
That 13.8V is coming from the driver and not the BMS. That is what this ticket is about, but reading it from the BMS and not the settings in the code. https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install#how-to-increase-the-default-limits
If you want to change that, you can edit the cell min/max voltage in the utils.py file.

You can't run both BT and UART->USB at the same time. UART works on a single master <--> slave comms concept. If your BMS have RS485, that can have multiple slaves, but then you have more ports that just the UART already. 
One user did wire in a small transfer switch, so he can switch between BT and the USB-UART connection with the switch. You need a 2 pole Y switch and switch the RX/TX wires.

neoneddy, 2021-12-10:
Ok, thanks.

coca7, 2022-01-17:
I have three 230a lithium batteries with three 150a JBD BMSs. I have edited the min/max voltages to 2.5 and 3.65 in utils.py, however, I have not been able to get the batteries to charge up to 14.6.  The BMS is set for 3.65.  The battery does not seem to want to charge over 13.5volts or 3.375 volts.  The smarthsunt shows the battery as only 75% full.  There are no current error in the logs.

Louisvdw, 2022-01-17:
@coca7 check in your Remote Console -> SerialBattery -> Parameters -> CVL. This should be 3.65V x amount of cells (14.6V in your case). If it is (which I expect it is) then the driver is doing what it should. 

Most likely what is happening is that your cells in your battery is not perfectly balanced. The JBD will raise an alert and stop the charging when a cell reach 3.65V on it's default settings and there is always one cell reaching that sooner than the others (even on well balanced cells). That is why for normal opperation I suggest you use 3.45V per cell and not 3.65V.
100% SOC is not on the cell voltage but on the capacity. You don't want the cells to reach 3.65V.
If your battery charge to 13.5V, look at the Min/Max cell voltage graph in VRM. I expect you will see one cell reaching 3.65V while the rest are still at 3.3V. The min and max should be close to each other on a well balanced battery.
[image]

What did you set your capacity for the smartshunt? If this is a new battery then you will have to give it a few cycles to balance everything and get the SOC up correct.

coca7, 2022-01-17:
@Louisvdw The battery is new, but I’ve had it for a few weeks.  All the cells were within .5-1v of each other when I started.  I set the capacity on the smartshunt to 690 AH, which is sum of my 3 230AH batteries in parallel.  I think you have a point about the individual cells getting full.  I did notice that the cells do not all discharge at an equal rate.  

[image]

Hakuna-23, 2022-04-24:
Hello
Thank you for your work.
Can you please help me?
I instal the driver on my CreboGx like in the wiki and for my first time with puty it worked well.
But now i whant to chance the CVL and DCL limits and i don't understand how i can do this. 
I Download the driver to my pc and change the data like in the wiki. But how can i now instal this driver on my CreboGx. 
Thank you

Marvo2011, 2022-10-28:
@smagicld can you help here? We need the registers for "max charge current" and "max discharge current". Many thanks.

smagicld, 2022-10-28:
Hi
The JBD FW supports something like max discharge and charge current.
but JBD is very cheap made (at least the BMS board's I've tested) - it will not limit the current to max value - it will disable charging / discharging port for certain time (release time set in seconds) whenever the current is over the set limit.

you cen set it here:
[image]

Marvo2011, 2022-10-28:
@smagicld Many thanks for your response, we would to read this fields over TTL/RS232/UART. Can you help us with the registers we have to read?

smagicld, 2022-10-28:
```            
BMSconfigParam(
                name: "Charge overcurrent",
                unitType: "10mA",
                valueType: Int(0),
                commandCode: BMScommandCode.CHGOC_val,
                checkFunctions: [FilterFunction(function: checkIfValueGreaterThan, parameter: 0)],
                lastValueReadTimestamp: nil, valueHasBeenModified: false, value: nil, newValue: nil,
                exampleRealReadResponse: [0xDD, 0x28, 0x00, 0x02, 0x1B, 0x58, 0xFF, 0x8B, 0x77], // (7000 + 0)
                supportedFromVersion: 0
            ),

         BMSconfigParam(
                name: "Discharge overcurrent",
                unitType: "10mA",
                valueType: Int(0),
                commandCode: BMScommandCode.DSGOC_val,
                checkFunctions: [FilterFunction(function: checkIfValueGreaterThan, parameter: 0)],
                lastValueReadTimestamp: nil, valueHasBeenModified: false, value: nil, newValue: nil,
                exampleRealReadResponse: [0xDD, 0x29, 0x00, 0x02, 0xE4, 0xA8, 0xFE, 0x72, 0x77], // (negative value. displayed 7000. written: 58536. callculation: 65536 - 7000 = 58536)
                supportedFromVersion: 0
            ),

    case CHGOC_val = 0x28
    case DSGOC_val = 0x29
```

Marvo2011, 2022-11-02:
@Louisvdw Does it help you ?

paulsteigel, 2023-10-26:
Hi All,
Can you have any advise on making JK-BMS to request the Inverter to apply a linear charging scheme like in the image below?
[image]
Mine with PYLON CAN bus protocol will always be imposed with a stressful charging scheme like in the image below:
[image]
Please advise with thanks!
Best regards,
Ngoc

mr-manuel, 2023-10-26:
@paulsteigel I don't see any correlation of your request with this topic. Please open a discussion. Additionally it's not fully understandable what you mean.

paulsteigel, 2023-10-26:
> @paulsteigel I don't see any correlation of your request with this topic. Please open a discussion. Additionally it's not fully understandable what you mean.

Many thanks Mr. Manuel,
Please be sorry for the wrong posting, please delete or remove with thanks. I tried to find a way to ask the inverter (Dye, LuxPower) to  apply a linear charging to the battery rather than applying a fixed charging scheme with constant power.
If you can, please advice me on this with thanks.
Best regards
Ngoc

Louisvdw, 2023-10-26:
@paulsteigel that is implemented by the Pylontech battery. You cannot change it on your side, they will have to code that as part of the Pylontech CANbus implementation.

paulsteigel, 2023-10-26:
> @paulsteigel that is implemented by the Pylontech battery. You cannot change it on your side, they will have to code that as part of the Pylontech CANbus implementation.

If you can, please advice me on the routine to inform the Inverter (via PYLON Can bus protocol) to apply a proper charge scheme to the battery.
For example (I assume the routine as below):
Microcontroller will monitor the SOC when is reach 80%, it will start informing the inverter about new Current limit (x A) instead of the limit set by the BMS (y A)

Louisvdw, 2023-10-26:
Please ask your questions in the [discussion section](https://github.com/Louisvdw/dbus-serialbattery/discussions)

mr-manuel, 2024-02-16:
This was implemented for most BMS where the intel was given.

closed 2024-02-16 by mr-manuel