Louisvdw/dbus-serialbattery #5: ANT BMS integration

Louisvdw, 2021-02-04:
Add the ANT-BMS communication to the driver.

- [ ] SOC
- [ ] Battery voltage
- [ ] Charge/discharge current
- [ ] Temprature
- [ ] Min/max cell voltages
- [ ] protection notifications

osantiano, 2021-04-12:
Would be great !

JBecker32, 2021-05-23:
Would like to see that too :-))

pahorpet, 2021-07-15:
Hi, any news here?

Louisvdw, 2021-07-16:
I have not started to look at the ANT BMS. I don't have an example of the BMS to do testing with and all other users that has one does not seem to be able to connect the BMS to their controllers or PC. If we can solve that it should be fairly simple as the ANT protocol is about 90% to what we use for the LLT/JBD BMS already.

I'm looking for anyone that can help as a sponser for hardware I can do test with.
Or if anyone has their ANT connected to a Pi or GX and is wiling to help with testing and technical debugging.

JBecker32, 2021-07-16:
> Hi, any news here?

does your ANT BMS have a free serial interface? I have two of them but the serial is occupied by the internal BT module. The only external interface is that for the display.

pahorpet, 2021-07-16:
> > Hi, any news here?
> 
> does your ANT BMS have a free serial interface? I have two of them but the serial is occupied by the internal BT module. The only external interface is that for the display.

300A version what I have have it the same, I have just ordered 450A version, so 7 days delivery, mean next week I should be able to check it.

JBecker32, 2021-07-16:
I will check the function of the display interface when I'm home after holidays in two weeks.

liufuyang, 2021-07-27:
Hi there. Looking forward to this. I am very new to the solar/battery business and I am waiting the ANT BMS and raspberrypi to arrive, perhaps later I can help give some tests. I currently have a victron mppt 100V/30A and it works well to a small bluetti power station and I am waiting to get the ANT BMS to build my own LTO battery pack, and I hope later it can be all wired onto the victron VRM page via the raspberrypi. 

Also, I saw some of the BMS (like [this MGod one](https://aliexpress.com/item/1005002085277123.html)) saying they have the so called `UART/RS485` connection port. Does that means as long as a BMS having this type of connections then we can use this driver to connect it to VenusOS?

Thank you and sorry for asking so many questions :)

Louisvdw, 2021-07-27:
@liufuyang if the BMS has a UART, RS485 or other serial connection port that only mean you can connect it to another device via that serial port for communication. But unfortunately each manufacturer implement their own protocol. 
So to make a new BMS work with this driver we first need to write the code to implement their protocol and commands.

liufuyang, 2021-07-27:
I see. Thank you for the info. I asked the seller of the MGod BMS and they say Daly is their cooperating entity so I guess it will probably work. While I am waiting for the Ant to arrive I will just order a MGod as well as I feel they having a better service than Daly on aliexpress.  Will probably get it after a month and if I ever test it I will let you know how it goes :)

pahorpet, 2021-07-28:
> I have not started to look at the ANT BMS. I don't have an example of the BMS to do testing with and all other users that has one does not seem to be able to connect the BMS to their controllers or [...]

Finally I am ready to test. ANT BMS connected via serial display interface to USB I have rpi to on.

osantiano, 2021-07-29:
This guy did some work on ANT: https://github.com/juamiso/ANT_BMS (via Bluetooth)

pahorpet, 2021-08-10:
An news here? I am ready to do some testing

Louisvdw, 2021-08-12:
> An news here? I am ready to do some testing

Hi @pahorpet. I need to finish a few pieces of work. I'll get a script to you to test as soon as I can.

JBecker32, 2021-08-12:
Hi Louis, you can send it to me too for testing. I still have the ANT lying around ....

Louisvdw, 2021-08-17:
@pahorpet @JBecker32 
I created [a branch ](https://github.com/Louisvdw/dbus-serialbattery/tree/ant-bms) with the first changes to test the ANT. 
Give it a test and see if you get a response. It should read the SOC, Voltage and Current.

pahorpet, 2021-08-17:
Great tomorrow I will test it.

liufuyang, 2021-08-18:
My ANT bms should soon arrive today as well. I would like to help give it a test as well but I haven't installed our driver/plug-in before, so all I need is to follow the installation doc and see how to test that branch @Louisvdw mentioned above? And what do you mean by it should read "SOC, V, A", where do I expected to see the readings? (From some logs via ssh into Venus OS or they will already show up on Victron VRM pages?)

I am not sure yet how to test a branch but I guess one could ssh into Venus OS (just tried this earlier and it works) and would it be possible to install git there and pull this code directly to some directory to install the pulg in(like restarting the system to trigger it)?

Louisvdw, 2021-08-18:
The best would be to follow the install steps to install an released version first. This will set everything up and set the file permissions. 
Then you can copy the files from the branch over the existing install.  You only need to update the *.py files in the dbus-serialbattery/etc/dbus-serialbattery/ directory. All the other files and folder does not change.

If it all works, then the battery will show up in the Remote Console of your GX device and SOC, V and A will be displayed. If not then look at the log files. See the troubleshoot section in the wiki.

liufuyang, 2021-08-18:
Thank you. Sounds like something I can try. Do you or @JBecker32 @pahorpet perhaps knows how to wire the ANT BMS's 4 pin to the RS485->USB cable?   I guess that is the suggested way to connect? The RS485->USB chip I have around is [this one](https://amazon.se/gp/product/B07TFSZ3ZP) - which I used now to wire a Victron solar charger to Raspberry Pi and it works well. Do you know which pin is Rx/Tx/Gnd and so on? I am a bit unsure how to wire the BMS to the Raspberry Pi yet :)

JBecker32, 2021-08-18:
My ANT BMS has no RS485! The display interface you mention is certainly **not** RS485, but UART with 5V or even 3.3V level. An RS485<->USB adapter will not work!
This is the problem with the ANT, that it has no dedicated (wired) interface for such things like we want to do ... (and atm this is what also prevents me from trying the new code :-(()

pahorpet, 2021-08-18:
So I give it a try today but without succes I was trying to install from USB, so put in the root folder and place in cerbo gx. And after connect the usb from BMS I have got but disconnect 5V. So I will give it another try next week again.

liufuyang, 2021-08-18:
I see. Sorry I mixed with RS485 with UART. The device I have at hands are these https://www.amazon.se/gp/product/B07TFSZ3ZP, so perhaps they are UART to USB?  I used them on Victron Solar charger and it can connect. So perhaps Victron Solar charger has a UART output?

liufuyang, 2021-08-18:
I still have some question for the installation process. 

1. Are we sure now that we will have to use an external USB stick to do this? 
2. (As I can only find one USB stick around me) Does it require the USB stick to be formatted in some special format? I think the only USB storage I have is used when I had Windows machines before (NTFS), as when I just mount it on my Mac now I seem cannot write but only able to read from it. Perhaps I need to reformat it into something else or? 

Thank you :)

JBecker32, 2021-08-18:
I tried to understand the display interface of the ANT today, spying on the receive and transmit lines, but without success. This is certainly not the using the documented ANT serial format. So for now I'm out as I not have an ANT BMS with UART or RS485 interface ...

Louisvdw, 2021-08-19:
That seem to be the problem with the ANT. To get a USB->serial connection to the BMS.

langepil, 2021-08-19:
I'm really not a professional, but is the USB Adapter for PC possibly the solution?

[I'm really not a professional, but is the possibly the solution?](url)

I have such one at home, but unfortunately currently less time to test.

Louisvdw, 2021-08-19:
That USB adapter for PC will work, yes. But it seems that most ANT BMS does not come with the required connection to use that adapter.

langepil, 2021-08-19:
The adapter is connected to the same port where the external display is connected. They should all have this port, right?

Louisvdw, 2021-08-19:
@langepil it might be that the adapter has some conversion curcuit inside. Does it have a box as part of the cable and can you open it and take a few pictures?

langepil, 2021-08-19:
Unfortunately, I don't know at the moment. I'll take a look this afternoon and report back.

liufuyang, 2021-08-19:
After asking the seller several times to let him/her contact their tech people, I finally got this wiring connecting graph from them:
[image]

Perhaps my serial->USB connecting device can work there, will have to test it later.

langepil, 2021-08-19:
Here are the pictures of the original Adapter

langepil, 2021-08-19:
Here are the pictures of the original Adapter

[image]
[image]
[image]
[image]

Louisvdw, 2021-08-19:
That looks like a CH340G chip (the big one) which is a USB-serial chip. This can be RS232 or RS485 depending how it is set up. Most likely RS485 in theis instance.

JBecker32, 2021-08-19:
> The adapter is connected to the same port where the external display is connected. They should all have this port, right?

Maybe you didn't read my comments above! The display port uses a completely different protocol! As long as nobody can provide the details, this is not usable at all!

pahorpet, 2021-08-19:
http://mysharegadget.com/265290407
here is the pc sw and some docu  i will give another try as soon possible

S pozdravem a přáním hezkého dne
Petr Pahorecký
www.g-kolo.cz

liufuyang, 2021-08-19:
BTW you might need this change in order to test the ANT BMS https://github.com/Louisvdw/dbus-serialbattery/pull/35

pahorpet, 2021-08-20:
I will, for connect battery to the pc the baudrate is 19200.

JBecker32, 2021-08-20:
Ok now when I try to connect to the ANT display port at 19200 Baud with a USB-adapter I have the following situation:
When I send 0x5A 0x5A 0x00 0x00 0x01 0x01, I receive the following (the answer is 140 bytes long!):
0xAA 0x55 0xAA 0xFF 0x01 0x16 0x0D 0x97 0x0D 0x97 0x0D 0x97 0x0D 0x98 0x0D 0x97 0x0D 0x97 0x0D 0x98 0x0D 0x97 ....
where the 0x0d 0x97 are the 8 cell voltages  ....
But this is a completely different protocol than what is documented for ANT serial communication!?!

JBecker32, 2021-08-20:
OK, same result when I send 0xDB 0xDB 0x00 0x00 0x00 0x00 (or 0xDB 0xDB 0x00 0x00 0x01 0x01 or 0xDB 0xDB 0x00 0x00 0x02 0x02  ... )
The length of 140 bytes is ok, but the position of e.g. voltage seems wrong!?!

pahorpet, 2021-08-20:
What documentation are you looking at?
S pozdravem a přáním hezkého dne
Petr Pahorecký
www.g-kolo.cz

JBecker32, 2021-08-20:
???
is there a documentation of the display interface?
I don't have it!

JBecker32, 2021-08-20:
could we please clarify something? Are we all connecting with the ANT BMS via the display interface? Or is anyone using a dedicated UART interface on an ANT? Or RS485?

liufuyang, 2021-08-20:
My board is not arrived yet but I am planing to connect it via the display interface. They are also selling a USB connection device with that interface - which I didn't order previously.

pahorpet, 2021-08-21:
Yes display interface, I don't have it as well, but where did you get:
"But this is a completely different protocol than what is documented for ANT serial communication!?!"

I had asked the seller if they can provide a protocol.

JBecker32, 2021-08-21:
I have now managed to get the communication to work with some mods to the code. Not a final solution, but showing the basic values.
I think that the display interface is not giving all information, but it is ok.
I see the voltage, current, SOC, cell voltages, min and max cells, remaining capacity, state of the MOSFETs, temperature ...

pahorpet, 2021-08-21:
Great,
Looking forward to test it.
It has to give all information as I am using with pc to set it. But not
need it.

JBecker32, 2021-08-21:
Not sure how to use github correctly  to make a pull request ...
Therefore I will try to attach the changed files here:
[ANT_BMS.zip](https://github.com/Louisvdw/dbus-serialbattery/files/7026297/ANT_BMS.zip)

pahorpet, 2021-08-22:
So I get to the testing instal new version in cerbo GX but I am getting :
@4000000061226c7721d6ecfc INFO:__main__:Testing Ant
@4000000061226c7732c5006c ERROR:utils:>>> ERROR: No reply - returning
@4000000061226c773346174c INFO:utils:>>> ERROR: Incorrect Data
@4000000061226c7815f9d8cc ERROR:__main__:ERROR >>> No battery connection at /dev/ttyUSB0

pahorpet, 2021-08-22:
So I had made it to run with cerbo gx:
First need to modify util a bit 
```
while toread < (length_pos+1):
  **sleep(0.01)**
  toread = ser.inWaiting()
  count += 1
   if count > **150**:
    logger.error(">>> ERROR: No reply - returning")
   return False 
...
```
And the discharge current is divided by 4.
only total capacity is missing. do you know how to get it from?
And in details it shows number of modules blocking charging 1 what does it mean?

[image]
[image]
[image]

JBecker32, 2021-08-22:
as far as I can say, total capacity is not included in the answer of the ANT. 
Nr. of modules blocking charge???
no idea how this is generated ...

pahorpet, 2021-08-23:
Is this correct? 
[ANT_communication_protocol_EN (1).pdf](https://github.com/Louisvdw/dbus-serialbattery/files/7028450/ANT_communication_protocol_EN.1.pdf)

JBecker32, 2021-08-23:
Looks quite OK. Although in my case (display interface) the baud rate certainly is 19200 and not 9600.
And the physical capacity is empty (00 00 00 00) although correctly set and shown via BT (Android App).

pahorpet, 2021-08-23:
In my case is as well 19200 as default.
The change need to be made is mark.
It show -xA when charging instead of +.
And then alarm / notification can be done as well.

JBecker32, 2021-08-23:
@pahorpet 
'And in details it shows number of modules blocking charging 1 what does it mean?'
This value is determined in the driver (dbushelper.py). It shows the state of the Charge MOSFET combined with a variable showing if the battery is allowed to be charged.

JBecker32, 2021-08-23:
"In my case is as well 19200 as default.
The change need to be made is mark.
It show -xA when charging instead of +.
And then alarm / notification can be done as well."

OK, then 19200 seems to be true for the majority of users (2 of 2) :-))
Which change do you mention? (For me it is working the way I send it in the .zip)
Yes, sign of battery current is possibly 'wrong'. (BMS shows that Battery is **delivering positive** current when it is discharged). 
For all the bitwise things I am to stupid to manage this in python ....

pahorpet, 2021-08-23:
I have made some changes in chargin / discharging as it was to strict for me and also not relevent for my size of pack. so now is relevant only to the max_charge/discharge constant. I will made few cycles to prove that is correct than I will share my final version.

JBecker32, 2021-08-23:
> I have made some changes in chargin / discharging as it was to strict for me and also not relevent for my size of pack. so now is relevant only to the max_charge/discharge constant. I will made few cycles to prove that is correct than I will share my final version.

Yes, that is something that should be addressed in a final version. Everybody might have his own idea about it and this should not be 'hardcoded' like it is ....

Louisvdw, 2021-08-23:
I made some changes from the feedback you gave.

[27 more replies]