Louisvdw/dbus-serialbattery #8: Handle multiple battery banks

Louisvdw, 2021-02-08:
Handle multiple battery banks

rocky00717, 2021-04-28:
Any updates on this?

Louisvdw, 2021-05-04:
I'm waiting for my second BMS to arrive. Seem that the factory has run out of RS485 chips.

xerootg, 2021-07-28:
Do you have ideas about what it would take to make this happen? I have two batteries, both ble (nordic) central mode uart, that I'd like to use with your library, but seems like even before jumping into ble code, multiple battery support should happen.

Louisvdw, 2021-07-28:
@xerootg you will first need to get the driver to talk to your one battery, so I would suggest starting with that. There is a battery_template.py file that you can copy as a base for your BMS. Be sure to submit a pull request when you are done. If you need help just should out in the forum.
Once you have your BMS talking to the driver all features will be availble, including this one for multiple battery modules.

liufuyang, 2021-08-11:
Interesting. I am also waiting for my BMS to arrive and I have 2 battery packs. Perhaps I can help testing as well. But these "driver" thing is definitely new to me. Maybe some of you have some good source I can take a quick look somewhere? I also purchased a QUCC BMS I think this will support that well. Looks like the US sanction on China really taking some effects as the chips are now feels really in low supply... The QUCC BMS is not shipped yet as the seller says the RS485 will arrive for them in a few days then he can ship everything to me together.

sideburnie, 2021-09-06:
following

dakoal, 2021-10-01:
Hi. What if 2 TTL-USB Adapters are connected? Not RS485 and two dalys at one Adapter but 2 dalys on two adapters?

ttyUSB0 and ttyUSB1

Louisvdw, 2021-10-01:
> What if 2 TTL-USB Adapters are connected? Not RS485 and two dalys at one Adapter but 2 dalys on two adapters?

You have the same issue with 2x TTL links or 1x RS485 linked to2 BMS. The one battery will override the data from the next. I already have 2 TTL linked BMS connected on my setup so that should work. You can see the data in Remote Console and the alarms from any one will show in VRM, but only the first BMS data is used and display.

Hhalewijn, 2021-12-23:
I have 2 BMS connected. A third one will be connected next week. The two show up in remote console but as stated before, only one can be selected for advanced reporting. I hope this will be added. Please keep me updated on this.

coca7, 2022-01-04:
I have three JBD BMS's in my system.  I have the first one connected using the driver to see how it works with my system.

coca7, 2022-01-04:
Reading through the comments I noticed that of the people who have tried connecting multiple batteries, I don't think anyone has connected more than two.  I connected my three today and two of the three are visible along with my smartShunt.
[image]

Hhalewijn, 2022-01-04:
Hi Coca7,
I've three up and running. It is as Louis says, the number does not matter, it is the behaviour of Victron that does not allow to have more than one BMS that can be selected. I think we would like to have some kind of virtual device that can be selected for Battery Monitoring and that aggregates the information from the individual devices.
[image]

coca7, 2022-01-04:
Thanks @Hhalewijn I knew what Louis said.  I just wanted to log the behavior if no one had observed it yet.

damonblumenstein, 2022-02-17:
@Louisvdw do you have an ETA on this and can it be done over BT with the CerboGX so you don't need to run USB-UART adapters eliminating the ability to use the XiaXang app?

Louisvdw, 2022-02-24:
Bluetooth is not is the near term plan, but there is an ticket for it https://github.com/Louisvdw/dbus-serialbattery/issues/13
The multi bank should be the next feature completed as I am currently working on it and the V2.80 firmware issue.

centro-max, 2022-04-04:
@Louisvdw can you foresee when you will get the update for the multi-bms problem with the victron version 2.8X?

your work is fantastic, compliments!

If you need a tester, i connected 2 daly-bms to my raspberrypi

herrep, 2022-04-28:
@Louisvdw Great work!

At present, I have not set up my Victron inverter/charger + BMS so far, I am gathering information for the correct setup where I will have at least two battery banks in parallel.

What I understand from the screenshots posted above is that only very basic monitoring is possible when connecting more than one BMS via RS485 to the Venus OS based hardware. In this case, all connected BMS are shown and its associated SOC, present voltage and present current. But the detailed analysis will show only one BMS.

Is my understanding correct or did I miss something?

Louisvdw, 2022-05-03:
Hi @herrep 
All the same details are available for each connected BMS when you look at it inside the Remote Console. (and venusOS not V2.80 or higher).
But as a Battery Monitor only the values from one BMS is used in the GX and published to VRM. That is what this ticket is looking to solve.

pedrorampolla, 2022-06-15:
Any updates on this? I have 3 JBD BMS and even renaming them as separate instances would be awesome.    Willing to test.  Using Firmware 2.73  and v0.10 so I can see them all in the device list.

Louisvdw, 2022-06-15:
There is a new 0.11beta4 release that should solve many of the issues the driver had on the newer v2.80 and higher firmware. You should be able to see all the BMS in the device details list like on the V2.73 firmware. If I can get a bit more feedback from more people if this solves the issues in 2.80 I can release it as the next version.

Those issues have been holding back this feature as you can't have multiple banks if you cannot see them.

centro-max, 2022-06-15:
OS 2.87 now shows 2 Daly-BMS with 011beta4 - that is fine!

BUT

the fill level, the total voltage and the current [A] are read out correctly, only the minimum and maximum cell voltage in the detail page is incorrect.
[image]
[image]
[image]

pedrorampolla, 2022-06-15:
Updated to 2.87 and installed beta .11b4.  Now only one JBD BMS shows up at a time.
[image]

Louisvdw, 2022-06-15:
> Now only one JBD BMS shows up at a time.

What install method did you use to upgrade the driver?
And did you reboot afterwards?

Louisvdw, 2022-06-16:
> Blind install with a USB drive. and then a reboot.

I did a change in how the auto install calls the installer in the new v0.12 release. Try and see if that solves your auto install only showing 1 BMS. It might also help to install a new venus firmware - this resets and force the driver to reinstall.

arnoc-intuvision, 2022-06-23:
Hi @coca7,
I would like to know if all 3 your LLT/JBD BMS's are connected to one /dev/ttyUSBx port ? Or is each BMS connected to its own serial port ?

I've added a driver to integrate with Polarium Gen5 batteries, and I can see the first BMS in the Cerbo GX Web UI, but I would like to have multiple Polarium BMS's connected in cascade (RS485), Modbus ID 1, 2, 3, x etc, and have them all show up on the Cerbo GX Web UI as well as in the VRM Portal.

[image]

coca7, 2022-06-23:
@arnoc-intuvision They are on different ports.  On my physical gx device one is plugged directly the other two are plugged in through a hub.

Louisvdw, 2022-06-23:
You need a seperate serial connection for each BMS. The connection works the same for RS485, RS232 or TTL and of these only RS485 have the option to do multiple devices on the same data lines.

arnoc-intuvision, 2022-06-23:
Hi @Louisvdw ,
Yes, I'm using a USB-to-RS485 cable to connect with the Polarium BMS(s). Each battery has 2 ports, A & B, I'm connecting the Cerbo GX device to BMS-1's port A, then port B of BMS-1 connects to port A of BMS-2. So the first BMS will have slave id 1, and the second BMS slave id 2 and so on. It would be great if I only have to use 1 x USB-to-Serial (RS485) cable and read at least 5 x BMS batteries connected in a serial cascading manner.

Louisvdw, 2022-06-23:
Currently that will not work. You need to use a USB-RS485 converter for each BMS. So if you have 3 BMS then you will need 3 USB ports, one to plug in each of the BMS.

janekx, 2022-06-23:
Hi, i have make some progress and now i can monitor 3BMS eqch connected by usb to hub. 
One is 135Ah and two 105Ah battery setups. Driver can set capacity i remeber and becuase i use at first one 135Ah i set 135Ah now on rest 105Ah says capacity 135Ah available 105Ah it means Daly send capacity what is set in BMS. In deiver set 0 to get real from BMS for each connected battry setup?

KRUP85, 2022-07-25:
I have a question about CCCM. If multiple BMS are used (three for example) will this feature work? Will the current (charge/discharge) be limited when any BMS reaches a limit value? I plan to probably also use SMARTChunt, but only as a display of total values.

Thanks

Louisvdw, 2022-07-25:
That is one of the more involved  pieces of the multi bank change. At the moment CCCM will only follow the bank that is published, but when multi banks are supported it will have to look at all the banks, so there will be some changes

XBeg9, 2022-08-15:
I am curious if I have 2 serial batteries in my GX listed without no issues (using 2 JBDs BMS with separate USB-serial connections), but the question is which one to select as the main battery, you can do either one, but not both at the same time, looks like there is a need for the shunt in this case?

Marv2190, 2022-08-24:
This is the issue about.
The driver "needs" to collect all BMS data (of any number of devices), aggregate them and publish it into the venus-os as one device.

herrep, 2022-08-29:
I try to summarize the current behavior of the v0.12 driver when having multiple battery modules with a BMS for each battery module:

1. The BMS information for each battery module is displayed separately.
2. Only the BMS information of the first battery module will be used for DVCC related controlling.
3. Therefore, the DVCC limitations calculated for the other battery banks will be ignored as only the limitations of the first BMS will become active for all battery modules.

Is my understanding correct?

himse1f, 2022-09-05:
is it possible to rename each BMS? I have multiple would love to be able to change their names, if possible. I am running CergoGX v2.89 and the v0.12 release version. I love this driver, thank you!!!!!!!!!
[image]

Louisvdw, 2022-09-06:
There is a ticket oped to do that 
https://github.com/Louisvdw/dbus-serialbattery/issues/100

merlep, 2022-09-12:
> After installing v.12 on my Cerbo GX (v2.87) I can see all three BMS's and the smart shunt. [image]
With 3 JK's and one smart shunt can you force the smart shunt to be the unit that's displayed in the VRM?  If so, is it also the one which the Venus considers the master ( in terms of certain ESS descisions based on SOC )

merlep, 2022-09-12:
Cool, can one disable the CCCM feature ? I do not need/want it to operate - main reason being is that 2 or my 3 JK's jump to 99% SOC as soon as charging resumes after sunrise even when SOC is actually 70% . Really annoying.  The 2 are both 150A JK's , whilst the 200A JK works reasonably well  ( no where near as good as the BMV 702 )

Louisvdw, 2022-09-12:
Change CCCM_ENABLE in utils.py (around line 25/27

corneel27, 2022-09-30:
I am very pleased with your driver. I used it for two JBD 16s connected with two usb - rs485 (with a hub) to a Multiplus gx. I found out that the max current limits are two low with two bms. I cannot charge more than 25 A per battery pack. Also discharge is a factor 2 to low. I have manually changed (100 and 120A) these limits (in utils.py) and now its working good.

Louisvdw, 2022-10-01:
Yes that is part of what it will do. At the moment it thinks there is only one battery to charge, but they will accumulate.

jvaartjes, 2022-10-05:
[image]

Here it works well with 3x Daly BMS and a BMV 712 Battery monitor.
Only thing that would be nice is to somehow being able to rename the the BMSes in GX or in the driver itself. 
Like Daly BMS ttyUSB0, ttyUSB1 etc. that way I can label connectors and USB ports so I always know what Battery bank I am looking at

jvaartjes, 2022-10-05:
[image]
Nevermind, I figured it out. Here the dirty fix for the other users for now. I will fork and find a nicer way to change and create pull request. But I don't have time now.

Edit /data/etc/dbushelper.py

Search for ProductName and change the line to:
self._dbusservice.add_path('/ProductName', 'Batt(' + self.battery.type + ' '+ self.battery.port +') v' +
                                   str(DRIVER_VERSION) + DRIVER_SUBVERSION)

merlep, 2022-10-05:
> [image] Nevermind, I figured it out. Here is the dirty fix for now. I will fork and find a nicer way to change and create pull request. But I don't have time now. Keep in mind that the order of ttyUSBx can change if you move USB connectors around, or if you update Venus OS (I run 2.91). If you make changes, it is wise to plug in one by one and check if your labels match what you see in the remote console. The original Daly USB cables come with a CH340 chip which does not have storage for a serial number.
> 
> Edit /data/etc/dbushelper.py
> 
> Search for ProductName and change the line to: self._dbusservice.add_path('/ProductName', 'Batt(' + self.battery.type + ' '+ self.battery.port +') v' + str(DRIVER_VERSION) + DRIVER_SUBVERSION)

Would you be able to provide a short .csv file from the VRM for your system. I'd like to see exactly what the smartshunt and 3 JK's  provide to vrm, thanks

pos-ei-don, 2022-10-05:
How did you connect an external mqtt Broker to the gx?

jvaartjes, 2022-10-05:
> How did you connect an external mqtt Broker to the gx?
Add the following to your mosquitto.conf (replace the IP 192.168.0.1 with your GX ip) 

connection victron-bridge
address 192.168.0.1:1883
topic # both 0 victron/

merlep, 2022-10-05:
> > 
> 
> I am not sure how to export a CSV with hardware, just statistics. Feel free to message me if you have specific questions. My setup: 2x Multiplus 2 48/3000/35-32 1x Multiplus 2 GX 48/3000/35-32 (buildin GX) 1x BMV 712 Smart (Victron) 3x Daly BMS 16S 150A with USB Uart Serial connector (not RS485) 1x LogiLink UA0148 USB 2.0 HUB (which is powered but use without powersupply) 48x EVE LF280K (AliExpress) AC Coupled SolarEdge SE7K on Modbus TCP SunSpec 1x ET340 on Victron RS485 but I am not using at them moment, due to incorrect values 1x MQTT broker connected to the GX

Go to your system on the Victron VRM portal, select Advanced - specify a time period - Last 1 Hour should suffice , then hit the Cloud icon top right and download .csv file ,thanks
[image]

cbbstewart, 2022-10-13:
> Hi @coca7, I would like to know if all 3 your LLT/JBD BMS's are connected to one /dev/ttyUSBx port ? Or is each BMS connected to its own serial port ?
>
> I've added a driver to integrate with Polarium [...]

Is this driver one you have made or does it work with the standard version? I have a load of these batteries and have no way currently to communicate with them. Would be great to get them connected to a CerboGX. cant find much info online about the rs485 on them either

corneel27, 2022-10-13:
The LLT/JDB bms don't have an rs485 address facility. Si they must each connect with an usb/rs485 converter. You probably need an usb hub to connect all these usb devices.

Louisvdw, 2022-10-13:
There are 2 features that I added to the new beta v0.14~1 that are steps towards multi bank functionality.
You can now give your batteries a [custom name](https://github.com/Louisvdw/dbus-serialbattery/issues/100). 
And more importantly each bms connected now have a device instance with data in VRM
[image]

Give it a test
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.14beta1

arrow1800, 2022-10-14:
installed and rebooted. white screen on the cerbo gx it seems, any idea? 

[image]
[image]

Louisvdw, 2022-10-14:
The latest version also does the GUI changes to see the cell values when the driver is installed. Previously I have seen that it could give a white screen sometimes. The pages that is changed is the battery page.

You can undo the battery page changes by running this script in SSH
> cp /opt/victronenergy/gui/qml/PageBattery.qml.backup /opt/victronenergy/gui/qml/PageBattery.qml
svc -d /service/gui
sleep 1
svc -u /service/gui

arrow1800, 2022-10-14:
> svc -d /service/gui

thix fixed my issue!

arrow1800, 2022-10-14:
> The latest version also does the GUI changes to see the cell values when the driver is installed. Previously I have seen that it could give a white screen sometimes. The pages that is changed is the [...]

> > installed and rebooted
> 
> Which VenusOS firmware are you using? Is it 2.91 or something older?

2.89 should i update?

corneel27, 2022-10-14:
It all works well (venus os version 2.91)
Except one thing: I cann't give my batteries a Custom Name
After battery->device I get:
[image]
No possibilty with a name to change (or do I somethng wrong)?

Louisvdw, 2022-10-15:
> 2.89 should i update?

I think so yes. I merged the GUI elements with the latest changes that was in 2.91 so it is possible that is what give you the blank screen. 
If that is the case I will have to check if there is a better way that the GUI changes will work over more firmware versions

arrow1800, 2022-10-15:
> > 2.89 should i update?
> 
> I think so yes. I merged the GUI elements with the latest changes that was in 2.91 so it is possible that is what give you the blank screen. If that is the case I will have to check if there is a better way that the GUI changes will work over more firmware versions

updated to 2.91 and its working. 

[image]

nice! also great to be able to connect both bms'es. 

this is new; new seen this before, does it have anything to do with the new driver? obviously ess mode is NOT set on External control but on Optimized without battery life

[image]

Louisvdw, 2022-10-16:
External Control is a lable change in the new venusOS v2.91. That is not the driver.

smurfix, 2022-10-19:
Multiple batteries would be a great feature, but I have to admit that I don't see why that should be implemented in dbus-serialbattery.

Shouldn't a program that aggregates the data from all the batteries it sees into one large virtual battery be independent of battery tech? After all, I could have two serial and a CAN battery and the code should still work. They're all visible on DBus, after all …

In other words, I recommend to code this as a standalone program, not as (integral) part of dbus-serialbattery. The code should get its data from the individual batteries' DBus data. It doesn't need any information that the batteries don't already publish on the bus.

[82 more replies]