Louisvdw/dbus-serialbattery #9: Chargery BMS integration

Louisvdw, 2021-02-11:
Add the Chargery BMS basic communication to the driver.

- [ ] SOC
- [ ] Battery voltage
- [ ] Charge/discharge current
- [ ] Temprature
- [ ] Min/max cell voltages
- [ ] protection notifications

LeaseOnLife, 2021-07-25:
Yes please :-)

Louisvdw, 2021-11-09:
If anyone has a extra BMS not being used, I need one to run tests on to make this work.
Or if someone has a bit of technical/programming skills and is willing to test pieces of code for us.

Jabber-se, 2021-11-10:
> If anyone has a extra BMS not being used, I need one to run tests on to make this work. Or if someone has a bit of technical/programming skills and is willing to test pieces of code for us.

Hi Louis, I have Chargery 16S BMS in use but it is not connected to the Venus device yet. I have couple of USB-TTL adapters also, but it is not fully clear for me because in some Chargery related places said about serial ( TTL or RS232? ). Currently I have to check some thinks regarding to the galvanic isolation between Chargery's port and Venus USB port. Chargery BMS has LVD on the negative wire but Chargery and Venus USB ports have connection over negative for signaling wire also. Please give me a several days for double check, planned for weekend. The adapter with galvanic isolation is best solution but I do not have it.
I have good technical and litle bit programming skills, for Venus OS also.

ocopusasia, 2021-11-28:
Hi I have a 16S Chargery BMS  connected to a Venus  GX on 2.80~24  that I can give you access to my machine via Team View and you can use it then for testing

Louisvdw, 2021-11-30:
@ocopusasia  that would be a great help. I'll get in contact when I have something that we can test

ocopusasia, 2021-12-01:
Great please let me know

 

Regards

Paul Bushell

 

Octopus Electrical Service

P:  +66 (0)620596161 

P: Landline  076600143

F:  +66 (0)76600142
E:  ***@***.***> ***@***.***

bloddy, 2022-02-08:
@Louisvdw Hi there, if you have any code wrote already you can publish it and I can take a look/test it, I'm thinking to move to the new victron image with node-red on it .. 

Thank you, and keep the good work.

Louisvdw, 2022-02-24:
Hi @bloddy 
I have not had time yet to look at the Chargery BMS much yet. It's still in the planning ToDo stage

Louisvdw, 2023-01-11:
Nothing has been done. We have been working on other BMS and changes