Louisvdw/dbus-serialbattery #13: Add bluetooth support

Louisvdw, 2021-03-15:
It would be good to have the driver use bluetooth to connect from the VenusOS device to the BMS
The latest rPi images has bluetooth enabled. A GX device will need a USB bluetooth dongle.

It will need to

- [ ]  Pair the device and the BMS
- [ ] Send commands to connected bluetooth
- [ ] Read values from connected bluetooth

damonblumenstein, 2022-08-24:
I would love to see this come to fruition. I have 2 propane sensors on BT now and its awesome. This would make total sense!

TheEldar, 2022-09-19:
did some tests with a Raspi and my Renogy RBT200LFP12-BT. Sorry it's not for the ANT BMS. But maybe it will also Help to implement the BT Support.

The Solar Monitor from Olen works and connects to the BT module of the Renogy. The scripts are meant for MQTT but maybe you can use the BT part and for integration in the GX devices?
[Olen on GitHub](https://github.com/Olen/solar-monitor)
[](url)
And an other Script on reddit
[Link to Reddit Post](https://www.reddit.com/r/SolarDIY/comments/frafmv/comment/hoqc2wu/?utm_term=38500721598&context=3&utm_medium=comment_embed&utm_source=embed&utm_name=a9879548-ce16-11ec-9708-ae6ecb37ff8a)

baranator, 2022-11-27:
Last week i got the last parts for my victron battery system and tried to wire it up. the dbus-serialbattery wouldn't work. After some testing i came to the conclusion out that the serialport of my jk-b2a24s20p is broken as i couldn't get it running with any serial/dongle-combination other people describe as working.  Therefore i had to look for an alternative:

I spent my weekend on figuring out how to communicate with my jk-b2a24s20p via python-bleak via Bluetooth. Good news: I think i've already manages the hardest part and have all relevant readings (threshold & protection-voltages, per-cell-voltages and so on in a python-dict. 

In the next days i would have a closer look at your dbus-serialbattery implementation and how to fit this in. I would assume the easiest way would be to create another Battery-Subclass as a wrapper for the functions i wrote today in etc/dbus-serialbattery/jkbms.py e.g. jkbms_ble.py, so that while running dbus-serialbattery simply wouldn't only poll all the currently implemented serial-connections, but would also try to build up a bluetooth-connection and poll data from there.

Is this a/the way to go? Anything else to consider?

Louisvdw, 2022-11-28:
The protocol would be the same via normal wired serial communication and bluetooth. The setup of bluetooth and the address would be the big differences. But to make it easier you can just make a new battery type like you suggest and add that to the types to check. As long as your type has all the requered function calls it will be tested. 
Make a draft Pull Request even though you have not finished it. That way we can see what you have done and give you feedback and we can merge when it is done.

baranator, 2022-12-02:
Okay, i forked the repo, but for now i will concentrate on building a class that solely concentrates on retrieving the jkms-data via ble. I put that pre-alpha-test on github, see: https://github.com/baranator/jkbms_brntr 

This is a single file, that should already be kind of runnable and print some values from the battery every few seconds; the library is going to create a background-thread, that handles all the incoming bluetooth-stuff and will provide a get_status() function that can be called any time to get the freshly updated values. I took many parts from https://github.com/syssi/esphome-jk-bms and translated the c-code to python. As i do not have that much experience with python i dont dare to guarantee for any kind of code-quality :D

baranator, 2022-12-10:
Did some first steps on my fork: https://github.com/baranator/dbus-serialbattery/commit/fe6b89cfb2a1623bd835d60f76b5bcd9102c684d 

this isn't running just yet, just trying to figure out how to put things together...
to make it work you need to  install the `bleak ` python-package, what won't do fully automatic
to get it installed, run `opkg install python3-pip` and then bring it up to the newest version `pip install --upgrade pip`. if you then try to install `bleak` via `pip install bleak` you will get some weird errors, that i  provisionally got rid of by installing all available python-modules :S via `pip install python3-modules` ... this should be reduced to the really needed packages later :) 

(just wanted to give a sign of life)

mr1978, 2022-12-14:
Perhaps i can help by installing the required modules. The trick is you have to download the modules via wget, extract an install it manually. 

Example:

cd /opt/victronenergy/inst
wget https://github.com/hbldh/bleak/archive/refs/heads/develop.zip
unzip develop.zip

By my testings I had the problem that some modules are missing:

cd /opt/victronenergy/inst
wget https://github.com/Bluetooth-Devices/dbus-fast/archive/refs/heads/main.zip
unzip main.zip.1

cd /opt/victronenergy/inst/dbus-fast-main/src
mv dbus_fast /opt/victronenergy/

cd /opt/victronenergy/inst
wget https://github.com/aio-libs/async-timeout/archive/refs/heads/master.zip
unzip master.zip

cd async-timeout-master
python3 setup.py install

cd /opt/victronenergy/inst
wget https://github.com/dreadnought/python-daly-bms/archive/refs/heads/main.zip

unzip main.zip
cd python-daly-bms-main/
python3 setup.py install

I hope this helps.

idstein, 2022-12-19:
> this isn't running just yet, just trying to figure out how to put things together... to make it work you need to install the `bleak ` python-package, what won't do fully automatic to get it [...]

Thanks for those valuable instructions. For me the following worked to get custom python packages on board:
`
opkg update
opkg install python3-pip python3-modules
`

For the concrete serial rfcomm with a BLE GATT device, I use `ble_serial`:

`
pip3 install ble_serial
`

For JBD BMS the following service, rx and tx UUIDs work to construct a local serial socket forwarding:
`
ble_serial -p /tmp/ttyBLE1 -s 0000ff00-0000-1000-8000-00805f9b34fb -r 0000ff01-0000-1000-8000-00805f9b34fb -w 0000ff02-0000-1000-8000-00805f9b34fb
`

Sample output running on a RPi4 Venus OS:
`
root@raspberrypi4:~# ble-serial -s "0000ff00-0000-1000-8000-00805f9b34fb"
21:20:30.075 | INFO | linux_pty.py: Port endpoint created on /tmp/ttyBLE -> /dev/pts/2
21:20:30.076 | INFO | ble_interface.py: Receiver set up
21:20:30.077 | WARNING | ble_interface.py: Picking first device with matching service, consider passing a specific device address, especially if there could be multiple devices
/usr/lib/python3.8/site-packages/ble_serial/bluetooth/ble_interface.py:28: FutureWarning: This method will be removed future version, pass the callback to the BleakClient constructor instead.
  self.dev.set_disconnected_callback(self.handle_disconnect)
21:20:31.233 | INFO | ble_interface.py: Trying to connect with 70:3E:97:D2:F4:D0: JBD-AP21S002
21:20:32.163 | INFO | ble_interface.py: Device 70:3E:97:D2:F4:D0 connected
21:20:32.165 | INFO | ble_interface.py: Found write characteristic 0000ff02-0000-1000-8000-00805f9b34fb (H. 20)
21:20:32.167 | INFO | ble_interface.py: Found notify characteristic 0000ff01-0000-1000-8000-00805f9b34fb (H. 16)
21:20:32.225 | INFO | main.py: Running main loop!
^C21:29:59.848 | WARNING | main.py: Shutdown initiated
21:29:59.849 | INFO | linux_pty.py: Serial reader and symlink removed
21:30:02.017 | WARNING | ble_interface.py: Device 70:3E:97:D2:F4:D0 disconnected
21:30:02.018 | INFO | ble_interface.py: Stopping Bluetooth event loop
21:30:02.020 | INFO | ble_interface.py: Bluetooth disconnected
21:30:02.021 | INFO | main.py: Shutdown complete.
`

This will construct a local tty file endpoint backed by BLE GATT device with a RX/TX characteristics using write / notify. Unfortunately, I haven't found a way to inform serial-starter about this new pseudo terminal device. But reading this here it does not sound like an easy route https://unix.stackexchange.com/questions/240352/control-pseudo-terminal-dev-pts-assignments.

Any interesting on the Python code can take a look at the attached patch file
[jbd_ble.patch](https://github.com/Louisvdw/dbus-serialbattery/files/10262866/jbd_ble.patch)

baranator, 2022-12-26:
slowly getting somewhere :)

https://bilderupload.org/bild/7e0a90548-jkbms

baranator, 2022-12-30:
created a pull-request draff, so further progress will be handled there: https://github.com/Louisvdw/dbus-serialbattery/pull/372

mr-manuel, 2023-05-01:
See this discussion for the big progress: https://github.com/Louisvdw/dbus-serialbattery/discussions/485

The Bluetooth connection for the JKBMS is now working fine since 4 months and the cornerstone for adding other Bluetooth drivers was placed.

mr-manuel, 2024-02-16:
Was implemented for most used BMS by community. Further PR for more BMS are very welcome.

closed 2024-02-16 by mr-manuel