Louisvdw/dbus-serialbattery #15: self.publish_battery_dbus() crash, stops logging data

mstealth22, 2021-04-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install successful, and device found, and logs data for around an hour or so.

This is error is found logs: 

@40000000608770e81d459954 *** CCGX booted (0) ***
@400000006087711627f5a31c INFO:__main__:dbus-serialbattery
@40000000608771170076dbbc INFO:__main__:JBD-SP04S020-L4S-100A-B-U
@4000000060877117189ce164 INFO:__main__:Battery connected to dbus from /dev/ttyUSB1

@400000006087868c15d7b29c ERROR:__main__:>>> ERROR: No reply - returning
@400000006087868c352bc73c ERROR:__main__:>>> ERROR: No reply - returning
@400000006087868d342933ec ERROR:__main__:>>> ERROR: No reply - returning
@400000006087868e356015dc ERROR:__main__:>>> ERROR: No reply - returning
@400000006087868e3587d98c Traceback (most recent call last):
@400000006087868e35ca783c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 414, in publish_battery
@400000006087868e35ea21b4     self.publish_battery_dbus()
@400000006087868e35ed80fc   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 234, in publish_battery_dbus
@400000006087868e35f2053c     self._dbusservice['/System/MaxVoltageCellId'] = 'C' + str(max_cell + 1)
@400000006087868e35f4fb0c TypeError: coercing to Unicode: need string or buffer, int found

**Expected behavior**
To keep running and logging data.  It is very possible I have an issue with my setup or an error in my install.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: [e.g. VenusGX, Cerbo, Raspberry Pi]
 - Firmware Version [e.g. 262]

 ***Raspberry PI 3B+, VenusOS 2.66

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: [e.g. LTT, Daly]  JBD-SP04S020-L4S-100A-B-U
 - Cells: [e.g. 16] 4
 - Interface: [e.g. USB-RS485, USB-RS232, Ve.Direct] USB-RS232 box that came with Overkill Solar BMS.  I do have and tried a VEDirect USB cable but it did not try to run the driver for it.

**Additional context**
Add any other context about the problem here.

Louisvdw, 2021-04-30:
Hi @mstealth22 
These BMS will sometimes shutdown if a critical error happens and then cannot be contacted anymore until the problem is resolved. It mostly happens when the battery is close to full (or empty) and the cells are not balanced. 

Check if you see any cell imbalance messages in the Victron VRM -> Alarm logs. If you do, then one of your cells have reached the limit you set up in in the BMS (using the App on your phone).
Also check what your SOC % are when you get this error.

If you do find that there are cell imbalance notifications, then setting the charge voltage lower for a few weeks should give the BMS a chance to balance the cells. You can do this override in the latest VenusOS versions under the DVCC. I suggest you use V13.4 (V3.35 x 4 cells). The default values for your BMS should be so that the cells will be balancing from V3.30.

Louisvdw, 2021-05-16:
Hi @mstealth22 
Your log show that the BMS gave no reply after being queried. So for some reason it is going into a shut down state. 
The best would be to find out why that state happens, but I have added a few checks that should get you past the error you were seeing. 

Try this build and let me know if it solves your error.
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.4beta2

closed 2021-05-26 by Louisvdw

Louisvdw, 2021-06-18:
There is a new build that should fix this issue.