Louisvdw/dbus-serialbattery #20: dbus-serialbattery.py crashing with an unpack error

robisonjoel, 2021-06-11:
**Describe the bug**
I installed the package manually on a dev build of VenusOS (2.70~17), ran basic ethernet configuration and enabled SSH to install this package.

I have two USB TTL adaptors installed, neither or them are Victron units - one is connected to a VE.Direct for my SolarCharger (which is still working great) and the other is connected to the JBD Modbus/USB adapter.

The Log contains this information:
```
root@raspberrypi2:~# cat /data/log/dbus-serialbattery.ttyUSB0/current
@4000000060c29d5336ae8af4 INFO:__main__:dbus-serialbattery
@4000000060c29d540f24462c INFO:__main__:JBD-SP10S009-L8S-100A-B-K
@4000000060c29d542329b38c Traceback (most recent call last):
@4000000060c29d54232b807c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 534, in <module>
@4000000060c29d54233505fc     main()
@4000000060c29d542335adf4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 521, in main
@4000000060c29d54233ef8dc     battery.read_gen_data()
@4000000060c29d54233f951c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 376, in read_gen_data
@4000000060c29d5423465f64     = unpack_from('>HhHHHHhHHBBBBBHH', gen_data)
@4000000060c29d542347651c struct.error: unpack_from requires a buffer of at least 27 bytes
```
I can confirm this by running it manually as well via `/data/etc/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB0`
```
root@raspberrypi2:~# /data/etc/dbus-serialbattery/dbus-serialbattery.py /dev/ttyUSB0
INFO:__main__:dbus-serialbattery
ERROR:__main__:>>> ERROR: Incorrect Reply
INFO:__main__:JBD-SP10S009-L8S-100A-B-BK
Traceback (most recent call last):
  File "/data/etc/dbus-serialbattery/dbus-serialbattery.py", line 534, in <module>
    main()
  File "/data/etc/dbus-serialbattery/dbus-serialbattery.py", line 521, in main
    battery.read_gen_data()
  File "/data/etc/dbus-serialbattery/dbus-serialbattery.py", line 376, in read_gen_data
    = unpack_from('>HhHHHHhHHBBBBBHH', gen_data)
struct.error: unpack_from requires a buffer of at least 27 bytes
```

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version: 2.70~17

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JBD (https://overkillsolar.com/product/bms-100a-8s-lifepo4/ - re-soldered to be 7s)
 - Cells: 7
 - Interface: USB-RS485 + USB-RS232 w/Ve.Direct

**Additional context**
Add any other context about the problem here.

robisonjoel, 2021-06-11:
I want to point out that I'm very excited to see this project, its just what I'm looking for - Thank you for your efforts, ill certainly donate

Louisvdw, 2021-06-16:
Hi @robisonjoel 
It seems that your BMS is not sending the full message back from the BMS. It is possible that your BMS does not send the 2 bytes for the high balance status, but that is just speculation. I'll check if there is an updates protocal document that perhaps mention this for the 8s BMS.

Louisvdw, 2021-06-29:
@robisonjoel there is a new release v0.5 that has some improved checks implemented. Can you test your issue with v0.5 and let me know if you still get the same error.

Louisvdw, 2021-08-17:
Is this still a problem or can the ticket be closed?

robisonjoel, 2021-08-18:
Works great now - Thank you!

closed 2021-08-18 by robisonjoel