Louisvdw/dbus-serialbattery #24: 3S and 4S BMS Support

yannickudb, 2021-06-28:
3S And 4S BMS have different communications.
My contact at Daly says this should be the document that describes the communication between 3 and 4S BMS and PC.

[Daly RS485+UART protocol.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/6728032/Daly.RS485%2BUART.protocol.pdf)

dreadnought, 2021-06-28:
The attached PDF is the same as in #3, but it is for a different protocol. Daly sells BMS with two different chipsets under the same brand. You can read through this discussion if you want to know all the details: 
https://github.com/dreadnought/python-daly-bms/issues/1

I didn't find documentation neither, but I've implemented most of what it can by capturing the communication of the Windows tool. Here is the code:  
https://github.com/dreadnought/python-daly-bms/commit/8799c7bf70ef9205e428aeab3c635ece5afdc1cc

unifiedcommsguy, 2021-07-19:
I have a Daly 4s with sinoweath protocol and a VenusOS and happy to assist in testing with this.

I'm just looking at the code now and will see if I can provide some feedback/assistance.

liufuyang, 2021-08-20:
I have a Daly/MGod 10S BMS and I have some issue using it https://github.com/Louisvdw/dbus-serialbattery/issues/36

Perhaps this is related to it? Thanks.

Carstijn, 2021-09-01:
I have a Daly Smart BMS 250A, already connected it to my venus GX, and started on a implementation. For now I can read the first values, still needs some work before I can make a pull request. (need to polish my python skills a bit, the python 2.7 in the venus OS does not help)

For the implementation, there is actually quite a bit information on the protocol in the PDF from the BMS tool installer. This is partly in Chinese, but with translate and the original side by side there is a lot to make out.  Even bitmaps for some of the status/config commands are given.
[Sinowealth BMS Tool Introduction V0.2.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/7092766/Sinowealth.BMS.Tool.Introduction.V0.2.pdf)

Carstijn, 2021-09-02:
I just pushed a first version to my repo. It needs some cleanup and fetching a bit more info from the BMS in stead of hardcoded variables, and probably has some bugs. Feel free to test/comment. I guess I will continue the implementation the coming days, and will post a pull request when I feel its ready for a first version.

But hey, it works!
[image]

Carstijn, 2021-09-04:
First version of the pullrequest is up, so have a look! #41

Louisvdw, 2021-09-06:
Nicely done @Carstijn 
I'll review the pull request as soon as I can.

closed 2021-09-06 by Louisvdw

Louisvdw, 2021-09-08:
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.7beta2
includes the Daly with Sinowealth chip for those that would like to try it and give feedback

unifiedcommsguy, 2021-09-12:
> 
> 
> https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.7beta2
> includes the Daly with Sinowealth chip for those that would like to try it and give feedback

Thankyou for this. Will test out and let you know this week.