Louisvdw/dbus-serialbattery #25: DIY BMS V4

dakoal, 2021-06-30:
https://github.com/stuartpittaway/diyBMSv4

This BMS would be fine to be integrated

Louisvdw, 2021-06-30:
Hi @dakoal 
Is there a communication protocol document available?
I can see there is an MQTT and NodeRed server, but is there a serial interface?

dakoal, 2021-06-30:
I am in contact with Stuart. Maybe in the near future the MQTT JSON output will be delivered on the USB/Serial Port.
When you connect a DIY BMS on the USB Port of a cerbo, you got the data every second (now its only debug data, but its coming on /dev/ttyUSBXY).

I will inform here when it's available, provide Test data and the format.

Louisvdw, 2021-06-30:
That would be perfect. It it connects to /dev/ttyUSB then the serial-started service on the Venus can start my driver. If we can read the battery data from there it will work.

Louisvdw, 2021-06-30:
As a minimum we need to know: 
The amount of cells in the battery 
And then also the SOC, Voltage, Current.

Cell data, temp data, alarms, charge limits are all extras

CuriousTimo, 2022-06-23:
Stuart has implemented CanBus on the diyBMSv4. This is directly compatible with Victron.
So I see no need to connect the diyBMSv4 over serial.

dakoal, 2022-06-24:
That is true when using a cerbo.

But VenusOS on a raspi doesnt have canbus

CuriousTimo, 2022-07-28:
Just add an MCP2515 CAN-Bus HAT.
https://github.com/victronenergy/venus/wiki/RaspberryPi-CAN-Interfaces