Louisvdw/dbus-serialbattery #26: Daly Smart BMS Discharge / Charge Mosfet Switching

MeisterQ, 2021-07-05:
I would like to ask, if you could support for switching on / off the FETs from Daly Smart BMS?

With the original bluetooth dongle, i can turn them on or off with the app.

Id like to disconnect the whole battery on low SoC for example to save energy in critical situations.

Louisvdw, 2021-07-07:
@MeisterQ just to confirm. 
1) Do you want to manually switch the battery off? 
2) Do you want to stop battery usage when the SOC gets to a low value? 
The FETs will only stop power flowing out, but the BMS will still use some power to opperate, so I cannot see a benefit for 1). 
You can already do 2) to stop the use of the battery at a SOC level using the settings inside the GX device.

MeisterQ, 2021-07-07:
@Louisvdw 

1. Yes, exactly
2. Yes, absolutely, i know about the BMS will still take power from the Battery, but my Solarinverter will stop getting energy out of the battery if i disable discharge MOSFETS, then only solaroverproduction will charge batterys, but there is no way for energy to leave the batterys at low SoC.

My inverters minimum powerproduction is around 27W, if i turn off the discharge FETs,  it will stop producing any watt.

With bluetooth dongle it is possible, so i think with this addon i could turn them on/off also, right?

Louisvdw, 2021-07-07:
The driver does not have an interface UI option to set your FET like in 1). If you connect the BMS to the GX using a serial port that does not remove your bluetooth, then you can still use the Daly app on your phone and switch the FET manually. Some devices only have one UART so you need to plug the serial connection in where the bluetooth was, so then you do not have bluetooth app access anymore.

The driver can limit the power usage, but that is only through the GX device currently. It will read the FET status and pass that on to the GX. So the min SOC settings that is set up i the BMS will be what limit the FET. However these alerts are not fully implemented in the Daly yet.

I think what you want to do would have somefulness in an off-grid use, but not that much for grid tied.

MeisterQ, 2021-07-07:
I know, i dont want to use the bluetooth anymore, i just want to connect it to my raspberry where venus OS is running on.

With this possibility, i could switch the FETs on or off when i want with MQTT. 

Thats my thought about this. It is also for grid tied operation usefull, because then i can limit the inverter power to the power my PV system is producing at the moment, so it cant draw power from battery.

Do you think this is possible?

Bagarre, 2022-01-17:
Interested in this capability as well (turning on/off charge/discharge mosfets without using bluetooth app)
Does anyone know what the commands look like being sent to the BMS? I can find lots of examples of querying the BMS for info but none for configuring a SinoWealth BMS and can't find an easy way to sniff bluetooth traffic.

mr-manuel, 2023-05-12:
I assume your Venus OS is also connected after the BMS and not before. In this case you anyway have to use the smartphone to turn it on again or am I wrong?

mr-manuel, 2023-05-26:
This was implemented.

closed 2023-05-26 by mr-manuel