Louisvdw/dbus-serialbattery #28: Support for ECS Lipro BMS

MeisterJoda1979, 2021-07-23:
Hi, 
German company ECS has some nice BMS called LiPro, with very high balancing current and high measurement accuracy. The have active BMS too, because it is cell bases it supports more then 200 cells. 
They use modbus RTU with 19200 baud and even parity. Register Description is at the end of user manuals. 
https://ecs-online.org/BMS:::2.html

It would be great to implement this Lipro BMS. 
If there is any question you can contact support at mail@ecs-online.org

Louisvdw, 2021-07-27:
I see the German manual has the commands at the end. It looks like it should be able to be added to the driver if there is enough intrest from users

FalkoJahn, 2021-08-02:
Integrating our components would be great. With us, the SOC comes from a device called the greenMeter. This is similar to the BMV 700 (only more precisely :-))
The LiPro's are held for cell monitoring and charge balancing. We are currently revising the LiPro's, if you want to integrate them, it would be best to implement the latest version right away.

We are happy to send you LiPro's and greenmeter free of charge if you integrate them.

Louisvdw, 2021-08-02:
Hi @FalkoJahn 
Send me a PM on [the forum ](https://energytalk.co.za/u/louisvdw)and we can talk.

kazham, 2021-08-05:
Hi @Louisvdw! It would be great if you could implement ECS devices driver! Thanks for your unimaginable efforts.
By the way, greenMeter is the most precise current sensor in the industry.

Chrischan-git, 2022-04-23:
Hi @Louisvdw 
I‘m also really Interested for supporting the ECS BMS in your project. Are there any news about a upcoming implementation?
If I can support with testing or anything else, let me know…
Thanks

gaeguu, 2022-06-11:
I would be interested too!

Louisvdw, 2022-10-12:
For those that want to give the GrreenMeter and LiPro devices a test, have a look at the [new v0.14~1 build](https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.14beta1).
It requires a GreenMeter which will be the battery part, and then add your LiPro devices to that for the cell values. 

You need to edit utils.py around line 88 and set the following values to match your GreenMeter and LiPro setup with the Modbus address your set each device with. It works best if your LiPro devices have sequential address as the driver will be searching for the LiPro's between the start and end address spesified.

>nano /data/etc/dbus-serialbattery/utils.py

>GREENMETER_ADDRESS = 1
LIPRO_START_ADDRESS = 2
LIPRO_END_ADDRESS = 9
LIPRO_CELL_COUNT = 8

Reboot your GX after you have made these changes and it should pick up your battery. 
For now alarms are only from the GreenMeter. I still need to add alarms from each LiPro as well.
Let me know if it works.

gaeguu, 2022-10-12:
Hi @Louisvdw 
I'm using the Lipro's standalone - no greenMeeter nor greenView - together with the Victron Smart Shunt.
Can I still use your dbus-serialbattery to visualize/monitor the Lipro's?
The SoC etc. is coming from the Smart Shunt.

Louisvdw, 2022-10-12:
The GreenMeter gives the data to be able to create the battery object. The cell values from the LiPro is then attached to that. I'll have to think if we could make it work for the scenario that you are using them with, but it will not work with the current version.

Chrischan-git, 2022-10-12:
Yay, I was so pleased about the news but unfortunately the same here. I Only use the LiPro standalone with Winston cells.
That's why I asked before whether there would also be support for the Lipro…

Louisvdw, 2022-10-13:
So if you only have LiPro devices for the cells, then you don't have anything that can measure the battery's current flow, State Of Charge, Voltage and things like that?
I guess we could just publish all the cell voltages under a dummy battery shell. But you would not be able to use that as a Battery Monitor inside your GX. It will just have the cell data available. Is that all you need?

gaeguu, 2022-10-13:
To be honest I haven't looked into how GX is logically "organized".
From the top of my head I would imagine that the LiPro's/Cells can be assigned to my existing Smart Shunt (VRM Instance) and therefore extend the Smart Shunt's information (as you said: battery's current flow, State Of Charge, Voltage and things like that). Would that be feasible?
My current Smart Shunt has VRM instance id 276 - so I would want to configure my LiPro's to extend VRM instance 276 with cell voltage information ..

Chrischan-git, 2022-10-13:
Oh, sorry. of course I also have a Victron BMV712 in my setup. So the shunt is actually delivering the SOC and current flow to my Cerbo-GX.
I hope this is a workable scenario…

nichtdiemama84, 2022-12-12:
hi I would like to use the Lipro bms without the greenmeter. I have 4 Lipro devices and an BMV 702. is this setup possible right now?

Louisvdw, 2022-12-12:
Hi @nichtdiemama84 
At the moment you need the Greenmeter to be able to use the LiPro devices.

Louisvdw, 2022-12-12:
I still have to investigate if that is possible, but it is not off the list yet.

nichtdiemama84, 2022-12-12:
Your work is really cool thanks a lot for that. If you manage to add the lipro informations to the BMV that would be great.

again thanks a lot for your work.

Chrischan-git, 2023-04-15:
Any news/progress here for standalone LiPro-BMS?

Chrischan-git, 2023-05-05:
@mr-manuel what do you mean by that comment/link?

mr-manuel, 2023-05-05:
It's only linking another issue.

kletteroli, 2023-07-02:
Thanks a lot for your commitment!! 
I would also highly appreciate the inclusion of the option: Lipros WITHOUT the GreenMeter/GreenView (but in combination with a BMV700/Smartshunt etc.)

closed 2024-02-22 by mr-manuel