Louisvdw/dbus-serialbattery #29: JKBMS Venus OS integration

Marcodit, 2021-08-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Marcodit, 2021-08-01:
> 
[BMS-CAN2-english (1).pdf](https://github.com/Louisvdw/dbus-serialbattery/files/6913139/BMS-CAN2-english.1.pdf)

Marcodit, 2021-08-01:
[BMS-CAN2-english (1).pdf](https://github.com/Louisvdw/dbus-serialbattery/files/6913141/BMS-CAN2-english.1.pdf)
[BMS CAN instruction(2).pdf](https://github.com/Louisvdw/dbus-serialbattery/files/6913143/BMS.CAN.instruction.2.pdf)
[BMS 485 instruction(1).pdf](https://github.com/Louisvdw/dbus-serialbattery/files/6913144/BMS.485.instruction.1.pdf)

juanmijm, 2021-08-02:
Ohhh, I think there are a lot of people interested in this, hopefully it can be done. THANK YOU!

domoticadcm, 2021-08-02:
wow!! it´s just what I need!!! thanks in advance

baizan1, 2021-08-02:
Yes, i need this integration too, my JK don't comunicate with Victron. Thanks.

gcruzsan, 2021-08-02:
A great BMS, and with this project would be even bigger.

Louisvdw, 2021-08-02:
Is anyone willing to sponser a BMS that we can use for development?

Marcodit, 2021-08-02:
> Is anyone willing to sponser a BMS that we can use for development?

How it can be done?

Louisvdw, 2021-08-02:
> > Is anyone willing to sponser a BMS that we can use for development?
> 
> How it can be done?

If you have a BMS that you do not use, you can send that to me.
Or I can look at price to order one and we can get sponsers through the Ko-fi sponser link. Just mention JKBMS in Ko-Fi and I will know.

Marcodit, 2021-08-02:
> > > Is anyone willing to sponser a BMS that we can use for development?
> > 
> > 
> > How it can be done?
> 
> If you have a BMS that you do not use, you can send that to me.
> Or I can look at price to order one and we can get sponsers through the Ko-fi sponser link. Just mention JKBMS in Ko-Fi and I will know.

Ok, thank you,  I will try to get sponsors, can you have a look at the BMS, and if you need just one with the less AMP is less expensive.. because for the same model, there are various AMP's.

diybateriaslifepo4, 2021-08-02:
> 
> 
> Is anyone willing to sponser a BMS that we can use for development?

Hello, I need to talk with you about this, how can we talk in private?
send me a email for example: DIY.LIFEPO4.europeangroup@gmail.com

domoticadcm, 2021-08-02:
https://www.aliexpress.com/item/4001192978656.html
there you can get BMS, rs485, and/or CAN modules for the easiest connection

Louisvdw, 2021-08-02:
> > Is anyone willing to sponser a BMS that we can use for development?
> 
> Hello, I need to talk with you about this, how can we talk in private?
> send me a email for example: [DIY.LIFEPO4.europeangroup@gmail.com](mailto:DIY.LIFEPO4.europeangroup@gmail.com)

You can send PM to me on the forum
https://energytalk.co.za/u/louisvdw

juanmijm, 2021-08-02:
> You can send PM to me on the forum
> https://energytalk.co.za/u/louisvdw

ok, you have MP ;)

Aea263, 2021-08-03:
Having support for jkbms would be awesome!!!!!

sgregori, 2021-08-04:
Wow, that's really interesting to use it in my JK BMS. Thanks!

RTW22, 2021-08-04:
I think what you do is incredible

sapcmc, 2021-08-08:
Made a contribution via kofi. Let's get this done

JoanQA, 2021-08-22:
That's really interesting.

Louisvdw, 2021-09-03:
I've created the first beta build for the JKBMS. 
If anyone wants to try it out and let me know if you find any problems.
Large features still outstanding is any cell data (min/max cells and voltages) and alarms.

https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.7beta1
https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install

Louisvdw, 2021-09-08:
v0.7 beta2 adds support for cell values and a few fixes.
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.7beta2

Louisvdw, 2021-09-09:
JKBMS is now supported in the latest release build

closed 2021-09-09 by Louisvdw