Louisvdw/dbus-serialbattery #31: unloading ON/OFF from vitronOS + installed and available capacity + Read alarms

Fabbec1, 2021-08-05:
1. Allow unloading ON/OFF MosFet as a function from victronOS - as a function Controllable e.B of ModBus address 280. 
[image]

2. Get installed and available capacity from the daly 
[image]

3. Read alarms
[image]
[image]

thx for your Work on this Driver

Louisvdw, 2022-02-06:
Alarms have been included in PR #90 and should be in the next version released

transistorgit, 2023-04-28:
Installed capacity is read from the config
remaining capacity is installed capacity * SOC
both is already displayed
[image]

Is this not what is wanted? Should it be read from the bms?

mr-manuel, 2023-04-28:
It would be nice, if the installed capacity is read from the BMS. Is this data available? This would help to identify multiple BMS by changing slightly the capacity on the BMS.

transistorgit, 2023-04-28:
as of the protocol snippets @teefix posted, it should be possible to read the rated capacity. But also the BMS board number. Will look into both and also try to set SOC, as it is usefull for me also.

mr-manuel, 2023-04-28:
That would be great!

transistorgit, 2023-04-28:
capacity is working.  It is read from bms now.
[image]

board number not. I can read it, but when I use the daly pcmaster software to set a number other than 1 the bms stops working with "aquisition aef error" or so... so I stoped working on it.

mr-manuel, 2023-04-28:
Unfortunately I have no hint.

teefixx, 2023-04-28:
I use a flow in Nodered, which switches to float voltage, after the max charge is reached for 15min and stays there until SOC is below 95%. Still the BMS always is only on 99.x%. My plan is to trigger SOC reset to 100% when I change to float voltage...
I'm not an experienced programmer, so don't blame me if this could be done in a shorter way, I anyway attach the flow for your information. The flow is used to control CVL via a PID controller not to overcharge the cells, the "float or not" function controls if CVL is changed to float voltage.

[image]
[float.txt](https://github.com/Louisvdw/dbus-serialbattery/files/11356266/float.txt)

teefixx, 2023-04-28:
> capacity is working. It is read from bms now. [image]
> 
> board number not. I can read it, but when I use the daly pcmaster software to set a number other than 1 the bms stops working with "aquisition aef error" or so... so I stoped working on it.

It's odd that changing the board number does not work, there is a guy at the diy solar forum who did it successfully :https://diysolarforum.com/threads/interfacing-several-daly-bms-via-rs485.17998/#post-215831

After that he was able to send gather serial data from the different boards by exchanging X40 to X41 for board No2: https://diysolarforum.com/threads/decoding-the-daly-smartbms-protocol.21898/page-2#post-385365

mr-manuel, 2023-05-01:
1. Is handled with https://github.com/Louisvdw/dbus-serialbattery/issues/26
2. Is done.
3. Is done.

closed 2023-05-01 by mr-manuel