Louisvdw/dbus-serialbattery #33: Would our plug in be used with those smart shunt with RS485/USB output?

liufuyang, 2021-08-11:
Hi there, as I can see the Victron SmartShunt is quite expensive, it might be interesting or benefit to buy a SmartShunt like [this one](https://www.aliexpress.com/item/32916627116.html) then use the Raspberry-Pi + Venus OS + this very nice plug-in to monitoring the data via Victron VRM web site. 

This could be especially useful in case someone's BMS doesn't have a RS485 output then a simple shunt like this might be used.

I also noticed [on Youtube someone](https://www.youtube.com/watch?v=vvQcoQesAD0) used this device to read voltage and current info from the RS485 output.  

So perhaps it is possible? If so, I might want to order a part like this to try it out. And if you also want to play it out, I can also a similar part from aliexpress and deliver it to you directly :)

I have very limited USB/485 knowledge, but if this can be not difficult I then would like to learn more about it :)

Thank you.

Louisvdw, 2021-08-12:
Hi @liufuyang 
The short answer is that an external shunt would be a waste of money and is not really useful for Lithium batteries.

In almost all scenarios you would not need a smartshunt or similar when using a Lithium battery, as Lithium batteries require the use of a BMS. And most BMS does their measurements with a shunt already. So you will just add a duplicate.

Shunts are an acurate way to measure current flow. The Victron Smartshunt is a very good shunt (not all are made the same) and it will give you State of charge, some history, the battery voltage and current and is already capabile to talk to the VenusOS without the help of a driver like this. But it does not give you all the data the BMS can.

Lithiums need a BMS for safety. If any cell is overcharged it can cause a fire, so a BMS is used to monitor each cell and stop any charging when a cell reach an overvoltage state. Those are the balancing wires you see going to each cell. This extra information is what is valuable to us and gives the advantange to the BMS over any external measurement device.

Now if you were using Lead Acid or Gel batteries, these do not have a BMS. So you want to add a battery monitor like a BMV or the SmartShunt.

Louisvdw, 2021-08-12:
You might be intrested in this thread. That opening poster has both a BMV (the non "Smart" version of the SmartShunt. i.e. it has not bluetooth interface) and compares those with the BMS values. He also has other threads in the same line.
https://energytalk.co.za/t/here-hold-my-beer-have-you-tried-this-before/964/9

liufuyang, 2021-08-12:
Aha, interesting, I think your suggestion makes a lot of sense.  Then I will just focus on waiting my BMS to arrive and see if this driver can work with it, otherwise I can perhaps get a Victron Smartshun. Thank you very much for the detailed info :)

liufuyang, 2021-08-12:
Thanks again, and I will just close this for now 👍 

I think my underling question is - how difficult it could be if there is a new device which has RS485 output and we use it with this driver. I wonder how different each manufacturer is doing with their RS485 output. Hopefully in the future all these could be standardised. But this could be asked in some other thread so don't worry if you don't reply here.

closed 2021-08-12 by liufuyang