Louisvdw/dbus-serialbattery #36: Daly BMS (MGod branded) shows discharging state as charging?

liufuyang, 2021-08-19:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
Connect the MGod BMS via RS485 to USB cable to Raspberry Pi.
Logs would be like:
```
root@raspberrypi2:~# tail -f /data/log/dbus-serialbattery.ttyUSB1/current 
...
@40000000611ea5c536bf82b4 INFO:__main__:dbus-serialbattery v0.5
@40000000611ea5c536d05364 INFO:__main__:Testing Daly
@40000000611ea5c53947fe6c INFO:utils:DalyBMS 10 cells
@40000000611ea5c6017fcfdc INFO:__main__:Battery connected to dbus from /dev/ttyUSB1
@40000000611ea5d014d71734 ERROR:utils:>>> ERROR: No reply - returning
@40000000611ea5d6185c9474 ERROR:utils:>>> ERROR: No reply - returning
@40000000611ea61219cdbc5c ERROR:utils:>>> ERROR: No reply - returning
@40000000611ea62818ad583c ERROR:utils:>>> ERROR: No reply - returning
@40000000611ea64c23e6161c ERROR:utils:>>> ERROR: No reply - returning
@40000000611ea67a24c2e18c ERROR:utils:>>> ERROR: No reply - returning
...
```

Then when discharge the battery, I saw this on the VRM website:
[image]
Which shows charging state. 

Also from the control UI seeing the amp and power is not negative perhaps that means charging as well?
[image]

One more thing I should mention is perhaps the Amp reading, which in this 2.4 to 2.6A, but with my other physical meters it reads around 2.1A. Not sure why there is some difference and what could be causing it.

Another thing is what I am not sure about those  `ERROR:utils:>>> ERROR: No reply - returning` output means. Perhaps that means something is not working correctly?

**Expected behavior**
It should show power usage when the current is going out?

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version v2.72

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: [Daly branded with MGod, LTO version]
 - Cells: [10]
 - Interface: [USB-RS485]

**Additional context**
I have asked the seller why MGod BMS looks very similar with Daly BMS, he told me they are corporative companies. I suppose the Daly driver can work out of the box but perhaps there is a few things I need to tweak for the MGod BMS to work?

As I can see the driver is already output some of the key values such as voltage and max,min cells, SOC, so I guess now it should be simple to fix it?

Thank you very much

liufuyang, 2021-08-19:
I tried to charge it now and then I saw the readings then become 0. 
[image]
[image]

In reality it is charged with 0.5A.

Perhaps there is some way I can dump all the info from the BMS so we can easily fix this? Thank you :)

liufuyang, 2021-08-20:
It seems that I can hack the line `self.current = (current - self.CURRENT_ZERO_CONSTANT) / -10` to divide to positive 10 to allow it to show discharging. But I guess this won't change the effect that when charging there is no reading? 🤔

Louisvdw, 2021-08-20:
Most BMS are not good at reading small currents accurately (<1A normally is an issue)

The Daly works with a constant and when the current is more or less than that it will show a charge or discharge. 

If you can charge it with more power it would show the charge/discharge better. It could be that the BMS brand you have is using another constant, but I would not expect that. They normally only put another sticher on the same hardware.

liufuyang, 2021-08-20:
Interesting. I will have to see that later when the sun is up :)  I also asked seller to send me the protocol manual and it seems looks the same as the Daly one mentioned here https://github.com/Louisvdw/dbus-serialbattery/issues/3 ?
[Mgod 485 协议中英文.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/7019608/Mgod.485.pdf)

liufuyang, 2021-08-20:
From this repo it seems that the current is just divide by 10, but in our code base it is divided by -10. Perhaps in our code it is wrong? I wonder how other Daly users see about this?
https://github.com/dreadnought/python-daly-bms/blob/main/dalybms/daly_bms.py#L147

Louisvdw, 2021-10-31:
In the next release you can change the utils.py to invert the current measurement
`INVERT_CURRENT_MEASUREMENT = -1`

closed 2021-10-31 by Louisvdw