Louisvdw/dbus-serialbattery #39: It would be nice to have QUCC BMS support?

liufuyang, 2021-09-01:
**Is your feature request related to a problem? Please describe.**

As you might know it or not, I had bought 2 Ant BMS and they both didn't start on my LTO nor LFP bettery packs. 
Then I tried a Daly 300A LTO BMS on my LTO battery site with your plug in it works nicely with Venus OS but the Daly BMS I got having two problems: 1. There is no amp reading when current is less than 2A, which makes the SOC easily go off after a few days for my application. 2. The SOC often (like yesterday) caps and stucks at a value (33.3% last time) - which seems a firmwire bug to me.

Then I finally got my [2 QUCC BMS](https://www.aliexpress.com/item/1005002608874846.html) arrived yesterday and they work very nicely, build quality is great and nice service as well. And their SOC estimation seems much better than the Daly ones. The package came with an usb storage key which contians phone app, pc software, detailed doc and serial communication protocols. I think some of their product is having the same protocal as the Jaibaida one but other product, like the one I got, have a different one. I think it is a really good product and company, the only down side I can think of right now is they don't have a good marketing outside China and they App's English UI needs more polishing for sure.

I will attach their protocal doc here as well.
[QUCC BMS-ModbusRTU Communication ProtocolV1.6-User Edition.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/7089699/QUCC.BMS-ModbusRTU.Communication.ProtocolV1.6-User.Edition.pdf)
(Above one is for 485 I suppose, but I also put CAN doc here just in case)
[Qucc CAN communication datasheet-V2.0.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/7089706/Qucc.CAN.communication.datasheet-V2.0.pdf)

**Describe the solution you'd like**
Support our plugin (just for monitoring or reading parameters, for my use case I don't need to nor want to do any settings on the BMS via our plugin) with the QUCC BMS protocal

**Describe alternatives you've considered**
Not sure what alternatives there could be...

**Additional context**
I natually assume you would be busy to support all different BMSs but if there could be some guide or help, I could perhaps give it a try and see if I could come up with some simple status readings from the QUCC BMS using python. I think the most important values for me to know is SOC, current, voltage, min/max cell voltage (and temperature perhaps). So to achieve this minimum requirement it shouldn't be too hard as we already have a protocol doc? 

I also have the usb to 485 adaptor bought together with this bms and it has that ch340 chip on it and can work with their PC softwares. So I am able to test any python or raspberry pi code to read BMS status.

Or if you do have time and want to have a QUCC BMS on your side to tese, perhaps I could let the seller know about it and we can ship you one with a low cost (I will cover the cost for you). Please let me know what your application requirement is, how many series and what is the max current limit you need.

Thank you :)

Louisvdw, 2021-09-02:
I read through the protocol for this BMS and there are a few items that are unique to this BMS. If you are not very comfortable reading and interpreting hex bytes, summs, crc, etc. it might be faster if I can get my hands on an example bms. 
The CH340 is a popular usb serial chip, but Venus OS does not provide support for it out of the box. I did make it work before on an Pi Zero, so it should work, but I might need to play and see how we can incorporate that into the autoinstaller. Venus OS do include support for the FTDI chipset (the Victron cables use those chips)

If you do want to send me an BMS, send me [a PM](https://energytalk.co.za/u/louisvdw) on one of the forums like https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot  and we can arrange that. It only needs to use the same protocol, but can be a smaller model.

liufuyang, 2021-09-03:
Thanks a lot, I will see if I could fine time later to take a look at your forums. I am not usually working with hex or crc but these days I tried to write some simple code in Rust to fetch some data in my QUCC board, for example reading all the cell voltages and it indeed works. I used some crc16 Rust package that someone already made. Seems working nicely.
https://github.com/liufuyang/usb-1/blob/master/src/qucc/mod.rs

Though I haven't done this much before in Python but I guess perhaps it is not that hard?

Also about CH340, I think it already supported? As my Daly BMS apparently have a CH340. I needed to install the CH340 driver on windows so to use the PC software to read and write settings on it. And I basically just plugged it in the Raspberry Pi with your plugin installed and it indeed can read status.  Besicdes this small issue https://github.com/Louisvdw/dbus-serialbattery/issues/36 where I need to edit the python code to fix the sign of the current it reports.

So maybe I could give it a try when I have time?

liufuyang, 2021-09-04:
BTW, I sent you a message on https://diysolarforum.com/ :)