Louisvdw/dbus-serialbattery #43: JKBMS: Sometimes wrong readings

trixing, 2021-09-10:
**Describe the bug**
I sometimes get wrong readings from the jkbms code. I already added code to test the checksum and the various framing bits (command type, etc), but e.g. the SoC (offset 0x85) sometimes goes to 133%, while battery voltage stays in the correct range (offset 0x83).

The captured status_data in this case is
`
Invalid soc: [bytearray(b"\x01y0\x01\r\'\x02\r,\x03\r+\x04\r\'\x05\r-\x06\r(\x07\r,\x08\r+\t\r2\n\r\'\x0b\r-\x0c\r(\r\r+\x0e\r-\x0f\r\'\x10\r,\x80\x00\x1d\x81\x00\x18\x82\x00\x18\x83\x15\x10\x84\x85\xc3\x85M\x86\x02\x87\x00\x03\x89\x00\x00\x03\x00\x8a\x00\x10\x8b\x00\x00\x8c\x00\x07\x8e\x16\x80\x8f\x10@\x90\x0e\x10\x91\rH\x92\x00\x05\x93\n(\x94\x0b\xb8\x95\x00\x05\x96\x01,\x97\x00d\x98\x01,\x99\x002\x9a\x00\x1e\x9b\x05\xdc\x9c\x00\x03\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x00F\xa4\x00F\xa5\xff\xec\xa6\xff\xf6\xa7\xff\xec\xa8\xff\xf6\xa9\x10\xaa\x00\x00\x01\x18\xab\x01\xac\x01\xad\x04\x0f\xae\x01\xaf\x01\xb0\x00\n\xb1\x14\xb2123456\x00\x00\x00\x00\xb3\x00\xb4Input Us\xb52101\xb6\x00\x00\xd6\x00\xb7H6.X__S6.1.0S__\xb8\x00\xb9\x00\x00\x01\x00\xbaBT307202012000")]
`

A correct case right next to the wrong frame is like this:
`
[bytearray(b"\x01y0\x01\r\'\x02\r,\x03\r+\x04\r\'\x05\r-\x06\r(\x07\r,\x08\r+\t\r8\n\r,\x0b\r+\x0c\r(\r\r-\x0e\r\'\x0f\r,\x10\r+\x80\x00\x1d\x81\x00\x18\x82\x00\x18\x83\x15\x11\x84\x86\x01\x85M\x86\x02\x87\x00\x03\x89\x00\x00\x03\x00\x8a\x00\x10\x8b\x00\x00\x8c\x00\x07\x8e\x16\x80\x8f\x10@\x90\x0e\x10\x91\rH\x92\x00\x05\x93\n(\x94\x0b\xb8\x95\x00\x05\x96\x01,\x97\x00d\x98\x01,\x99\x002\x9a\x00\x1e\x9b\x05\xdc\x9c\x00\x03\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x00F\xa4\x00F\xa5\xff\xec\xa6\xff\xf6\xa7\xff\xec\xa8\xff\xf6\xa9\x10\xaa\x00\x00\x01\x18\xab\x01\xac\x01\xad\x04\x0f\xae\x01\xaf\x01\xb0\x00\n\xb1\x14\xb2123456\x00\x00\x00\x00\xb3\x00\xb4Input Us\xb52101\xb6\x00\x00\xd6\x00\xb7H6.X__S6.1.0S__\xb8\x00\xb9\x00\x00\x01\x00\xbaBT307202012000")]
`

In this particular case (it's unfortunately not the only ones), the following bytes are flipped (some might be due to different values):
`
Offset 0x1d Wrong 0x32 OK 0x38
Offset 0x20 Wrong 0x27 OK 0x2c
Offset 0x23 Wrong 0x2d OK 0x2b
Offset 0x29 Wrong 0x2b OK 0x2d
Offset 0x2c Wrong 0x2d OK 0x27
Offset 0x2f Wrong 0x27 OK 0x2c
Offset 0x32 Wrong 0x2c OK 0x2b
Offset 0x3e Wrong 0x10 OK 0x11
Offset 0x40 Wrong 0x85 OK 0x86
Offset 0x41 Wrong 0xc3 OK 0x01
`

What I believe is that the find() function in get_data is brittle as it doesn't differentiate between value and idcodes. I think it would be better to parse the bytes continuously, ie. first read the ID, then read the number of bytes associated to it, then continue, etc. I'm happy to write up a patch if you agree with the approach.

ps:
Modified code to test for checksum and print framing bits (in read_serial_data_jkbms):

`
        start, length = unpack_from('>HH', data)
        terminal = unpack_from('>L', data[4:])[0]
        cmd, src, tt = unpack_from('>bbb', data[8:])
        frame = unpack_from('>H', data[-9:])[0]
        end, crc_hi, crc_lo = unpack_from('>BHH', data[-5:])

        s = sum(data[0:-4])
        logger.info('S%d C%d C%d' % (s, crc_lo, crc_hi))
        logger.info('T%d C%d S%d F%d TT%d' % (terminal, cmd, src, frame, tt))

        if s != crc_lo:
            logger.error('CRC checksum mismatch: Expected 0x%04x, Got 0x%04x' % (crc_lo, s))
`

Louisvdw, 2021-09-10:
Hi @trixing 
Thank you for the detailed feedback. You might be correct about the find. I've been seeing some issues as well every now and then.
You are welcome to look at a patch and create a pull request, else I will look at it as soon as I get some time.

trixing, 2021-09-10:
I already got something rudimentary working and confirming my theory. I can
clean this up over the weekend and send a patch.

Louisvdw, 2021-09-10:
An easy way to still check for the identification byte and fix any posible false finds is to give it a starting position for the find. That was my original plan, but I did not get to implement that.

Louisvdw, 2021-09-10:
So the get_data function could look like this
```
    def get_data(self, bytes, idcode, start, length):
        start = bytes.find(idcode, start, start + 1 + length)
        if start < 0: return False
        return bytes[start+1:start+length+1]
```
and then called by passing the offset as the start

trixing, 2021-09-10:
What I thought is a bit simpler:

1. read the first byte of the response (that tells you the number of bytes for the cell voltages)2
2. . handle the cell voltages
3. pass the rest of the response to a giant unpack_from as the field order in the response seems always the same

Or

3b. instead of a really long line of code for unpack_from which is hard to read, loop through the bytes, skip forward by known length of field.

I've quickly implemented proof of concept of 3b as it's easier to read and allows for more error checking:

<pre>
wrong = bytearray(b"\x01y0\x01\r\'\x02\r,\x03\r+\x04\r\'\x05\r-\x06\r(\x07\r,\x08\r+\t\r2\n\r\'\x0b\r-\x0c\r(\r\r+\x0e\r-\x0f\r\'\x10\r,\x80\x00\x1d\x81\x00\x18\x82\x00\x18\x83\x15\x10\x84\x85\xc3\x85M\x86\x02\x87\x00\x03\x89\x00\x00\x03\x00\x8a\x00\x10\x8b\x00\x00\x8c\x00\x07\x8e\x16\x80\x8f\x10@\x90\x0e\x10\x91\rH\x92\x00\x05\x93\n(\x94\x0b\xb8\x95\x00\x05\x96\x01,\x97\x00d\x98\x01,\x99\x002\x9a\x00\x1e\x9b\x05\xdc\x9c\x00\x03\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x00F\xa4\x00F\xa5\xff\xec\xa6\xff\xf6\xa7\xff\xec\xa8\xff\xf6\xa9\x10\xaa\x00\x00\x01\x18\xab\x01\xac\x01\xad\x04\x0f\xae\x01\xaf\x01\xb0\x00\n\xb1\x14\xb2123456\x00\x00\x00\x00\xb3\x00\xb4Input Us\xb52101\xb6\x00\x00\xd6\x00\xb7H6.X__S6.1.0S__\xb8\x00\xb9\x00\x00\x01\x00\xbaBT307202012000")
ok = bytearray(b"\x01y0\x01\r\'\x02\r,\x03\r+\x04\r\'\x05\r-\x06\r(\x07\r,\x08\r+\t\r8\n\r,\x0b\r+\x0c\r(\r\r-\x0e\r\'\x0f\r,\x10\r+\x80\x00\x1d\x81\x00\x18\x82\x00\x18\x83\x15\x11\x84\x86\x01\x85M\x86\x02\x87\x00\x03\x89\x00\x00\x03\x00\x8a\x00\x10\x8b\x00\x00\x8c\x00\x07\x8e\x16\x80\x8f\x10@\x90\x0e\x10\x91\rH\x92\x00\x05\x93\n(\x94\x0b\xb8\x95\x00\x05\x96\x01,\x97\x00d\x98\x01,\x99\x002\x9a\x00\x1e\x9b\x05\xdc\x9c\x00\x03\x9d\x01\x9e\x00Z\x9f\x00F\xa0\x00d\xa1\x00d\xa2\x00\x14\xa3\x00F\xa4\x00F\xa5\xff\xec\xa6\xff\xf6\xa7\xff\xec\xa8\xff\xf6\xa9\x10\xaa\x00\x00\x01\x18\xab\x01\xac\x01\xad\x04\x0f\xae\x01\xaf\x01\xb0\x00\n\xb1\x14\xb2123456\x00\x00\x00\x00\xb3\x00\xb4Input Us\xb52101\xb6\x00\x00\xd6\x00\xb7H6.X__S6.1.0S__\xb8\x00\xb9\x00\x00\x01\x00\xbaBT307202012000")

from struct import unpack_from

MAP={
        0x79: ('B', 1000.0, 'cell_voltages', 1), # magic
        0x80: ('H', 1.0, 'temperature_internal', 2),
        0x81: ('H', 1.0, 'temperature_box', 2),
        0x82: ('H', 1.0, 'temperature_battery', 2),
        0x83: ('H', 1000.0, 'battery_voltage', 2),
        0x84: ('H', 100.0, 'current', 2),
        0x85: ('b', 1.0, 'soc', 1),
        0x86: ('b', 1.0, 'temperature_sensor_count', 1),
        0x87: ('H', 1.0, 'cycles', 2),
        0x89: ('L', 1.0, 'total_capacity', 4),
        0x8a: ('H', 1.0, 'strings', 2),
        0x8b: ('H', 1.0, 'protection', 2),
# TODO names
        0x8c: ('H', 1.0, 'status', 2),
        0x8e: ('H', 1.0, 'status', 2),
        0x8f: ('H', 1.0, 'status', 2),
        0x90: ('H', 1.0, 'status', 2),
        0x91: ('H', 1.0, 'status', 2),
        0x92: ('H', 1.0, 'status', 2),
        0x93: ('H', 1.0, 'status', 2),
        0x94: ('H', 1.0, 'status', 2),
        0x95: ('H', 1.0, 'status', 2),
        0x96: ('H', 1.0, 'status', 2),
        0x97: ('H', 1.0, 'status', 2),
        0x98: ('H', 1.0, 'status', 2),
        0x99: ('H', 1.0, 'status', 2),
        0x9a: ('H', 1.0, 'status', 2),
        0x9b: ('H', 1.0, 'status', 2),
        0x9c: ('H', 1.0, 'status', 2),
        0x9d: ('b', 1.0, 'status', 1),
        0x9e: ('H', 1.0, 'status', 2),
        0x9f: ('H', 1.0, 'status', 2),
        0xa0: ('H', 1.0, 'status', 2),
        0xa1: ('H', 1.0, 'status', 2),
        0xa2: ('H', 1.0, 'status', 2),
        0xa3: ('H', 1.0, 'status', 2),
        0xa4: ('H', 1.0, 'status', 2),
        0xa5: ('H', 1.0, 'status', 2),
        0xa6: ('H', 1.0, 'status', 2),
        0xa7: ('H', 1.0, 'status', 2),
        0xa8: ('H', 1.0, 'status', 2),
        0xa9: ('b', 1.0, 'status', 1),
        0xaa: ('L', 1.0, 'status', 4),
        0xab: ('b', 1.0, 'status', 1),
        0xac: ('b', 1.0, 'status', 1),
        0xad: ('H', 1.0, 'status', 2),
        0xae: ('b', 1.0, 'status', 1),
        0xaf: ('b', 1.0, 'status', 1),
        0xb0: ('H', 1.0, 'status', 2),
        0xb1: ('b', 1.0, 'status', 1),
        0xb2: ('10s', 1.0, 'status', 10),
[... 36 lines cut ...]
            if x == 0x79 and expected == 1:
                expected = unpack_from('>B', chunk)[0]
                chunk = bytearray()
            if len(chunk) == expected:
                res = None
                if x == 0x79:
                    cells = []
                    for i in range(len(chunk)/3):
                        p = chunk[(3*i):(3*i+3)]
                        n, v = unpack_from('>BH', p)
                        # print(i, MAP[x][0], p, len(p), n, v)
                        cells.append((n , v))
                    res = cells
                else:
                    res = unpack_from('>' + MAP[x][0], chunk)

                print('Complete %02x' % x, MAP[x], res)
                x = None
                chunk = bytearray()
                expected = None

parser(ok[1:])
parser(wrong[1:])

</pre>

Louisvdw, 2021-09-10:
I have reopened the jkbms branch and created a fix using the offset method that seems to be working well. You can check it out.

trixing, 2021-09-10:
I think you could just remove the find() now given that you are hard coding
essentially the offset.

But yeah, this looks like it will work, though it obviously makes the code
a bit less readable.

trixing, 2021-09-11:
I assume you are going to merge that branch as a quick fix, given my
approach would be a lot more intrusive?

Louisvdw, 2021-09-12:
> I assume you are going to merge that branch as a quick fix, given my approach would be a lot more intrusive?

Correct, that is the idea.

PR #54 (2021-10-02): Bugfixes0.9
Fix #53 - midpoint function error on <4 cell battery
Fix #50 - Minimum cell voltage returns False for Sinowealth BMS
Fix #43 - JKBMS add checksum from @trixing and check the byte size of the lenght field to read
Fix #48 - Add option to log cell voltages - need to set log level to INFO or lower to activate
* resolve deptrecated warnings in python 3
* default logging level set to WARNING
* Add scripts to run for installing, upgrading and disabling the driver
closed 2021-10-02 by Louisvdw