Louisvdw/dbus-serialbattery #46: ElectroDacus SBMS0 integration

Louisvdw, 2021-09-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

willobrien, 2022-10-16:
+1 on this. IMO, what's needed is a gateway to ship BMS values across canbus.
On my camper build, just some form of integration would be great. On my boat, everything must speak canbus: BMS, Cerbo GX, WS500 alternator controller...  I'm using a different BMS on the boat, but am getting tempted to install a SBMS0 for monitoring/exposing the data a bit more. Whatever comes from this, will be useful to me. Happy to help.

H0lb0X, 2022-12-18:
Any news about the SBMS integration?