Louisvdw/dbus-serialbattery #47: Single Temperatur BMS issue

Iolinar, 2021-09-15:
Hey there,

I am using a JBP-SP04S010 via UART.
In a former release I could manage to make it work, I had to change the parsing of the BMS status message. 
As far as i could follow the answer contains the count of temp sensors and depending on that count the length of the answer differs.
Since my BMS has only one temp sense and the code expects at leased two it crashes when reading the RAW data and reports an array length missmatch.

In the most recent version many things changed and were definetly improoved, but my old trick does not work out any more. (simply removed any occurence of temp sense two in code or gave a static value)

Any chance to get a single temp sensor running?
I can provide any data that could be usefull to solve the issue!

best regards

Iolinar

Louisvdw, 2021-09-15:
Hi @Iolinar 
Can you run the attached script on your device and give me the output. You will first need to change the serial port on line 57 to the one your device is connected too. And then run it with:
`python serialtestJBD.py`
[serialtestJBD.zip](https://github.com/Louisvdw/dbus-serialbattery/files/7172355/serialtestJBD.zip)

Iolinar, 2021-09-19:
Hey,

there is an argument missmatch in that line:

Traceback (most recent call last):
  File "serialtestJBD.py", line 74, in <module>
    read_gen_data()
  File "serialtestJBD.py", line 57, in read_gen_data
    res = read_serial_data(commandGenData, '/dev/ttyUSB1', 9600, 3, 6)
TypeError: read_serial_data() takes exactly 6 arguments (5 given)

Added length_fixed, with True and False, in both cases it stated:

**/dev/ttyUSB1
serial data toread 6**
Traceback (most recent call last):
  File "serialtestJBD.py", line 74, in <module>
    read_gen_data()
  File "serialtestJBD.py", line 57, in read_gen_data
    res = read_serial_data(commandGenData, '/dev/ttyUSB1', 9600, 3,6, False)
  File "serialtestJBD.py", line 30, in read_serial_data
    print(res.hex())
AttributeError: 'str' object has no attribute 'hex'

Louisvdw, 2021-09-19:
That parameter should be False for the JBD.
It seems the hex() function is not available, so here is a version modified to print out with the hexlify function. Give it a try and send me the result.
[serialtestJBD.zip](https://github.com/Louisvdw/dbus-serialbattery/files/7192534/serialtestJBD.zip)

Louisvdw, 2021-09-28:
@Iolinar did you had a chance to run this last script?

commit e6f0b732 (2021-10-04): Bugfixes0.9 (#59)

Fix #47 JBD single temp sensor
Fix deviation calc
closed 2021-10-04 by Louisvdw