Louisvdw/dbus-serialbattery #48: single cell values in logfile / debugfile

dakoal, 2021-09-17:
HI.

I am always logging BMS data of the different BMS's I am using.
The single cell voltages are also logged, so I can see in a graph over time if one cell is faster dropping/rising.

Is there a way to get the (even the raw hex values) out of your wonderful piece of code into a /var/log/serial-starter/debug.txt ?

Thank you for your work.

Louisvdw, 2021-09-17:
Hi @dakoal 
There is no such logging currently, but you can add it yourself. Logging to a log file tend to be slow and we poll the BMS data for most batteries every second so I try to keep it as light as possible.

For the battery that you are using, it's driver file will have a refresh_data(self) function. Create your own log function that you can call at the end of this function.

```
def log_data(self):
   # this will print to screen
   for c in range(self.cell_count):
       print ("Cell {0} voltage    {1}V".format(c, self.cells[c].voltage)
```

dakoal, 2021-09-17:
Ok, I will try. It's a daly BMS. I will see if I can handle this in memory without file operation.

Louisvdw, 2021-09-17:
> It's a daly BMS. 
I hope you have the Sinowealth Daly version. It has the data for each cell. The normal Daly we only use the min/max cell data that the BMS publish. 
In the log file it will either say "Daly/Sinowealth BMS" or "Daly BMS" connected.

Carstijn, 2021-09-17:
If its a Daly with Sinowealth chip then it should already print the cell voltages to the log.

dakoal, 2021-09-17:
> If its a Daly with Sinowealth chip then it should already print the cell voltages to the log.

This one:  /data/log/dbus-serialbattery.ttyUSB0/current ?
There is nothing logged, only if an error occurs.

Louisvdw, 2021-09-19:
That is the correct log file for the USB0. You can chech which USB port your battery use in Remote Console uder your battery -> Device -> Connection

dakoal, 2021-09-20:
> If its a Daly with Sinowealth chip then it should already print the cell voltages to the log.

I think it's not. No single cell values there. But with a selfmade script I can check them.

Carstijn, 2021-09-20:
Here is an example from my log:

root@beaglebone:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current 
@40000000614859681c9f27f4 INFO:utils:>>> INFO: current pack current: 0.000000
@40000000614859681d9c391c INFO:utils:>>> INFO: Cell 1 voltage: 3.334000 V
@40000000614859681e94ae94 INFO:utils:>>> INFO: Cell 2 voltage: 3.318000 V
@40000000614859681f9e2784 INFO:utils:>>> INFO: Cell 3 voltage: 3.334000 V
@400000006148596821386adc INFO:utils:>>> INFO: Cell 4 voltage: 3.335000 V

If the info stuff is not in there you might not have a Sinowealth BMS? What is in your log?

Louisvdw, 2021-09-29:
@dakoal the latest beta has cell logging as an option. You need to edit the utuls.py line 10 and change the loggin level from WARNING to INFO to activate it. 
`logger.setLevel(logging.INFO)`
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.9beta1
https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install

PS @SanderV4n if you want to keep the same level of logging for your Sinowealth you need to do the same.

SanderV4n, 2021-09-30:
@Louisvdw I was looking at your changes, they look good! The logging was mostly for development, so its fine to have it at a lower level.

commit f6845d05 (2021-09-29): Add optional cell logging

fix #48 Cell logging
PR #54 (2021-10-02): Bugfixes0.9
Fix #53 - midpoint function error on <4 cell battery
Fix #50 - Minimum cell voltage returns False for Sinowealth BMS
Fix #43 - JKBMS add checksum from @trixing and check the byte size of the lenght field to read
Fix #48 - Add option to log cell voltages - need to set log level to INFO or lower to activate
* resolve deptrecated warnings in python 3
* default logging level set to WARNING
* Add scripts to run for installing, upgrading and disabling the driver
closed 2021-10-02 by Louisvdw