Louisvdw/dbus-serialbattery #49: DalyBMS 8cell: wrong value min Volt

riegler-econsulting, 2021-09-20:
The values on a Raspberry Pi v2.73 with (since?) v0.8beta1 are not the same like the PC Windows App with the same USB-UART cable. (with v0.8 the BMS is not running)

Between the two screenshots are only some seconds apart.

Min/max difference is according v0.8beta1 since days about 0.008V - and according PC-App/Android App 0.003 V.
The min Volt seems to be too low.

[image]

[image]

riegler-econsulting, 2021-09-20:
additional info:
[image]

Louisvdw, 2021-09-21:
There are two things I can think of that can influence this on the Daly.

The Daly use seperate commands for each value you want, but the baud rate (speed) is actually low for the overhead. So while for most BMS we retrieve data every second, for the Daly we can only update every 2 seconds or else the data from the previous request has not finished yet. So this 2 second delay could be influencing the values that you see.

The other thing is that the Daly has a command to return just the min/max cell voltages (so not all the cells) and this is what we currently use in the driver. I think this is most likely the reason for the difference as I don't know how fast the min/max values are updated against the cell values inside the BMS. 

I'll look at making a test version with the individual cell voltages at that you can check against.

dakoal, 2021-10-01:
I am waiting also for the individual cell voltages.

closed 2022-06-16 by Louisvdw