Louisvdw/dbus-serialbattery #50: Minimum cell voltage path on D-Bus has invalid value (False)

mpvader, 2021-09-24:
**Describe the bug**
Minimum cell voltage on D-Bus has value False. Seen on the system of de Longis which ahs a Daly 4S BMS, running the sinoweath protocol (?).

This causes issues in Venus OS and also in the VRM data ingestion system.

Please don't only fix that path, but make sure that also other paths cannot do this.

Its seen only sometimes, so maybe this is during startup, or some other exceptional situation.

The data as sent to VRM is:
```
    ["mcV"]=>
    array(1) {
      [1]=>
      string(5) "False"
    }
```

And the error is:
```
InfluxException: HTTP code: 500. {"error":"write failed: partial write: field type conflict: input field \"value\" on measurement \"173\" is type boolean, already exists as type float dropped=1"}
```

**To Reproduce**
I don't know

**Expected behavior**
Either publish a valid numeric value, typically a float. Or a dbus-invalid value, (let me know if you don't know what that is), or don't publish the path at all.

**Screenshots**

**VenusOS (please complete the following information):**
- RaspberryPi, Venus OS v2.66.

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly 4S
 - Cells: 4
 - Interface: I don't know, its not my system.

**Additional context**
I'll ask the user of the system to add more information to this issue if necessary. I'm from Victron, and just posting this to reduce the amount or errors and warnings seen in our monitoring system.

mpvader, 2021-09-24:
Ps. looking at the data, perhaps whats being sent here is supposed to be a what we call "D-Bus invalid". But just a slightly wrong one.

See here the Python definition of dbus-invalid: https://github.com/victronenergy/velib_python/blob/master/ve_utils.py#L12.

Which is an empty array. Not an array with a boolean or string in it.

unifiedcommsguy, 2021-09-24:
Thankyou @mpvader for reporting this for me.

@Louisvdw I am able to assist with the permission of Victron once the code has been validated.

Louisvdw, 2021-09-27:
Thanks for reporting. I will see if I can find where the issue might be or let you know if I need more information.

Louisvdw, 2021-09-29:
@unifiedcommsguy can you please test using the new build
https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.9beta1
https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install

PR #54 (2021-10-02): Bugfixes0.9
Fix #53 - midpoint function error on <4 cell battery
Fix #50 - Minimum cell voltage returns False for Sinowealth BMS
Fix #43 - JKBMS add checksum from @trixing and check the byte size of the lenght field to read
Fix #48 - Add option to log cell voltages - need to set log level to INFO or lower to activate
* resolve deptrecated warnings in python 3
* default logging level set to WARNING
* Add scripts to run for installing, upgrading and disabling the driver
closed 2021-10-02 by Louisvdw