Louisvdw/dbus-serialbattery #52: Please add support for TinyBMS

gyrex, 2021-09-27:
I love your project! Could you please add support for TinyBMS: https://www.energusps.com/shop/product/tiny-bms-s516-150a-750a-36?category=4

There's a communication protocol PDF here too: https://www.energusps.com/web/binary/saveas?filename_field=datas_fname&field=datas&model=ir.attachment&id=21208

Many thanks!! :)

tenteb, 2022-10-17:
Good morning, thank you very much for your excellent work.
I have a victron system with multiplus and cerbo gx and 4 mercedes batteries in parallel on a bus bar and each one with a Tiny BMS. You could contact me and give me a price to be able to integrate these bms. I'm waiting. Greetings and thanks.

amarce, 2022-12-13:
Hi,

I woud love to see your driver working with Tiny BMS too. Can we hel you somehow in order to make it possible?

Thanks a lot!