Louisvdw/dbus-serialbattery #53: Daly bms connects en disconnects every 5 seconds

Barpo, 2021-09-29:
Daly bms connects en disconnects every 5 to 10 seconds.

tail -f /data/log/serial-starter/current also shows the start of the dbus-serialbattery at the same time.

I use  Venus Version 2.66 on my Raspberry Pi
Daly 3Cell for testing, interface USB-RS485 and the USB-RS232 from daly or the FTDI: all the same problem...

Louisvdw, 2021-09-29:
@Barpo can you please look at the dbus-serialbattery logs if you see anything in there that will help
See https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot

Barpo, 2021-09-29:
The serial communication:
Pi   A5 40 94 08 00 00 00 00 00 00 00 00 81
Daly A5 01 94 08 03 01 00 00 02 00 00 02 4A 

Pi   A5 40 90 08 00 00 00 00 00 00 00 00 7D
Daly A5 01 90 08 00 72 00 00 75 30 01 EF 45

Pi   A5 40 91 08 00 00 00 00 00 00 00 00 7E
Daly A5 01 91 08 0F 01 02 0E E1 01 01 EF 31

Pi   A5 40 92 08 00 00 00 00 00 00 00 00 7F
Daly A5 01 92 08 40 01 40 01 DF 01 01 EF 92

Pi   A5 40 93 08 00 00 00 00 00 00 00 00 80
Daly A5 01 93 08 00 01 01 2B 00 00 3A 02 AA

Barpo, 2021-09-29:
Serial port of Pi Sends all kind of requests...
FE FF FF FD FD
01 46 07 53 A2                                           
47 E2 42 47 77 FF 47 E2 42 47 77 FF 47 E2 42 47 77 FF 47 E2 42 23 77 FF 
47 E2 42 47 77 FF 47 E2 42 47 77 FF 47 E2 42 47 77 FF 47 E2 42 47 77 FF 
A7 E2 22 47 77 FF 47 E2 42 47 77 FF 47 E2 42 47 77 FF 47 E2 42 47 77 FF 
41 80                                                                   
A5 40 94 08 00 00 00 00 00 00 00 00 81                                  
A5 40 90 08 00 00 00 00 00 00 00 00 7D
A5 40 91 08 00 00 00 00 00 00 00 00 7E
A5 40 92 08 00 00 00 00 00 00 00 00 7F

Barpo, 2021-09-29:
Part of the error log:

@4000000061549b5011230814     midpoint, deviation = self.battery.get_midvoltage()
@4000000061549b501123db04   File "/data/etc/dbus-serialbattery/battery.py", line 190, in get_midvoltage
@4000000061549b50113017d4     half1voltage += self.cells[c].voltage
@4000000061549b50113207ec IndexError: list index out of range
@4000000061549b6020bfdb94 INFO:__main__:dbus-serialbattery v0.8
@4000000061549b6020d5bd24 INFO:__main__:Testing LltJbd
@4000000061549b6030b65c1c ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b6030cef4fc INFO:__main__:Testing Ant
@4000000061549b61050f670c ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b610524e6f4 INFO:utils:>>> ERROR: Incorrect Data
@4000000061549b61052d5eec INFO:__main__:Testing Daly
@4000000061549b611507f304 ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b61151e9fb4 INFO:__main__:Testing Daly
@4000000061549b6124f7302c ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b61250c6dac INFO:__main__:Testing Jkbms
@4000000061549b6134e5f824 ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b6134fa8dac INFO:__main__:Testing Sinowealth
@4000000061549b62093bcec4 ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b62272b0404 INFO:__main__:Testing LltJbd
@4000000061549b6236fdd7bc ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b623714a794 INFO:__main__:Testing Ant
@4000000061549b630b623904 ERROR:utils:>>> ERROR: No reply - returning
@4000000061549b630b791c64 INFO:utils:>>> ERROR: Incorrect Data
@4000000061549b630b8205a4 INFO:__main__:Testing Daly
@4000000061549b6313fcad24 INFO:utils:DalyBMS 3 cells
@4000000061549b6314054c2c INFO:__main__:Connection established to Daly
@4000000061549b63174ed9ac INFO:__main__:Battery connected to dbus from /dev/ttyUSB0
@4000000061549b652ff4b1fc Traceback (most recent call last):
@4000000061549b65302821f4   File "/data/etc/dbus-serialbattery/dbushelper.py", line 140, in publish_battery
@4000000061549b6530284134     self.publish_dbus()
@4000000061549b6530284cec   File "/data/etc/dbus-serialbattery/dbushelper.py", line 155, in publish_dbus
@4000000061549b6530285c8c     midpoint, deviation = self.battery.get_midvoltage()
@4000000061549b6530286c2c   File "/data/etc/dbus-serialbattery/battery.py", line 190, in get_midvoltage
@4000000061549b6530287fb4     half1voltage += self.cells[c].voltage
@4000000061549b653029374c IndexError: list index out of range

Barpo, 2021-09-29:
Super cool!!! the Midpoint fix #53 did the job :)

closed 2021-09-29 by Barpo

Louisvdw, 2021-09-29:
Great. Thanks for the feedback.

neoneddy, 2021-12-01:
I have same thing.  Venus OS 2.72  serial battery version 0.9.1 

Except of my log.

root@raspberrypi2:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current
@4000000061a6d5ff0e357d44     data = read_serial_data(self.generate_command(command), self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK)
@4000000061a6d5ff0e361d6c   File "/data/etc/dbus-serialbattery/utils.py", line 40, in read_serial_data
@4000000061a6d5ff0e3b0f0c     with serial.Serial(port, baudrate=baud, timeout=0.1) as ser:
@4000000061a6d5ff0e3ba764   File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
@4000000061a6d5ff0e4fd374     self.open()
@4000000061a6d5ff0e508b0c   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 288, in open
@4000000061a6d5ff0e69cbe4     self._update_rts_state()
@4000000061a6d5ff0e6a7bac   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 627, in _update_rts_state
@4000000061a6d5ff0e6c2574     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
@4000000061a6d5ff0e6dc76c IOError: [Errno 71] Protocol error
@4000000061a6d6100b964154 WARNING:SerialBattery:dbus-serialbattery v0.9.1
@4000000061a6d6100bb0ab0c WARNING:SerialBattery:Testing LltJbd
@4000000061a6d6101bf1483c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d6101bff5db4 WARNING:SerialBattery:Testing Ant
@4000000061a6d6102c612714 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d6102c71d0b4 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000061a6d6102c797d8c WARNING:SerialBattery:Testing Daly
@4000000061a6d6102d9c336c Traceback (most recent call last):
@4000000061a6d6102d9c4adc   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 108, in <module>
@4000000061a6d6102dce72b4     main()
@4000000061a6d6102dd47d94   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 78, in main
@4000000061a6d6102dd49504     battery = get_battery_type(port)
@4000000061a6d6102dd4a0bc   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 57, in get_battery_type
@4000000061a6d6102dd4b82c     if test.test_connection() is True:
@4000000061a6d6102dd4c3e4   File "/data/etc/dbus-serialbattery/daly.py", line 38, in test_connection
@4000000061a6d6102dd661f4     return self.read_status_data()
@4000000061a6d6102dd66dac   File "/data/etc/dbus-serialbattery/daly.py", line 54, in read_status_data
@4000000061a6d6102dd730fc     status_data = self.read_serial_data_daly(self.command_status)
@4000000061a6d6102dd74484   File "/data/etc/dbus-serialbattery/daly.py", line 125, in read_serial_data_daly
@4000000061a6d6102dd94ff4     data = read_serial_data(self.generate_command(command), self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK)
@4000000061a6d6102dda1344   File "/data/etc/dbus-serialbattery/utils.py", line 40, in read_serial_data
@4000000061a6d6102ddac30c     with serial.Serial(port, baudrate=baud, timeout=0.1) as ser:
@4000000061a6d6102ddb6b04   File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
@4000000061a6d6102ddf762c     self.open()
@4000000061a6d6102de029dc   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
@4000000061a6d6102de4e8b4     self._update_dtr_state()
@4000000061a6d6102de5a81c   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
@4000000061a6d6102df0e31c     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
@4000000061a6d6102df196cc IOError: [Errno 71] Protocol error
@4000000061a6d6213132b424 WARNING:SerialBattery:dbus-serialbattery v0.9.1
@4000000061a6d62131a91ca4 WARNING:SerialBattery:Testing LltJbd
@4000000061a6d62206ae8c64 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d62206bf16c4 WARNING:SerialBattery:Testing Ant
@4000000061a6d62217145534 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d6221722821c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000061a6d622172a5604 WARNING:SerialBattery:Testing Daly
@4000000061a6d62219f2fb84 WARNING:SerialBattery:Connection established to Daly
@4000000061a6d6221cf644bc WARNING:SerialBattery:Battery connected to dbus from /dev/ttyUSB0
@4000000061a6d6241e61bf84 ERROR:SerialBattery:[Errno 5] could not open port /dev/ttyUSB0: [Errno 5] Input/output error: '/dev/ttyUSB0'
@4000000061a6d6241e67bac4 Traceback (most recent call last):
@4000000061a6d6241e690ab4   File "/data/etc/dbus-serialbattery/dbushelper.py", line 159, in publish_battery
@4000000061a6d6241e76ba9c     self.publish_dbus()
@4000000061a6d6241e777a04   File "/data/etc/dbus-serialbattery/dbushelper.py", line 168, in publish_dbus
@4000000061a6d6241e791814     self._dbusservice['/Soc'] = round(self.battery.soc, 2)
@4000000061a6d6241e7a9eb4 TypeError: a float is required
@4000000061a6d63513c7c244 WARNING:SerialBattery:dbus-serialbattery v0.9.1
@4000000061a6d63513dedc54 WARNING:SerialBattery:Testing LltJbd
@4000000061a6d63524799004 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d635248b3b74 WARNING:SerialBattery:Testing Ant
@4000000061a6d635347ecdfc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d635348ce75c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000061a6d635349449fc WARNING:SerialBattery:Testing Daly
@4000000061a6d635370c9cfc WARNING:SerialBattery:Connection established to Daly
@4000000061a6d63539ae7cb4 WARNING:SerialBattery:Battery connected to dbus from /dev/ttyUSB0
[... cut ...]
@4000000061a6d63e0877425c   File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
@4000000061a6d63e088d3b5c     self.open()
@4000000061a6d63e088e48e4   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
@4000000061a6d63e08a7d7dc     self._update_dtr_state()
@4000000061a6d63e08a89f14   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
@4000000061a6d63e08aa6434     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
@4000000061a6d63e08ac0dfc IOError: [Errno 71] Protocol error

neoneddy, 2021-12-01:
I have same thing.  Venus OS 2.72  serial battery version 0.9.1 

Except of my log.

```
root@raspberrypi2:~# tail -f /data/log/dbus-serialbattery.ttyUSB0/current
@4000000061a6d5ff0e357d44     data = read_serial_data(self.generate_command(command), self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK)
@4000000061a6d5ff0e361d6c   File "/data/etc/dbus-serialbattery/utils.py", line 40, in read_serial_data
@4000000061a6d5ff0e3b0f0c     with serial.Serial(port, baudrate=baud, timeout=0.1) as ser:
@4000000061a6d5ff0e3ba764   File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
@4000000061a6d5ff0e4fd374     self.open()
@4000000061a6d5ff0e508b0c   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 288, in open
@4000000061a6d5ff0e69cbe4     self._update_rts_state()
@4000000061a6d5ff0e6a7bac   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 627, in _update_rts_state
@4000000061a6d5ff0e6c2574     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
@4000000061a6d5ff0e6dc76c IOError: [Errno 71] Protocol error
@4000000061a6d6100b964154 WARNING:SerialBattery:dbus-serialbattery v0.9.1
@4000000061a6d6100bb0ab0c WARNING:SerialBattery:Testing LltJbd
@4000000061a6d6101bf1483c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d6101bff5db4 WARNING:SerialBattery:Testing Ant
@4000000061a6d6102c612714 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d6102c71d0b4 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000061a6d6102c797d8c WARNING:SerialBattery:Testing Daly
@4000000061a6d6102d9c336c Traceback (most recent call last):
@4000000061a6d6102d9c4adc   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 108, in <module>
@4000000061a6d6102dce72b4     main()
@4000000061a6d6102dd47d94   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 78, in main
@4000000061a6d6102dd49504     battery = get_battery_type(port)
@4000000061a6d6102dd4a0bc   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 57, in get_battery_type
@4000000061a6d6102dd4b82c     if test.test_connection() is True:
@4000000061a6d6102dd4c3e4   File "/data/etc/dbus-serialbattery/daly.py", line 38, in test_connection
@4000000061a6d6102dd661f4     return self.read_status_data()
@4000000061a6d6102dd66dac   File "/data/etc/dbus-serialbattery/daly.py", line 54, in read_status_data
@4000000061a6d6102dd730fc     status_data = self.read_serial_data_daly(self.command_status)
@4000000061a6d6102dd74484   File "/data/etc/dbus-serialbattery/daly.py", line 125, in read_serial_data_daly
@4000000061a6d6102dd94ff4     data = read_serial_data(self.generate_command(command), self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK)
@4000000061a6d6102dda1344   File "/data/etc/dbus-serialbattery/utils.py", line 40, in read_serial_data
@4000000061a6d6102ddac30c     with serial.Serial(port, baudrate=baud, timeout=0.1) as ser:
@4000000061a6d6102ddb6b04   File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
@4000000061a6d6102ddf762c     self.open()
@4000000061a6d6102de029dc   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
@4000000061a6d6102de4e8b4     self._update_dtr_state()
@4000000061a6d6102de5a81c   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
@4000000061a6d6102df0e31c     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
@4000000061a6d6102df196cc IOError: [Errno 71] Protocol error
@4000000061a6d6213132b424 WARNING:SerialBattery:dbus-serialbattery v0.9.1
@4000000061a6d62131a91ca4 WARNING:SerialBattery:Testing LltJbd
@4000000061a6d62206ae8c64 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d62206bf16c4 WARNING:SerialBattery:Testing Ant
@4000000061a6d62217145534 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d6221722821c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000061a6d622172a5604 WARNING:SerialBattery:Testing Daly
@4000000061a6d62219f2fb84 WARNING:SerialBattery:Connection established to Daly
@4000000061a6d6221cf644bc WARNING:SerialBattery:Battery connected to dbus from /dev/ttyUSB0
@4000000061a6d6241e61bf84 ERROR:SerialBattery:[Errno 5] could not open port /dev/ttyUSB0: [Errno 5] Input/output error: '/dev/ttyUSB0'
@4000000061a6d6241e67bac4 Traceback (most recent call last):
@4000000061a6d6241e690ab4   File "/data/etc/dbus-serialbattery/dbushelper.py", line 159, in publish_battery
@4000000061a6d6241e76ba9c     self.publish_dbus()
@4000000061a6d6241e777a04   File "/data/etc/dbus-serialbattery/dbushelper.py", line 168, in publish_dbus
@4000000061a6d6241e791814     self._dbusservice['/Soc'] = round(self.battery.soc, 2)
@4000000061a6d6241e7a9eb4 TypeError: a float is required
@4000000061a6d63513c7c244 WARNING:SerialBattery:dbus-serialbattery v0.9.1
@4000000061a6d63513dedc54 WARNING:SerialBattery:Testing LltJbd
@4000000061a6d63524799004 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d635248b3b74 WARNING:SerialBattery:Testing Ant
@4000000061a6d635347ecdfc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000061a6d635348ce75c ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000061a6d635349449fc WARNING:SerialBattery:Testing Daly
@4000000061a6d635370c9cfc WARNING:SerialBattery:Connection established to Daly
@4000000061a6d63539ae7cb4 WARNING:SerialBattery:Battery connected to dbus from /dev/ttyUSB0
[... cut ...]
@4000000061a6d63e0877425c   File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
@4000000061a6d63e088d3b5c     self.open()
@4000000061a6d63e088e48e4   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
@4000000061a6d63e08a7d7dc     self._update_dtr_state()
@4000000061a6d63e08a89f14   File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
@4000000061a6d63e08aa6434     fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
@4000000061a6d63e08ac0dfc IOError: [Errno 71] Protocol error
```

neoneddy, 2021-12-03:
Ok, odd.  Left it connected for a few days.  Seems stable now.

I also installed this via USB for another friend and it all worked great out of the box, but was a 4s vs mine is an 8s .  Not sure this needs to be opened back up.

Louisvdw, 2021-12-06:
@neoneddy you error seem to be from the underlying serial library, which indicates it might be a low level error. 
My first though would be that is could be a wire/plug that is loose and that is hamperring signals. The wire might settle over time (gravity :) ). If you move the cable a bit and it start giving errors that would be a clear indication.

neoneddy, 2021-12-06:
@Louisvdw rebooted after updating to venus OS 2.73 , Started reconnecting in a loop again, been going on for a day or two now.

I also run GUI Mods if that matters.   I can try and reseat the UART cable.   I am also connecting 4 dongles via a USB hub, I could could try connecting directly, I ran a 10ft USB extension cable to add this 5th device.  I have BMV, 2x Solar, Multiplus, and now the Daly.  Maybe putting Daly on it's own USB line might help.

Thanks for response.

Louisvdw, 2021-12-06:
@neoneddy those are all good things to try.  Check that the error in the log file is the **IOError: [Errno 71] Protocol error**. (should be the last line of a loop). 
I also use a 4port USB hub (MPPTx2, MK3, BMSx2) on my Pi3 testing hub so in itself that would not be an issue, but long cables and extra things could lead to cable loss. 
If you can run a test with the BMS connected to the GX directly you can see if that then works or not, and go from there

neoneddy, 2021-12-07:
Ok, just switched the Daly BMS to its own USB port on the Pi and it’s online now and has a stable connection.

Louisvdw, 2021-12-07:
I found according to Wikipedia the USB specification allows for 5m cable for USB1.1 and USB2 and 3m cable for USB3. You might need to add a booster or powered hub to get longer distances.
https://en.wikipedia.org/wiki/USB_hardware#Cabling

commit fcd80fe9 (2021-09-29): Midpoint fix #53

Midpoint fix #53
PR #54 (2021-10-02): Bugfixes0.9
Fix #53 - midpoint function error on <4 cell battery
Fix #50 - Minimum cell voltage returns False for Sinowealth BMS
Fix #43 - JKBMS add checksum from @trixing and check the byte size of the lenght field to read
Fix #48 - Add option to log cell voltages - need to set log level to INFO or lower to activate
* resolve deptrecated warnings in python 3
* default logging level set to WARNING
* Add scripts to run for installing, upgrading and disabling the driver
commit 91704571 (2021-10-02): Bugfixes0.9 (#54)

* Fix exception if the length byte is longer than 1 like in JKBMS and not enough bytes have been read to read the length of the reply.
* New disable and install scripts added
* logging cleanup
* set default logging to level Warning
* Sinowealth fixes to how cell values are initiated
* Fix deprecate warning in python3
* Add optional cell logging
* Add jkbms checksum from @trixing
* Midpoint fix #53
* fix Sinowealth error for no Capacity
* min/max cell volt improvement