Louisvdw/dbus-serialbattery #58: /Io/AllowToCharge unexpected behavior on daly bms

Barpo, 2021-10-03:
AllowToCharge  is depending on battery.charge_fet AND control_allow_charge.  This sound strange becauls the fet is The parameter enabaling charging.
Becauls my AllowToCharge gives on an off at (seems) random moment i change the code a little:

        self._dbusservice['/History/TotalAhDrawn'] = self.battery.total_ah_drawn
        _self._dbusservice['/Io/AllowToCharge'] = 1 if self.battery.charge_fet else 0_
#        self._dbusservice['/Io/AllowToCharge'] = 1 if self.battery.charge_fet \
#                                and self.battery.control_allow_charge else 0
        self._dbusservice['/Io/AllowToDischarge'] = 1 if self.battery.discharge_fet else 0
        self._dbusservice['/System/NrOfModulesBlockingCharge'] = 0 if self.battery.charge_fet is None or \
                                (self.battery.charge_fet and self.battery.control_allow_charge) else 1
        self._dbusservice['/System/NrOfModulesBlockingDischarge'] = 0 if self.battery.discharge_fet is None \

Louisvdw, 2021-10-04:
**control_allow_charge** is managed by the [Charge Current Control Management](https://github.com/Louisvdw/dbus-serialbattery/wiki/Features) feature of the driver. It will stop charging the battery once it has reached 100% SOC.

Barpo, 2021-10-04:
Oke, I see it's managed. I changed it to my preference to fet status of the mbs. Still good to see the max charge capacity.
Thx

closed 2021-10-05 by Louisvdw