Louisvdw/dbus-serialbattery #67: Integrate with kwindrem/SetupHelper project to maintain updates and installation using Console GUI

ddarek2000, 2021-11-05:
It would be great to have a single update mechanism for all custom packages installed on Venus OS.

Since I already use SetupHelper project for their GUI changes , I would like to incorporate the setup and upgrade with their project. (console based update button)

It is cumbersome to ssh to the remote box once disconnected from home network and check for new version of the project.

SetupHelper project is written in SHELL and will require a standard version file indicating the installed and new version (hope to have this automatically updated during build / publish).  There are additional requirements during the installation.  The Feature request on kwindrem repository is included.

[https://github.com/kwindrem/SetupHelper/issues/3](https://github.com/kwindrem/SetupHelper/issues/3)

pau1phi11ips, 2022-01-17:
This is out of beta now btw https://github.com/kwindrem/SetupHelper

mr-manuel, 2023-05-05:
I think this is only useful, if all settings can also be changed through the remote console/GUI, since all settings has to be done through the `config.ini`. I would also prefer to keep the `config.ini` since it can be copied over to same systems and reduces the installation time. In my opinion with this much settings a GUI would also be overwhelming the user.

Surely this could be an advance for remote systems without SSH access to update the driver, but I would not do blindly an update, if I have no other access to the device.

mr-manuel, 2023-05-05:
I think it's less complicated to add a update function to this driver than integrating it with SetupHelper. Like a page with three options:

1. Install latest release
2. Install nightly
3. Install penultimate version

What do you think?

mr-manuel, 2023-05-12:
As alternative for updating over the remote console you can use this in the meanwhile: [venus-os_ngrok](https://github.com/mr-manuel/venus-os_ngrok)

It‘s installable with the package manager and allows you to access the SSH port without forwarding the port.

mr-manuel, 2023-11-25:
This is not compatible with this project.

closed 2023-11-25 by mr-manuel