Louisvdw/dbus-serialbattery #68: Show all Cells within the battery on console detail screen

ddarek2000, 2021-11-05:
My current BMS can either be connected to BT or Serial connection to VenusOS.  I get to see all cell voltages in BMS , but would like to have the BMS sending the low and high cell to the VenusOS to control charge / discharge of the battery.

I would like to display all 16 cells in the detail, and have this accessible using the Console.

I have decided to return my current BMS and replace with one that supports dual connection.  This functionality would still be very useful for single connection devices.

Louisvdw, 2021-11-05:
What benefit do you see in having all the cell voltages available vs. just the min/max cell voltages that VenusOS currently support?
I know it will be nice to see. I am just trying to guage priority and if this is only a nice to have or has some benifit not currently available elsewhere.

ddarek2000, 2021-11-05:
The detail about batteries don't have concrete benefits.  I find it is easier to see if the min/max when compared to the average.  It is easier for me to interpret the state of each battery cell that way.  Even with lower priority , I might take this one up on myself to complete.

Louisvdw, 2021-11-09:
The [Chargery BMS driver ](https://github.com/Tobi177/venus-chargerybms) has support to display all the cells. The QML will need to be changed to add to the display (see gui\qml). 
In Victron land anything that is None will not be displayed, and this would be preferred as the driver support batteries with few to many cells (3-32 mostly). 
Also if we bring in the extra battery banks then this will complicate the display more. Something to keep in mind.

neoneddy, 2021-11-19:
+1 .. one of the things I look a the most in the BMS app right now (about to actually connect this to Venus soon) is individual cell voltages.   it let's you know if a particular cell is going rogue, or is it just .05v deviation from the rest.

With the QML, are you saying if someone wanted that data, just add the pointers / references in the QML and it will work?

Louisvdw, 2021-11-22:
@neoneddy , The VenusOS only contain the min/max cell voltages. If you have a rogue cell you can see it as either the min or the max will have larger differenaces from the other. This is also there the BMS min/max cell graph come in.

However, if we would want to show all the cell voltages, then these will need to be populated as non standard extra fields on VenusOS, and we also need to create/edit a view to display them with the QML.

So no, it will not work only adding QML.

[image]

neoneddy, 2021-12-01:
@Louisvdw Ok, makes sense. I think the min/max will work fine.

Jaco-TTT, 2021-12-08:
Showing the Delta between the cells, a be a very useful visual indicator of a problem.

piapiacz, 2022-02-03:
@Louisvdw, I had the same need as @ddarek2000, so I added this feature. If you are interested I can send you required changes.
Updated files are : battery.py (making available individual cell voltages + cell balancing), dbushelper.py (publishing cell voltages to /Voltages/ and balancing to /Balances/), PageBattery.qml (added submenu for Cell Voltages, if available) and new file PageBatteryCellVoltages.qml.
4 cells are visible per line. When the cell is balancing the background on the cell turns red.
[image]

neoneddy, 2022-02-03:
This is awesome!  I love the Red balancing background.

Jaco-TTT, 2022-02-03:
> @Louisvdw, I had the same need as @ddarek2000, so I added this feature. The pull request #87 is in queue. Updated files are : battery.py (making available individual cell voltages + cell balancing), dbushelper.py (publishing cell voltages to /Voltages/ and balancing to /Balances/), PageBattery.qml (added submenu for Cell Voltages, if available) and new file PageBatteryCellVoltages.qml. **4 cells are visible per line**. When the cell is balancing the background on the cell turns red. [image]

What if one uses 18 cells on 48v system? 
Is that possible, today being the exception, slowly becoming a viable option between 16/17 and 18 cells.

piapiacz, 2022-02-04:
@neoneddy thanks. I am just giving back to the community.
@Jaco-TTT the backend publishes 24 cells, so all you need to do is to edit the gui file - /opt/victronenergy/gui/qml/PageBatteryCellVoltages.qml

after the line : 
<pre>	property VBusItem _b16: VBusItem { bind: service.path("/Balances/Cell16") }</pre>
add :
<pre>	property VBusItem _b17: VBusItem { bind: service.path("/Balances/Cell17") }
	property VBusItem _b18: VBusItem { bind: service.path("/Balances/Cell18") }</pre>

after the line : 
<pre>	property string c16: _b16.valid && _b16.text == "1" ? "#ff0000" : "#ddd"</pre>
add :
<pre>	property string c17: _b17.valid && _b17.text == "1" ? "#ff0000" : "#ddd"
	property string c18: _b18.valid && _b18.text == "1" ? "#ff0000" : "#ddd"</pre>

and after the :
<pre>		MbItemRow {
			description: qsTr("Cells (13/14/15/16)")
			values: [
				MbTextBlock { item { bind: service.path("/Voltages/Cell13") } width: 70; height: 25; color: c13 },
				MbTextBlock { item { bind: service.path("/Voltages/Cell14") } width: 70; height: 25; color: c14 },
				MbTextBlock { item { bind: service.path("/Voltages/Cell15") } width: 70; height: 25; color: c15 },
				MbTextBlock { item { bind: service.path("/Voltages/Cell16") } width: 70; height: 25; color: c16 }
			]
		}</pre>
you need to add :
<pre>		MbItemRow {
			description: qsTr("Cells (17/18)")
			values: [
				MbTextBlock { item { bind: service.path("/Voltages/Cell17") } width: 70; height: 25; color: c17 },
				MbTextBlock { item { bind: service.path("/Voltages/Cell18") } width: 70; height: 25; color: c18 }
			]
		}</pre>
the file can be edited using for example this command : <pre>nano /opt/victronenergy/gui/qml/PageBatteryCellVoltages.qml</pre>
then you need to restart the gui using these commands :
<pre>svc -d /service/gui
svc -u /service/gui</pre>

Louisvdw, 2022-02-04:
Nice work @piapiacz 
Thanks for sharing your changes. I'll review the PR as soon as I can

piapiacz, 2022-02-04:
@Louisvdw ok, thanks.

closed 2022-02-06 by Louisvdw