Louisvdw/dbus-serialbattery #70: reverse reading charging / discharging on Daly BMS

nhybrow, 2021-11-09:
Hello,
On the Daly bms, when the battery is charging, the BMS displays discharging, and when the battery is discharging it displays charging. On venusOS and VRM the direction of the current is therefore reversed, the reading of information is therefore mistaken. Is there a way to reverse the direction in order to interpret the values as they are displayed on the BMV, so that the current direction is correct?
Thanks in advance,

nhybrow, 2021-11-09:
[image]

Louisvdw, 2021-11-09:
Hi @nhybrow 
There is already an solution for this that will be included in the next release.
See issue #65

closed 2021-11-09 by Louisvdw