Louisvdw/dbus-serialbattery #74: Collaboration on 123SmartBMS Integration

CodeChief, 2021-11-17:
The 123electric 123SmartBMS has a similar DBUS serial integration in early stages of development which I am currently trying to test:

https://github.com/123electric/smartbms-venus

But currently it appears unclear how the charge and discharge enable is controlled. Could somebody here please help by confirming what the "System status" screen should look like with your DBUS integration for other BMS systems. As displayed below the solar and BMS control state/ability appear to be disabled:

[image]

Can your solution/other BMS show all as "On" or is this normal and the charge/discharge is controlled via some other API, or only ever by physical cables?

Louisvdw, 2021-11-22:
Those settings does not look correct for if you want the BMS to control the GX. 
The easiest might be to add the 123 to this driver, then all the Victron integration is already done. I'll have a look at the code in the repo when I get some time.

Here is what it should look like.
[image]

AlbertronicDev, 2021-11-22:
The 123\SmartBMS Victron software is currently only for monitoring the BMS on the display and via VRM - we have not a full "BMS" implementation. That is probably why not every value is ON.
It is a nice idea to have full BMS control over the Victron, although we still prefer our BMS to have a physical power relay connected to turn off the chargers/inverters in case of a fault.

We are working on full BMS control as it would be a nice feature. I suppose DVCC - on the Cerbo/Venus - needs to be turned on for this.

Louisvdw, 2021-11-22:
Yes, DVCC needs to be on. 
The BMS still stay in control. The BMS just tells the GX what it needs and if the GX does not do it then it still needs to protect with it's relays/FETs and control circuits.

All the GX is in control off is how much current it can charge or draw from the battery and at what voltage this is allowed. And these parameters come from the driver/BMS.

CodeChief, 2021-11-22:
This sounds great. Yes, the BMS physical relay as last resort/critical disconnect is a general feature people should be encouraged to have, whether directly supported by the BMS or achieved via Victron included ports or compatible add-ons.

My current experience with the Victron settings and DVCC suggest that the inverter/charger assistant code will take-over many settings. In residential use the "ESS Assistant" appears mandatory (from Victron documentation and the fact the old "hub" assistants are marked "depreciated"). This introduces an additional layer of complexity.

I hope that the complete DVCC support will work but wonder if both projects might need a custom assistant, like a "DBUS Virtual Switch", to fully cover the minimum and maximum allowed charge/discharge conditions from all sources, i.e. solar charger, DC-DC converter (generator) or inverter/charger (multi/quattro). Error messages and tooltips for greyed out settings I've seen in various places confirm the assistant becomes the "master" once configured.

Louisvdw, 2021-11-22:
The ESS Assistant do have to be included for a grid tie setup and it does run the GX and all the linked Victron devices. But it does this with the guidelines the BMS gives. 

So what my driver does is read the BMS to get the cell info. It will use the cell count to set the Charge Voltage Limit(CVL) and Discharge Current Limit(DCL). This tells the GX to charge the battery till you get to the CVL and to stop draining the battery below DCL. If it does go past these limits then the BMS protections will kick in, but the GX wil now try and stay in these limits.

Futher it will also limit the current draw to what the driver tells it in the Charge Current Limit(CCL)

Now the hole Victron ecosystem will use these limits. If you use MPPTs to change the battery in will limit the PV input to keep to those limits. So while the battery still accept charge the MPPTs will go at full speed, and as soon as the battery is full it will ramp down and the MPPTs will only generate power for the consumption.
Here is an image of my system doing that. On the marked time the PV is being limited as the battery reached full capacity.
[image]
If there are any other sources it will just follow what the GX will allow them to be able to keep with what the battery requests. 

The "Distributed" in DVCC meand it comes from the battery and not the GX settings (something else (the battery) is setting it).

If the system is off grid, then it will have [these features](https://www.victronenergy.com/media/pg/CCGX/en/dvcc---distributed-voltage-and-current-control.html#UUID-57d64cc6-d59b-7827-1b2d-9f2c18360975)
While if it is grid tie with the ESS assistant it will have [these features](https://www.victronenergy.com/media/pg/CCGX/en/dvcc---distributed-voltage-and-current-control.html#UUID-ab5f5267-b5d7-d9c2-40f1-8dde56821752)