Louisvdw/dbus-serialbattery #76: VRM firmware update Scanning VE.Direct devices

pahorpet, 2021-12-21:
Hi,
during a scanning - firmware update via VRM, I always get an error during Status: Scanning VE.Direct devices.
And the battery is then disconnected -> have to make a restart to connect to it again. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Vrm
2. Click on the firmware update 
3. See an error

**Expected behavior**
The driver should be able to handle that situation to reconnect again. And there has to be an answer to be able to update another device.

**VenusOS (please complete the following information):**
 - Device type: Cerbo
 - Firmware Version 2.80

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: ANT
 - Cells: 16
 - Interface: USB-RS232

pau1phi11ips, 2021-12-29:
What version of the driver are you on?

Just tried on mine (v0.9.0) and it has the same bug. There are some improvements for Venus 2.80 in v0.9.1 though
https://github.com/Louisvdw/dbus-serialbattery/releases

pahorpet, 2021-12-29:
V0.9.1.

Louisvdw, 2022-06-12:
In [beta4 release](https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.11beta4) I changed the way the driver is installed/reinstalled on the GX device. Please give it a test for this issue

squircle12, 2022-10-12:
Thanks Louis - I've just updated to 0.13 and I'm still experiencing this issue. 

Log into VRM > Navigate to Device list > Firmware Updates > error message reports "No VE.Direct ping reply" *screenshot attached* (note after about a minute the screen will still report devices with firmware updates, but the BMS software is non responsive and requires a reboot) 

VenusOS: 
Raspberry Pi 4 
VenusOS V2.90~28

Battery/BMS:
JKBMS 8 Cell USB to Serial
[image]

pau1phi11ips, 2022-10-13:
I'm on v12 beta 2 and Venus 2.9. The remote firmware update works ok here. I've just used it to update my MPPT.

squircle12, 2022-10-13:
> I'm on v12 beta 2 and Venus 2.9. The remote firmware update works ok here. I've just used it to update my MPPT.

Interesting - are you on a rPi at all? 

Some more specifics of my setup 
3 MPPT's connected via VE Direct (original cables) updated to latest firmware (150/100, 150/70, 150/35)
Multiplus 2 3000/24/70

pau1phi11ips, 2022-10-13:
I'm on a Cerbo GX @squircle12. It's in a campervan so having all the tank and temperature inputs was a no brainer.

Similar setup to you though:
1 MPPT via VE.Direct (100/30)
Multiplus 2 3000/24/70

squircle12, 2022-10-13:
Checking /data/log/vedirect.ttyUSB5/current log (this is the USB the battery is connected to, I noticed this - could be something, could be nothing, seems to occur when the scan occurs so It's bombing out and then restarting the service: 

`@400000006347d2771b4c49f4 -------- dumping backlog -------
@400000006347d2771b4c55ac 000.036 INF.task: using ttyUSB5 as unique service part
@400000006347d2771b4c6164 000.037 INF.VE.Direct: Setting status to VdUnknown
@400000006347d2771b4c6d1c 000.043 INF.settings: connected to settings on dbus (com.victronenergy.settings)
@400000006347d2771b4c7cbc 003.000 ERR.values: timeout connecting device
@400000006347d2771b4c848c 003.000 INF.task: done in 003.000`

coca7, 2022-10-13:
I checked my logs, and I'm seeing that too on all 3 of my USB connections.  I don't notice any other symptoms though.

@4000000063336fa43794c474 *** starting vedirect-interface ***
@4000000063336fa438ecedec vedirect-dbus 3.70
@4000000063336fa7394b80dc
@4000000063336fa7394ba01c -------- dumping backlog -------
@4000000063336fa7394ba7ec 000.029 INF.task: using ttyUSB0 as unique service part
@4000000063336fa7394bb3a4 000.029 INF.VE.Direct: Setting status to VdUnknown
@4000000063336fa7394bbf5c 000.042 INF.settings: connected to settings on dbus (>
@4000000063336fa7394bcb14 003.000 ERR.values: timeout connecting device
@4000000063336fa7394bd6cc 003.000 INF.task: done in 003.000
@4000000063336fd31030cc5c *** starting vedirect-interface ***
@4000000063336fd3118125d4 vedirect-dbus 3.70

Louisvdw, 2022-10-13:
It could be an issue on the Pi devices.

squircle12, 2022-10-13:
I have another install with a Pi 4 I'm setting up in the next couple of weeks - I'll try a vanilla build of Venus prior to installing third party stuff just to see if its a VRM / Pi thing and not related to serialbattery at all.

mr-manuel, 2023-05-04:
Is this issue still relevant? Unfortunately I have no VE.Direct device to test.

Please try with dbus-serialbattery `v0.14.3` or the nightly version and Venus OS `v2.92` or greater.

closed 2023-05-28 by mr-manuel