Louisvdw/dbus-serialbattery #77: Ant and USB rs485-et340 not compatible

pahorpet, 2022-01-01:
**Describe the bug**
When both devices(ant usb via ch340 and Original victron rs485 to read et340) are connected none of them is working.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect both device
2. Restart
3. None of them work

**Expected behavior**
Handle correct device

**VenusOS (please complete the following information):**
 - Device type: cerbo
 - Firmware Version 2.80

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: ant
 - Cells:  16
 - Interface:  USB-RS485, USB-ttl ch340

I am assuming that it is general error not only ant.

pau1phi11ips, 2022-02-18:
I've been having a bit of a headache getting an ET112 working. Since I have the USB RS485 going thru a ZigBee link I assumed it was that that was the problem. Took me a while to realise it was this blocking it.

Is there a way to lock your driver to a specific USB port @Louisvdw ?

pau1phi11ips, 2022-02-18:
@Louisvdw I'm on 2.80. I'm updating to 2.84 now.

Louisvdw, 2022-06-12:
There is a new [beta4 release](https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.11beta4) that should solve this issue. Please give it a test

Louisvdw, 2022-06-16:
release v0.12 fix this problem

closed 2022-06-16 by Louisvdw

pau1phi11ips, 2022-06-23:
I didn't spot this message but thought I'd give 0.12 a try anyway. Working well, thanks!