Louisvdw/dbus-serialbattery #78: running at nanopi

pahorpet, 2022-01-07:
As the code is running on several devices would it be possible to run it at nanopi?
multi/easysolar GX. 
It crash on :
file "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 595, in reset_input_buffer 
    termios.tcflush(self.fd, termios.TCIFLUSCH)
termios.error: (25, 'Inappropriate ioctl for device')

any hints to fix would be appreciated, also able to test.

Louisvdw, 2022-01-07:
It is suppose to work but seems that the IO control is getting an error. I suspect this would be a VenusOS issue. Or it might be the pyserial lib on that device where something is not supported.

bergyla, 2022-05-27:
My standard setup is a MultiplusII GX.
The integrated GX device calls itself nanopi.
At least one (of three connected DALY BMS could be seen at the same time without severe crash).
However I found some errors in the LOG which is unfimilar with yours and furthermore are not crashing the driver at all:
@40000000628f9dc02e73f5cc Traceback (most recent call last):
@40000000628f9dc02e7424ac   File "/data/etc/dbus-serialbattery/dbushelper.py", line 142, in publish_battery
@40000000628f9dc02e743064     success = self.battery.refresh_data()
@40000000628f9dc02e743c1c   File "/data/etc/dbus-serialbattery/daly.py", line 49, in refresh_data
@40000000628f9dc02e7447d4     result = result and self.read_fed_data()
@40000000628f9dc02e744fa4   File "/data/etc/dbus-serialbattery/daly.py", line 108, in read_fed_data
@40000000628f9dc02e745b5c     fed_data = self.read_serial_data_daly(self.command_fet)
@40000000628f9dc02e771a7c   File "/data/etc/dbus-serialbattery/daly.py", line 125, in read_serial_data_daly
@40000000628f9dc02e772e04     data = read_serial_data(self.generate_command(command), self.port, self.baud_rate, self.LENGTH_POS, self.LENGTH_CHECK)
@40000000628f9dc02e77418c   File "/data/etc/dbus-serialbattery/utils.py", line 61, in read_serial_data
@40000000628f9dc02e774d44     toread = ser.inWaiting()
@40000000628f9dc02e77512c   File "/usr/lib/python3.8/site-packages/serial/serialutil.py", line 590, in inWaiting
@40000000628f9dc02e77ddcc     return self.in_waiting
@40000000628f9dc02e77ed6c   File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 467, in in_waiting
@40000000628f9dc02e77f924     s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
@40000000628f9dc02e7804dc OSError: [Errno 5] Input/output error
@40000000628f9dc038ca71cc /opt/victronenergy/serial-starter/run-service.sh: line 15: kill: (2444) - No such process

So, i´m able to connect at least one DALY BMS via a isolated UART-USB Adapter with genuine FTDI device under Venus OS v2.85.
Are there additional USB-Serial devices connected via VE-Direct (USB-UART cable) ?
Could you figure out which TtyUSB devices are mapped to your BMS and to other functions (MK3 IF, internal CAN-IF etc,. and your connected USB-UARTs.
Furthermore there is a possibility of non-genius FTDI chips seen sometimes which seems to be less stable and tend to through errors via USB communication.

corneel27, 2022-09-30:
Also here no problems with nanopi on gx board in Multiplus ik. Venus version 2.90.
Perhaps van @pahorpet try it with version 2.90 of Venus os?

mr-manuel, 2023-02-10:
I have also no problems on my MultiPlus II-GX with Venus OS 2.92 and driver version 0.14.3

closed 2023-04-14 by mr-manuel