Louisvdw/dbus-serialbattery #83: Multiplus never goes into float

neoneddy, 2022-01-16:
**Describe the bug**
Multiplus never goes into float now.  I adjusted the battery cell voltages up to 3.55 to get them better top balanced.  But now the voltage never drops back down.  Is there anything that can be done or do I just set the cell voltages back manually.

**To Reproduce**
Steps to reproduce the behavior:
1. Change voltages to 3.55v
2. Charge with multi
3. wait
4. goto step 3

**Expected behavior**
Would be nice to eventually drop down to a float voltage of 3.4 or 3.45

Louisvdw, 2022-01-16:
Why would you want that?
Float if for lead acid batteries. Litiums don't use it as the BMS manage the cell voltages and the chemistry don't need a tricle charge.

Or is there any other reason?

neoneddy, 2022-01-16:
For the very reason I did it, to get the cells top balanced.   The variances in cell manufacturing and whatnot show themselves at higher voltages and state of charge.  The BMS  or Balancer can't do it's job unless there are differences in cell voltages.  At the same time I don't think it's good to hold the batteries at 3.55v for weeks or months on end.  Just my $.02

Attached below, you can see how they balanced out after a few cycles.  I've let them sit for a week now at 3.55 and now it's nearly in perfect balance.

[image]

bluetrepidation, 2022-10-11:
Default Victron behavior is to hit your absorption voltage (usually 3.45 volts per cell) then a 1 hour timer starts. Once 1 hour has elapsed the system then drops into "float" or maybe a better word is "resting" voltage. This set point is a much lower voltage per cell (usually 3.35) which causes no current to flow and allows the lithium chemistry to rest. They battery will be at 100% charge but the voltage will sag slightly. My understanding is that holding a voltage of the default of 3.45 for long periods of time will damage the cells. Most cell datasheets mention after the constant voltage charge stage to stop the charger after you reach 1 amp of current. A 1 hour timer basically accomplishes the same thing a different way.

Louisvdw, 2022-10-11:
Check out the Charge Voltage Control that was added to v0.13 release. 
This should give you the ability to do what you want. 
You can enable it in utils.py
> CVCM_ENABLE = False/True
And give us feedback on how it works.