Louisvdw/dbus-serialbattery #85: DALY BMS - Unreasonable CVL and DCL

Thomk-63, 2022-01-30:
**Describe the bug**
I am using a newer DALY BMS 4S12V, which can only communicate to the DALY PC Monitor, not anymore to Sinowealth.
In the parameter section of the remote console the following is displayed:
[image]

I am sure that the values for CVL and DCL are not correct. the CCL is correct!

**To Reproduce**
Just connect a newer DALY BMS model R05A, configure the CVL and DCL with the Bluetooth App or PC-Monitor.

**Expected behavior**
Display the correct values

**Screenshots**
[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version V2.73

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: DALY 4S 12V 100A model R05A
 - Cells: 4
 - Interface: USB-RS485

**Additional context**
none

Louisvdw, 2022-01-30:
Hi @Thomk-63 
CCL and DCL is not read from the BMS. There is already a [ticket open](https://github.com/Louisvdw/dbus-serialbattery/issues/4) to read those, but for now you need to set them yourself. Follow the guide [under install](https://github.com/Louisvdw/dbus-serialbattery/wiki/How-to-install#nb-before-you-begin) 

Then also the DCL (and CCL) will be lowered depending on the SOC % of the battery. See how the [CCCM feature](https://github.com/Louisvdw/dbus-serialbattery/wiki/Features) works and you will see where the 5A comes from on a 100% DOC.

So your model works as designed

Thomk-63, 2022-01-30:
Thanks for the fast reaction. I just looked to the respective links. I now better understand!
Is the CCCM feature automatically active and will it actively reduce the set allowed current in the DALY BMS? Or is this depending on an other setting in the RPi Venus? I did not recognize the effect, as I was discharging with more than 6A and was already below 10%SOC (just a first test, I would not do this regularly)

Thomk-63, 2022-01-30:
How can I easiest update the utils.py in the RPi?

JostB-ger, 2022-02-02:
> How can I easiest update the utils.py in the RPi?

Activate SuperUser in Console, create root password, activate SSH on Lan, login via ssh

For editing the utils.py file use a texteditor like nano:
`nano /data/etc/dbus-serialbattery/utils.py`

For editing the CCCM-Limits you have to edit the battery.py:
`nano /data/etc/dbus-serialbattery/battery.py`

For example I changed my Amp-Limit below 10% to 20A
```
        # Change depending on the SOC values
        if self.soc <= 10:
            self.control_discharge_current = 20
        elif 10 < self.soc <= 20:
            self.control_discharge_current = self.max_battery_discharge_current/4
        elif 20 < self.soc <= 30:
            self.control_discharge_current = self.max_battery_discharge_current/2
        else:
            self.control_discharge_current = self.max_battery_discharge_current

```

Louisvdw, 2022-02-04:
As long as you install the driver using one of the normal methods, you can then later edit the files inside the GX device using an text editor like what @JostB-ger described with nano, or you can edit the file using your normal PC (Windows, MacOS) and copy the new file over to the GX using a SFTP client (like Filezilla, Winscp, CyberDuck)

Louisvdw, 2022-02-04:
> Is the CCCM feature automatically active and will it actively reduce the set allowed current in the DALY BMS? Or is this depending on an other setting in the RPi Venus?

CCCM is active by default, but will only limit the charge current when your battery is close to full capacity. This allows for slower charge of the last bit without any cell going into an overvoltage state and the BMS not being able to balance it fast enough.
Similar, it only limit discharge when the battery is close to empty capacity and a large current draw can pull a cell below the low voltage protect limit.

If you want to disable CCCL completely you can comment out this code at line 156 in dbushelper.py
```
            # This is to mannage CCCL
            self.battery.manage_charge_current()
```

closed 2022-02-06 by Louisvdw