Louisvdw/dbus-serialbattery #86: DALY BMS: Reading Alarms and used Ah

Thomk-63, 2022-01-30:
**Is your feature request related to a problem? Please describe.**
The current driver is already really good! hank you for this and all the effort!
But there are 4 things missing:
1. Alarms/Errors of the DALY
2. Display of used Ah (possibly there is not such a value with the DALY?!)
3. Display of activated balancer
4. Display of installed capacity

[image]
[image]
[image]

**Describe the solution you'd like**
Display of the Alarms/Errors and the values

**Additional context**
DALY 4S 12V 100A model R05A
Raspberry Pi
V2.73

Louisvdw, 2022-02-06:
Pull request #90 now add support for alarms on Daly BMS

CaptKrisp, 2022-02-07:
@Thomk-63 The Daly docs do not specify how to pull configured capacity.
The current code now includes a BATTERY_CAPACITY variable in utils.py to specify your battery capacity.
Most Alarm data is now pulled from BMS.

Thomk-63, 2022-02-07:
Hi Louis,
which files need to be updated? all?

Regards,
Thomas
Thomas + Nicole Krüger
***@***.***

Thomk-63, 2022-02-07:
> Hi Thomas
> I'm still busy adding more fixes and these new changes have not all been tested together so there might be some errors.
> If you want to give it a test before the next betais released then update from the latest main branch the following files: battery.py, dbushelper.py, daly.py and utils.py
> If you find any issues report it and revert to the last release again.

Thanks, I will do and will report.

Regards,
Thomas
Thomas + Nicole Krüger
***@***.***

Thomk-63, 2022-02-11:
Hi Louis,

in the meantime I could test a little.
In general the software still works after your modifications. The capacity value of the battery can be seen in the Venus console.
But when I provoked a cell voltage high alarm (cell volt high level 2), no alarm was reported in the Venus console.

Here two screenshots at the same time of the alarm:

Regards,
Thomas

Thomas + Nicole Krüger
***@***.***

Louisvdw, 2022-06-16:
This is included in the new release v0.12
Note the cell voltages is disabled for now as it was giving issues

closed 2022-06-16 by Louisvdw