Louisvdw/dbus-serialbattery #95: Support for LiFePO4 48 V

stanhausc, 2022-02-18:
Hello, @Louisvdw, I'm a student at [PTI](https://www.pti.org.br), we have several renewable energy projects, one of them is the communication between the LiFePO4 48 V battery and the Venus OS.
Here's the battery communication protocol
[Protocolo de comunicação LFP - MODBUS_ rev.06.pdf](https://github.com/Louisvdw/dbus-serialbattery/files/8098726/Protocolo.de.comunicacao.LFP.-.MODBUS_.rev.06.pdf).
I'll work on it over the next few weeks, and if it goes well, I'll share the code.

Thanks for your great work.

emuman049, 2022-11-04:
Is the battery mentioned in the documentation a MasterBattery Upower UE-48Li100-3U? Any Progress so far?

stanhausc, 2022-11-07:
> 
This is not the battery (the pdf has the specifications), so far I'm able to read the voltage, current, SOC, alarms, number of cells and individual cell voltage data.

emuman049, 2022-11-09:
Okay I was Just thinking it is the same one, since 3U was mentioned in the PDF and Upower ist from Spain. What kind of BMS is this?

stanhausc, 2022-11-10:
> Ok, eu estava pensando que é o mesmo, já que o 3U foi mencionado no PDF e o Upower é da Espanha. Que tipo de BMS é esse?

The information I have about the BMS is:

[image]

cnidus, 2023-04-01:
+1 I have a stack do lifepower4 24v EG4 batteries I’d love to support with CerboGX.

drmiar, 2023-04-15:
May I ask? What Type of BMS is integrated in the batteries UE-48Li100-3U? Is there a possibility to make a firmware update? Software?  Thanks a Lot!