Louisvdw/dbus-serialbattery #98: dbus-serialbattery driver causes Smartshunt not recognised after Venus restart

smejfi, 2022-02-26:
**Describe the bug**
When this driver is installed into Venus OS and instead of serial BMS I connect Victron Smartshunt witch VE direct to USB cable, it is recognized correctly, but after Venus restart, it is not recognized at all or it is loaded as some of the BMS from Dbus-serialbattery driver database with nonsense values. A few times I have also experienced that the Bluesolar MPPT which in my case is connected to GX with Ve direct to Ve direct cable, wasnt recognized after Venus restart, but this happened only very rarely and I dont know how to reproduce it.

**To Reproduce**
Steps to reproduce the behavior:
1. Instal dbus-serialbattery
2. connect Victron Smartshunt
3. Restart Venus os
4. Smartshunt should not be recognized at all
5. If you disconnect the smartshunt and connect it again, it is recognized correctly.

**Expected behavior**
All devices should be recognized correctly after the venus os restart

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: VenusGX built in Multiplus II 48V/3000W
 - Firmware Version: 2.81

**Battery/BMS (please complete the following information):**
 - BMS/Battery type:
 - Cells:
 - Interface:

**Additional context**

hcaldicott, 2022-04-16:
I also have this issue. 

VenusOS on a Rpi4. Serialbattery running with a JBD BMS. Seems the serial battery library stops my SmartSolar 100/30 from recognising correctly and it appears I cannot have both connected at once. If I reboot the device with only one of the two devices connected I can reliably tell to that device. However connecting both generally results in only the BMS being recognised.

gmanic, 2022-04-16:
Same with venus os 2.82-large-30.

Removing the symlinks from /opt/victronenergy/(service(-template)?/)?dbus-serialbattery gets the smartshunt working again (but no JBD monitoring).

Might that be related to dbus-serialbattery just trying to get hold of any serial port, no matter what PRODUCT is connected?
My MK3 USB-Adapter always works just fine (connecting the Multiplus).

I'm not nearly close to fully understand serial-starter's behaviour and prioritization, but would it be possible(helpful to limit dbus-serialbattery to non VEdirect devices? Then the smart shunt and other VEdirect connected devices would work while non-VEdirect devices would be subject to scanning by dbus-serialbattery?

hcaldicott, 2022-04-16:
I actually use a multiport isolated UART (https://www.tindie.com/products/saimon/4-ports-isolated-usb-uart-converter-with-usb-c/) so for me all victron and JBD devices are on the same adapter. All ports appear as different serial devices though. 

Even the ability to temporarily configure serialbattery to only listen on a named tty device would be adequate - leaving all others to detect per victron standards.

smejfi, 2022-04-16:
Guys, I dont know if it helps to you, but for me the most of these issues were solved when I started using powered USB hub on my GX (even though I dont really need more than one USB device connected). Just try it and let me know if it helps.

Louisvdw, 2022-04-19:
The serial battery driver is set up to be tested last after all other drivers have failed. So something is making the correct driver to be disconnected and then the serial battery picks up that a result is returned. 
So it is possible that the powered hub is giving anough power so that those devices never disconnect and so keep their original driver connected.

luphi01, 2022-05-10:
Same issue here, Smartshunt and BMS (Daly) are directly connected via USB, no Hub.

Louisvdw, 2022-06-12:
There is a new [beta4 release](https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v0.11beta4) that should solve this issue. Please give it a test

closed 2022-06-16 by Louisvdw

smejfi, 2022-06-16:
> Komu: Louisvdw/dbus-serialbattery ***@***.***>
> Smartshunt not recognised after Venus restart (Issue #98)
> "
>
> This is fixed in the new release v0.12
>
> (https://github.com/Louisvdw/dbus-serialbattery/issues/98#issuecomment-1157384077)
> "

Thank you Louis! I will start testing new release as soon as possible!

Best regards,

Filip
---------- Původní e-mail ----------