Louisvdw/dbus-serialbattery #100: Create option to add custom name

Louisvdw, 2022-03-02:
Create option to add custom name.
This is useful if there are more than one BMS so you can name them to easily identify

pedrorampolla, 2022-06-15:
Any updates?   I’m willing to help in testing; I have 3 JBD BMS.

Louisvdw, 2022-10-04:
[image]
Should be in the next beta

mr-manuel, 2023-03-09:
I think before this can be addresses, the pairing of the `DeviceInstance` and the serial to USB adapter has to be overthinked. Actually there is no unique value from where a specific BMS can be identified, if multiple BMS are connected from the same make. If then the `DeviceInstance` changes because the ID of the serial to USB adapter changes, than the mess is complete.

Maybe a possible solution would be to assign a device ID by serial number of the BMS. This would require:
1. Extending all existing battery classes to read the serial number (biggest work)
2. Checking, if the `DeviceInstance` is already in use
3. Save the assigned pair of `DeviceInstance` and serial number somewhere, probably in the dbus
4. Delete an assigned pair, if the BMS is not seen for x days (if needed)

mr-manuel, 2023-04-14:
An easier way to identify batteries could be the battery capacity. It should not matter, if you have 10 batteries and change the battery by +/-5Ah to have an unique total capacity.

teefixx, 2023-04-15:
good idea
Tried this, but didn‘t work for daly, as, at least at my test, the capacity was not read out :-(

mr-manuel, 2023-04-15:
@teefixx thats true. Some Daly don't provide this value. Is there any other data we can use from daly to create a unique identifier?

teefixx, 2023-04-15:
Did not find one...
Anyway... thanks for your help. If I solve it I'll male a post for ither users, which have probably the same oroblem.

mr-manuel, 2023-05-03:
@teefixx with the latest version Daly also provides this value.

teefixx, 2023-05-03:
@mr-manuel got it already, installed your nightly ble branch.

thank you

transistorgit, 2023-05-04:
just read this again.
for my daly, there is a battery production date field that can be set by daly app. 
I added it to the hardware version string. So you can use different dates for your batteries to mark them. maybe better than adjust capacity?
[image]

mr-manuel, 2023-05-04:
Sure! Is this field empty by default?

transistorgit, 2023-05-04:
hm, cant remember...

transistorgit, 2023-05-04:
at least I set it myself to the date I built my battery... so it should be usable.
If someone wants to try: https://github.com/mr-manuel/venus-os_dbus-serialbattery/pull/5

mr-manuel, 2023-05-06:
It's now in the latest `jkbms_ble` branch. Custom BMS fields for Daly (battery code field) and JK BMS (User Private Data field) are shown on the "GUI -> SerialBattery -> Device" page as "Device Name".

seamaster101, 2023-05-09:
@mr-manuel I understand that you guys fixed that one, but I tested again after update to V 1.0.0-jkbms_ble (20230505)
[image]
still reverts to the default name after reboot
[image]
what could be causing that?

mr-manuel, 2023-05-09:
It's not fixed. But the value that you insert in "User Private Data field" in the JKBMS app is shown in the device page.

gurkc006, 2023-06-07:
Only to be shore not missing something: I updated to Version 1.0.20230531. The custom name still doesn't seem to be saved correct, is that right? After reboot I still see the default name. Any idea, when this will be fixed? Can I help with debugging? 
Thanks

mr-manuel, 2023-06-07:
Since this is not that easy as it seams, it will take time. There are a few options to set the name via Node-RED for example, so it will persist reboot and Venus OS update, but not if the VRM ID changes, which is currently the biggest issue.

mr-manuel, 2023-06-10:
Please try [this release](https://github.com/mr-manuel/venus-os_dbus-serialbattery/releases/tag/v1.0.20230610beta2) with all possible setups you have. The name currently is fixed to the port. If the port changes, the name disappears or the wrong one is loaded.

For this I added an unique identifier, which is displayed in the driver logs and in the remote console/GUI under Device -> Serial. Please check, if this value is different, if you have multiple BMS. If the value is not different, please increase/decrease the installed capacity on your BMS by 1 until all BMS have another capacity. In this way you can have 11 unique batteries with a deviation of +/- 5 Ah.

gurkc006, 2023-06-11:
Hi mr-manuel,
thanks a lot for the new beta version. I tested you new version (but I only have one BMS) and it works quite well. Assigned a new name, rebooted and name still ok. So for me that's fine now! Serial number says "Battery1_JK-B2A24S15P". If you need any other information, leave me a message. Thanks!!!

mr-manuel, 2023-06-11:
This is all I need, thanks

mr-manuel, 2023-06-29:
Is there no other feedback for this most wanted feature?

mr-manuel, 2023-07-31:
This is included in the latest beta and will be in the next stable release.

closed 2023-07-31 by mr-manuel

lfhoward, 2023-09-03:
Hello @mr-manuel,

Thanks for the nightly version. I completed the installation using (3) nightly and (2) dev. I then rebooted the system.

Unfortunately Venus cannot see the BMS anymore. It is not in the device list.

Should I have uninstalled the previous version before reinstalling the beta version over it? Or is there a setting I need to change once it boots? What am I missing?

Thanks

mr-manuel, 2023-09-04:
> Should I have uninstalled the previous version before reinstalling the beta version over it? Or is there a setting I need to change once it boots? What am I missing?

No, an uninstallation is not needed before updating.

Please check the issues, there you see that there are currently problems with the latest nightly and LLT/JBD BMS. Try the [latest beta](https://github.com/Louisvdw/dbus-serialbattery/releases/tag/v1.0.20230724beta).

PR #701 (2023-06-10): Signal strength, unique identifier, persist custom name, many improvements
* Added: Bluetooth: Show signal strenght of BMS in log by @mr-manuel
* Added: Create unique identifier, if not provided from BMS by @mr-manuel
* Added: JKBMS BLE - Show last five characters from the MAC address in the custom name (which is displayed in the device list) by @mr-manuel
* Added: Save custom name and make it restart persistant (fix #100) by @mr-manuel
* Changed: Fixed error in `reinstall-local.sh` script for Bluetooth installation by @mr-manuel
* Changed: Fixed typo in `config.ini` sample by @hoschult
* Chan [... cut]