Louisvdw/dbus-serialbattery #102: Cannot read property 'hasOwnProperty' of null" when using node-red

xury77, 2022-03-09:
**Describe the bug**
On battery monitor using node-red 
When no alarm property of dbus message in null but should be 0 
"TypeError: Cannot read property 'hasOwnProperty' of null"
**To Reproduce**
Steps to reproduce the behavior:
Connect to node-red use Baterry monitor node and choose any type of alarm
**Expected behavior**
Payload output should be 0 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**VenusOS (please complete the following information):**
 - Device type: Cerbo, 
 - Firmware Version 2.82 large

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly
 - Cells: [e.g. 16]
 - Interface:  USB-RS232

mr-manuel, 2023-04-15:
@xury77 does the issue still persist? In the meanwhile there were a lot of changes in the driver and Victron Node.

xury77, 2023-04-15:
I stopped using Daly BMS in favor of my own with your dbus  battery mqtt controller. So I don't know is still persist.

closed 2023-04-15 by mr-manuel