Louisvdw/dbus-serialbattery #116: Support for - SLPO48-200 Battery

NunoRgP, 2022-04-10:
Hello is it possible to add support to the  Sunstone SLPO48-200 Battery BMS communication ?

[https://www.sunstonepower.com/lithium-battery-with-bms.html](url)

I managed to find the following information on the communication protocol : 

[](https://www.vselektro.eu/files/edited/aku/slpo48-200-lifepo4-communication-data-sheet-ap.html)

A lot of people are buying these batteries , unfortunately for now they can only comunicate with sunstone inverters . 

Really nice work you did so far .

Louisvdw, 2022-04-10:
Yes, it looks like it would be possible to add.
Let's see if there are any intrest for people that would want it.

alwaysatronse9600, 2022-05-28:
Hello is it possible to add support to the Sunstone SLPO48-100 Battery BMS communication ?

hnestmann, 2022-09-24:
I am interested too. Happy to help as well

srux83, 2022-11-18:
Hi Louis,

I would like to contribute getting the Sunstonepower batteries to work with Victron using your driver. The SLPO48 100 and 200 have the same BMS.

Currently I use a Raspi to read some values from a RS Smart Solar and created some scripts to control the system and visualize everything with a munin plugin. 
My oversized project is working fine so far, but now I found your repo :)

Maybe we could get in touch to add support for the SLPO batteries.