Louisvdw/dbus-serialbattery #122: Ja bms triggers warning at Venus os

testpaul999, 2022-05-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 
Warning with high voltage appears at load end
2. Click on
Automatically  
3. Scroll down to
see list of warnings
5. See error
See first screenshot
**Expected behavior**
I have this 

MIN_CELL_VOLTAGE = 2.88
MAX CELL VOLTAGE = 3.64

In my Utilys.py so my max voltage are 29.12V
But the jk-bms script will bring up warnings about high voltage begins from 27.66V see screenshot.

**Screenshots**
[image]
[image]

[image]

**VenusOS (please complete the following information):**
 - Device type: Raspberry Pi
 - Firmware Version: 2.90-6

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: JK-BMS
 - Cells: 8
 - Interface: USB-RS485

**Additional context**
Have run the version 0.10

chickenbig, 2022-05-19:
[testpaul999](https://github.com/testpaul999) I think it would be helpful to see what the status on the JKBMS phone app is at the time of the warnings. Could it be that one of your cells is reaching 3.65V (or some other high value) far earlier than the others?

Also some additional logging in [Jkbms.to_protection_bits](https://github.com/Louisvdw/dbus-serialbattery/blob/f034d7c226d89a662f716da32335d61abd9d2ea5/etc/dbus-serialbattery/jkbms.py#L123) would help clarify what is going on; both the raw data and interpretation would give a better idea about what the BMS is complaining about and how the code interprets it.

On the subject of these protection bits, the JKBMS code was added under [this issue](https://github.com/Louisvdw/dbus-serialbattery/issues/29); the [RS485 protocol definition](https://github.com/Louisvdw/dbus-serialbattery/files/6916652/BMS.485.instruction.pdf) page 11 translation is a little difficult to understand. [DIY Solar Forum Resources](https://diysolarforum.com/resources/jk-bms-documentation-on-protocols-provided-by-hankzor.259/) links to [this document](https://github.com/NEEY-electronic/JK/blob/JK-BMS/BMS-RS485Communication%20protocol.pdf) which despite the Neey branding appears to be similar but perhaps slightly more understandable. Anyhow, I'm not sure about the Jkbms.to_protection_bits interpretation of these bits, at least from visual inspection.

testpaul999, 2022-05-19:
Hello,

Yes :-( this is my problem.

„Could it be that one of your cells is reaching 3.65V (or some other high value) far earlier than the others?“

If this will trigger the alarm I‘ll try initials my Akku. 

Regards
Viele Grüße

Jens-Uwe Franz

> Am 19.05.2022 um 10:04 schrieb chickenbig ***@***.***>:
> 
> Could it be that one of your cells is reaching 3.65V (or some other high value) far earlier than the others?

testpaul999, 2022-05-20:
THX… 👍🏻🙂
Really good explanation.
I’ve found my issue at the Busbar to the cell how have high voltage. 😩 my fault…
Have screws down an now all cells close together.
Edit my untils.py to high 3.4 and low 3.1 as your example.

Thx for help.
The issue can close.

Viele Grüße

Jens-Uwe Franz

Am 20.05.2022 um 10:03 schrieb Louis Van Der Walt ***@***.***>:

The default MAX_CELL_VOLTAGE and MIN_CELL_VOLTAGE that is set in the driver is to get your battery to full charge without any cell going into protection. This is used to calculate the max voltage for the battery that we ask the chargers to charge teh battery too.

3.65V will be the protection voltage of a cell of most BMS while 3.45V is considered full charge.
If you set your charger to aim for 3.65V it meant you are aiming to reach the protection voltage which it will reach. Even 3.64V is too close.
If you look at your cells you will see that it takes hours of charging to go from 3.2V to 3.4V, but from 3.45V to 3.6+V can be in just a second. This give no time for balancing to kick in, thus one cell will reach 3.65V while most of the rest will still be around 3.45V (assuming all your cells are closely balanced, much worse if they are not). This all gets multiplied by the cell count that you have.

An example might help:
If we aim for 3.45V per cell that will tell the charger to charge to 27.6V
If we aim for 3.64V per cell that will tell the charger to charge to 29.12V
Now say all our cells are almost full. That would mean they are close to 3.45V or 27.6V while we are requesting the battery to go up to 29.12V or another 1.52V.
The first cell that is a bit more full that the rest will jump from 3.45V to 3.65V and raise an alarm, but it is only an increase of 0.2V which mean while one cell should be protected we are asking the charger to increase the voltage another 1.32V cause it thinks we are not full yet.
If this continues the BMS will disconnect and your power will trip.

Now if you do want to aim for 3.65V per cell you will have to change the setting on your JKBMS so that the BMS protection only kick in at a higher voltage (e.g. 3.85V), but I WOULD NOT RECOMMEND THIS if you want your battery to last a long time.

closed 2022-05-20 by Louisvdw

Fraser86, 2023-09-19:
Hey there. 

I'm having a similar issue and despite reading through this I don't know how to fix the problem. 

I have my high-voltage alarm set on the Jkbms as 28.4. OVP is set to 3.55. 

but I am seeing it triggered at 27.7v 28v 28.04v. 

This happens almost daily. my mppts at set to 28v max. 

I am not sure about the following 

what is causing this?
do I have a problem? 
how can I stop this?
what setting do you prefer for OVP.

please help.

Louisvdw, 2023-09-19:
@Fraser86 what settings are you using for the max cell voltage in the driver set up? Is it the default 3.45V per cell or did you change this? Inside your GX if you go to the battery parameters what are the CVL setting?

Fraser86, 2023-09-19:
I'm using a raspberry pi. but I'm away from home right now. 
I the Jkbms app
I have it set up as 

ovp 3.55
ovpr 3.5

I will look now if I can see what you are asking with remote console

Fraser86, 2023-09-19:
where do I find CVL settings?

Louisvdw, 2023-09-19:
In Remote Console -> your SerialBattery -> Parameter

Fraser86, 2023-09-19:
no access currently Louis but I will try later. 
Could It be I need to update the driver for this?

Louisvdw, 2023-09-19:
What happens in most cases is that some of your cells are out of balance. The BMS will try and balance them but if you push to much amps then this all is very hard to do for the balancers and then some cells spike higher. This can be very fast and the BMS cannot react fast enough to avoid the high voltage alarm. 
You could also get this with wrong settings, so first we need to check the settings. Also check the VRM graphs for what the cell voltages is doing at the time you get the alarms. 
The troubleshoot section in the docs have more info on this.

mr-manuel, 2023-09-19:
@Fraser86 please open a new support request for this. There all needed informations are asked we need from to help you.

Fraser86, 2023-09-19:
@mr-mauel I am trying to fix remotely and making a support ticket requires that I upload a configuration file and log. I don't know how to do that remotely. 

@Louisvdw I looked at the parameters and they are set to 

cvl 27.6
ccl 25amp
dcl 60amp. 

this obviously does not match my previously stated values in jkbms app. 
I am also not able to adjust the values. when clicking them no adjustment is available. 
does this mean I have to adjust them in the jkbms to match?

sorry for posting here but I don't have another option. 
unless there is away to get config file remotely.

mr-manuel, 2023-09-19:
If you have installed the Setup Helper and/or GuiMods you can simply install [venus-os_ngrok](https://github.com/mr-manuel/venus-os_ngrok) over the remote console.