Louisvdw/dbus-serialbattery #125: 0.11beta1 reads wrong cell voltage with Daly-BMS

centro-max, 2022-05-22:
**Describe the bug**
0.11beta1 reads wrong cell voltage

The fill level, the total voltage and the current [A] are read out correctly, only the minimum and maximum cell voltage in the detail page is incorrect.

I tested it with 2 different Daly-BMS, sometimes the minimum, sometimes and/or the maximum of the cell voltage is wrong.

**VenusOS (please complete the following information):**
 - Raspberry Pi 4b
 - Firmware Version 2.85

**Battery/BMS (please complete the following information):**
 - BMS/Battery type: Daly Smart BMS 100A 16S LifePo4
 - Cells: 16
 - Interface: USB-RS485
[image]
[image]

YoBnEb, 2022-06-13:
Hi,
I have quite the same issue while using the 0.11 beta versions. Tried it also with the latest beta 4.
As you can see, the first 7 cell values are correct. Number 8 is the far too low and for cell 9 to 16 there are no values at all.
Running the Daly 16S-200A (version with fan) serialbattery is running on the MP-II 48/3000 GX.

[image]

Any idea what to check?

Thanks in advance!

By the way... Very nice project :-)

towatai, 2022-06-14:
Same Problem here on my Rasperry Pi 3 b+ with OS 2.73, Beta4 Driver and my 8S Battery. You can Chek the Values here:
https://vrm.victronenergy.com/installation/175068/share/037bcec3

centro-max, 2022-06-15:
Same problem with OS 2.87 and beta4. But first time, i can see both bms with OS > 2.73, that ist fine.

[image]
[image]
[image]

Louisvdw, 2022-06-15:
Please test if this daly.py solves the min/max cell voltage issue. I think it has to do with the reading of the cell values which is not correct. This then override the min_cell_voltage/max_cell_voltage that is read from the BMS. This this version I have disabled the cell values so that the min/max would be used again. 
I don't have a Daly BMS to test with, so someone will need to test this for me please.

Download this txt file and rename to daly.py and copy to the GX device overriding the existing one (to be replaced over 0.11beta4 build) on /data/etc/dbus-serialbattery/daly.py
Let me know if this works so that I can apply it to the next release.

[daly.txt](https://github.com/Louisvdw/dbus-serialbattery/files/8911190/daly.txt)

centro-max, 2022-06-15:
Yes it works now! Everything is fine, the min and max Voltage is correct with the new daly.txt file.
OS 2.97, 011beta4, raspberrypi4b, 2 Daly-BMS

YoBnEb, 2022-06-15:
I can also confirm that the min/max cell values seems to be correct again.
The single cell voltages are gone (as maybe expected...).

Louisvdw, 2022-06-16:
Thank you for the feedback.
I have made the new release v0.12 which includes the fix

closed 2022-06-16 by Louisvdw