Louisvdw/dbus-serialbattery #128: Installing this driver destroys a already installed one

stundenblume, 2022-06-09:
1.: Thank you for this very helpful driver!

**Describe the bug**
Installing this driver destroys a already installed one.

**To Reproduce**
Steps to reproduce the behavior:
1. working [dbus-mqtt-devices](https://github.com/freakent/dbus-mqtt-devices) driver
2. installing this driver
3. reboot
4. dbus-mqtt-devices driver doesn't work anymore

**Expected behavior**
Installing this driver should not influence other drivers.

**Solution**
Reinstalling dbus-mqtt-devices helped

**What could I find**
Your installation script replaced a already existing entry in `/data/rc.local`
Reentering it did not help. I had to reinstall dbus-mqtt-devices

Louisvdw, 2022-06-10:
This was a known issue for the autoinstaller that the VenusOS provides0. 
For serialbattery it will not be a major issue if the rc.local with it's commands don't exist anymore after installation, but then when you upgrade your VenusOS firmware next time you will have to install the driver again. 

The only way to get around this would be to not use the auto installer option, and then adjusting the ssh install script to not destroy any existing files.

stundenblume, 2022-06-10:
@ngardiner
the setup script of dbus-mqtt-devices [appends](https://github.com/freakent/dbus-mqtt-devices/blob/main/bin/setup.sh#L33) his line to `/data/rc.local`

For me everything is working new. But just to be aware that this can happen.

ngardiner, 2022-06-11:
> @ngardiner the setup script of dbus-mqtt-devices [appends](https://github.com/freakent/dbus-mqtt-devices/blob/main/bin/setup.sh#L33) his line to `/data/rc.local`
> 
> For me everything is working new. But just to be aware that this can happen.

That's going to work I'd nobody puts an exit 0 at the end of their script or runs any command which doesn't either fork or finish immediately, but we are at the point where people have to guess what other scripts are going to do.

kr0815, 2022-07-19:
i found this old topic, have exactly the same problems
In rc-local there was en entry that reinstalled dbus-serialbattery at each reboot - at the reinstall deleted my other entries there

I uncommented that entry - now i want to change max charge / discharge current in utils.py, but this has no effect?
utils.py says 20A/20A , Venus GUI says 5A / 60A , very strange
Do i really need a reinstall each time the unit reboots?

pos-ei-don, 2022-09-11:
Hm, sadly the installation deleted my rc.local without any backup. so i lost many settings.....

Louisvdw, 2022-12-29:
Fixed in #370

closed 2022-12-29 by Louisvdw