Louisvdw/dbus-serialbattery #130: v0.11beta3 BMS driver does not start anymore after adding other lines (rc.local)

e-i-k-e, 2022-06-11:
Hello,

just upgraded from beta 1 to beta 3.
The content of the rc.local was overwritten (already known).

But after I added my drivers, the driver for the BMS wouldn't start anymore.

Content of the file:

```
#!/bin/bash
ln -sfn /data/etc/dbus-serialbattery /opt/victronenergy/dbus-serialbattery
ln -sfn /data/etc/dbus-serialbattery/service /opt/victronenergy/service-templates/dbus-serialbattery
sleep 10 && /data/home/root/mqtt-Grid-Bridge0 > /dev/null 2>/dev/null &
/data/dbus-shelly-1pm-pvinverter/install.sh
/data/dbus-shelly-1pm-pvinverter2/install.sh
```

It helped though that I replaced -sfn with -s (whatever this does ;) )

Venus OS v. 2.9 on an RPI2.

Louisvdw, 2022-06-12:
Could it be that a reboot have solved your issue and not the change from -sfn to -s ?
The -fn set that it will force the creation and not add sub links. 
If your driver did not start it means the symlinks were not created.

e-i-k-e, 2022-06-12:
No.  I also did a reboot beforehand (to test).

e-i-k-e, 2022-06-12:
So I just tested it again.
It actually loaded the driver this time.

But this time he didn't immediately connect to the BMS. It said "not connected" and alternated between the text and the three empty fields.
After what felt like 4 minutes, he finally got connected!

Is this behavior a coincidence? At least I've never seen it before

[image]

[image]

Louisvdw, 2022-06-12:
@e-i-k-e your voltage for 16cells at 55.1V is around 3.45V. The Daly balancers are not the greatest, so that could mean you have some cells that are lower or higher. Look at your Alarms in VRM if you see any cell imbalance alarms. Also go into the Details view under the battery and check your min and max cell voltages. If the max is higher (or close to your max) then the BMS will disconnect (some will reboot) which could loose the connection. 
If this only happen when the battery is at 100% that might be the issue. See if lowering your max cell voltage in the utils.py solves it. If it does you might need an external balancer if you want to go back to 3.45V, but 3.40V will only loose 1%SOC so not that critical

closed 2022-06-16 by Louisvdw