Louisvdw/dbus-serialbattery #135: After install Ver. 0.12 bmv-712 over Ve.Direct stop work

danygi79, 2022-06-17:
after updating the driver to the latest version, the devices connected via VE.Direct have stopped working. Disabling the driver resumes normal operation.

Venus OS on Multiplus 2 5000/48/70 GX

Louisvdw, 2022-06-17:
@danygi79 can you provide the current log file for serial-starter when the driver is installed and VE.Direct fails. 
If you know what ports your ve.direct device work on and can provide the log file fot it and also the dbus-serialbattery log for the same port it would also help (I can give you more detail on which those are if you first send the seriar-starter log)
https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot#check-the-log-files-on-your-device

Louisvdw, 2022-09-18:
no feedback received

closed 2022-09-18 by Louisvdw