Louisvdw/dbus-serialbattery #138: Seplos BMS

gerdluthe, 2022-06-18:
Would be very nice to have the seplos BMS supported here.
The Seplos BMS got a CAN and two RS485 Ports, documentation is available at several places, can post if needed.
[
SEPLOS CANBUS Protocol V1.0 www.seplos.comhttps://solaranzeige.de › phpBB3 › download › file](https://solaranzeige.de/phpBB3/download/file.php?id=2373)
[
](https://us.v-cdn.net/6034073/uploads/OJ8IDWBXR179/seplos-bms-communication-protocol-v2-0.pdf)

gerdluthe, 2022-06-18:
https://us.v-cdn.net/6034073/uploads/OJ8IDWBXR179/seplos-bms-communication-protocol-v2-0.pdf

SamuelBrucksch, 2022-07-26:
Seplos supports connection to Victron out of the box via can bus, no need to support here.

pau1phi11ips, 2022-08-15:
@Louisvdw I think this can be closed as per SamuelBrucksch's comment above.

closed 2022-08-15 by Louisvdw

Rimco, 2023-01-31:
@Louisvdw I would like to re-open this item, looking at the results of Andy from the Off-Grid Garage: https://www.youtube.com/watch?v=EDM_acJyFxw

As Andy mentions, Victron takes the upper limit voltage of the BMS as charging target. As your FAQ mentions, MAX_CELL_VOLTAGE and MIN_CELL_VOLTAGE are in your case actually separately defined.

Planning to buy both Victron and Seplos, I would like to avoid that problem. Did not figure out yet if I can somehow adapt the VE.Can interpretation on the GX side, but otherwise your driver will be my alternative option (via RS485). When I actually get the HW, I may also be able to help you out on the implementation, but time is always a limiting factor.

wollew, 2023-03-29:
It also seems that the Seplos provides only limited information via can bus, no single cell voltages for example. They're available on RS485 though.

Lucifer06, 2023-09-05:
@Rimco Agreed with you, seems the only viable solution would be to have Seplos supported with dbus-serialbattery
I believe it would be based on the RS485 protocol as it exposes everything.
Any tentative yet?

cheers

wollew, 2023-09-05:
Check out the [documentation](https://louisvdw.github.io/dbus-serialbattery/general/supported-bms/), it is supported for quite a while now.

andreas-bulling, 2023-09-18:
Do I need a separate RS484 cable/USB connection or does it also work through the CAN bus?

mr-manuel, 2023-09-19:
You need a serial connection.

andreas-bulling, 2023-09-19:
the one with two wires connected only or with all? (sorry for using this issue to ask this but I don't know where else to ask)

mr-manuel, 2023-09-19:
Please open a discussion for this.