Louisvdw/dbus-serialbattery #141: 'LltJbd' object has no attribute 'min_battery_voltage'

chadodes, 2022-06-23:
Hello Team!

I have only seen this with 1 other user, and it was reported after a **wifi dongle** had been removed, but a resolution was never posted.
I have wifi built in, however I have been trying several different USB to serial adapters in the process of trying to get things to work. My latest is a dual serial startech with a prolific chipset.

I also own two Cerbo Gx's but one I damaged while trying to hook up tank sensors, so I am sensitive to that, and wanted to get this all working in my lab before I take it out to my motorhome at my farm 2 hours drive away.

The Venus unit is a RPI4 running **v2.82-large-30**
The unit also has a touch screen and is using HDMI 1
The BMS is a **JBD** in a 4s config with 4 x 100AH LIFEPO4 EVE Cells

**Log when running:**
 cat /data/log/dbus-serialbattery.ttyUSB0/current | tai64nlocal

_2022-06-23 08:24:48.445116500 INFO:SerialBattery:Starting dbus-serialbattery
2022-06-23 08:24:48.445610500 INFO:SerialBattery:dbus-serialbattery v0.12
2022-06-23 08:24:48.446815500 INFO:SerialBattery:Testing LltJbd
2022-06-23 08:24:48.844155500 INFO:SerialBattery:Connection established to LltJbd
2022-06-23 08:24:48.867069500 INFO:SerialBattery:DeviceInstance = 3
2022-06-23 08:24:48.867531500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
2022-06-23 08:24:49.256784500 Traceback (most recent call last):
2022-06-23 08:24:49.256789500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 115, in <module>
2022-06-23 08:24:49.256793500     main()
2022-06-23 08:24:49.256795500   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 100, in main
2022-06-23 08:24:49.256799500     if not helper.setup_vedbus():
2022-06-23 08:24:49.256801500   File "/opt/victronenergy/dbus-serialbattery/dbushelper.py", line 76, in setup_vedbus
2022-06-23 08:24:49.256805500     self._dbusservice.add_path('/Info/BatteryLowVoltage', self.battery.min_battery_voltage, writeable=True)
2022-06-23 08:24:49.256873500 AttributeError: 'LltJbd' object has no attribute 'min_battery_voltage'
2022-06-23 08:25:10.477830500 INFO:SerialBattery:Starting dbus-serialbattery
2022-06-23 08:25:10.478318500 INFO:SerialBattery:dbus-serialbattery v0.12
2022-06-23 08:25:10.479511500 INFO:SerialBattery:Testing LltJbd
2022-06-23 08:25:10.874135500 INFO:SerialBattery:Connection established to LltJbd
2022-06-23 08:25:10.895664500 INFO:SerialBattery:DeviceInstance = 3
2022-06-23 08:25:10.895975500 INFO:SerialBattery:com.victronenergy.battery.ttyUSB0
root@raspberrypi4:~#_

Settings inside /data/etc/dbus-serialbattery/utils.py:
_# Constants - Need to dynamically get them in future
DRIVER_VERSION = 0.12
DRIVER_SUBVERSION = ''
zero_char = chr(48)
degree_sign = u'\N{DEGREE SIGN}'
# Cell min/max voltages - used with the cell count to get the min/max battery voltage
MIN_CELL_VOLTAGE = 2.6
MAX_CELL_VOLTAGE = 3.55
# battery Current limits
MAX_BATTERY_CURRENT = 100.0
MAX_BATTERY_DISCHARGE_CURRENT = 100.0
# Charge current control management enable (True/False).
CCCM_ENABLE = True

# Daly settings
# Battery capacity (amps) if the BMS does not support reading it
BATTERY_CAPACITY = 100
# Invert Battery Current. Default non-inverted. Set to -1 to invert
INVERT_CURRENT_MEASUREMENT = 1_

My understanding is that this occurs when it can not accurately get a cell count due to voltage.
Could it be that my BMS is in a cell overvolt state and thus communicates differently? especially if in a mosfet disconnect state? 
I did run a slow and low amp (7 watts) charge on it before hand to help allow everything to balance out.

As I don't see this reported with many other users, I am sure it is something on my end, like GPS, or the touch screen fighting over serial ports.

I hope my logs and info help narrow things down.
Thanks!

IamSierraCharlie, 2022-09-11:
Daly BMS here - seeing the same issue on a raspberry PI 4

Louisvdw, 2022-09-12:
@IamSierraCharlie which version of the driver are you using when you get this?
Try the latest beta v0.12b3 and also post the battery settings from the log file when it starts

IamSierraCharlie, 2022-09-12:
@Louisvdw thanks for your message.
I'll try as you suggest; Although I think my issue is either the USB to Serial converter or the fact that I am trying to connect to the Daly BMS via RS485.  I have installed the USB to serial driver in Windows and run the Daly software - still no dice.  I have a USB to UART adaptor on the way.  I also installed Victron S/W on a raspberry Pi 2 today - same fault but I dont think its software.  If I find some compelling info that suggests otherwise, I'll report back.  Thanks again for responding!

IamSierraCharlie, 2022-09-14:
Just to close out my comment above - new USB to RS485 adapter turned up today.  Swapping this with the one I have fixed the issue I was having.  Id suggest the old one "kind of worked", but the new one actually works properly. 

[image]