Louisvdw/dbus-serialbattery #150: Phantom BMS generates alarm notification

centro-max, 2022-07-14:
**Describe the bug**
I have only connected 2 Daly BMS but unfortunately a non-existent ANT BMS is also regularly recognized which then generates various alarm messages.

**To Reproduce**
Steps to reproduce the behavior:
I can delete the non existent ANT BMS but after a few days then the Phantom BMS will appear.

**Screenshots**
[image]
[image]

**VenusOS (please complete the following information):**
Venus OS 2.87 with Raspberry Pi 4 and serial battery v0.12

**Battery/BMS (please complete the following information):**
2 *  Daly BMS 48V with 16 cells and 120A/100A

Louisvdw, 2022-07-16:
Hi @centro-max 
There should be a log file that get generated for that ANT bms. Could you upload that so that I can see what it is doing. 
https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot#check-the-log-files-on-your-device

centro-max, 2022-07-16:
Hi,ok i will do this next week because i am on holiday.

https://github.com/Louisvdw/dbus-serialbattery/wiki/Troubleshoot#check-the-log-files-on-your-device

centro-max, 2022-07-26:
Now the Sinowealth BMS was recognised, here are the log files i found.

Could it be a solution to disable the search for alle types of BMS? I know i have only Daly, so how could i deactivate the search for the other BMS-types?

[current.txt](https://github.com/Louisvdw/dbus-serialbattery/files/9187746/current.txt)
[current-2.txt](https://github.com/Louisvdw/dbus-serialbattery/files/9187753/current-2.txt)
[current-0.txt](https://github.com/Louisvdw/dbus-serialbattery/files/9187777/current-0.txt)

Johanson2020, 2022-10-16:
HI @centro-max 
Hello @Louisvdw 

I have exactly the same problem
how did you solve it?

For me the problem started when I installed a BlueSolar Charger MPPT 150/35 and added a diy Ve.Bus serial connection 
I use only one JKBMS and would also like to simply stop the search for further BMSs?

I have just installed the update to 0.13.
but I assume this update does not fix the problem?

Thanks a lot
Johannes

Louisvdw, 2022-10-16:
I plan to add an option that you can select which BMS must be seached for. That should also help with this issue