Louisvdw/dbus-serialbattery #153: Daly 4s smart bms - no battery connection at /dev/ttyUSB0

chavalaloco, 2022-07-22:
I can´t get the driver to work, the bms is not connecting.

My Setup:
Raspberry PI 3 
Venus OS 2.89
Daly 4s Smart BMS 150a
[image]
Daly uart usb converter cable
[image]

here is the log "/data/log/dbus-serialbattery.ttyUSB0/current" it can´t open any connection and then it´s crashing
```
000000062db0cf1242aa4e4 INFO:SerialBattery:Testing LltJbd
@4000000062db0cf13423d8d4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf1343837ac INFO:SerialBattery:Testing Ant
@4000000062db0cf2089a5abc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf208aa8b44 ERROR:SerialBattery:>>> ERROR: Incorrect Data
@4000000062db0cf208b312dc INFO:SerialBattery:Testing Daly
@4000000062db0cf218bbe2e4 ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf218d70c04 INFO:SerialBattery:Testing Daly
@4000000062db0cf228baeeec ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf228cb6d94 INFO:SerialBattery:Testing Jkbms
@4000000062db0cf238c875fc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf238d9564c INFO:SerialBattery:Testing Sinowealth
@4000000062db0cf30d3dd2ec ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf30d508be4 INFO:SerialBattery:Testing Renogy
@4000000062db0cf31d42bedc ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf31d53222c INFO:SerialBattery:Testing Revov
@4000000062db0cf32d38782c ERROR:SerialBattery:>>> ERROR: No reply - returning
@4000000062db0cf32d4fad94 ERROR:SerialBattery:read_serial_data_revov::Serial Data is Bad
@4000000062db0cf40f999184 ERROR:SerialBattery:ERROR >>> No battery connection at /dev/ttyUSB0
@4000000062db0cf40fbe4bdc Traceback (most recent call last):
@4000000062db0cf40fbe634c   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 115, in <module>
@4000000062db0cf40fbe76d4     main()
@4000000062db0cf40fbe7ea4   File "/opt/victronenergy/dbus-serialbattery/dbus-serialbattery.py", line 90, in main
@4000000062db0cf40fbe922c     os.exit(1)
@4000000062db0cf40fbe99fc AttributeError: module 'os' has no attribute 'exit'

```
Here is the "/data/log/serial-starter/current" log
```
@4000000062db0e6729947e3c INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000062db0e723b5e631c INFO: Start service gps-dbus.ttyUSB0 once
@4000000062db0e7c1237b434 INFO: Start service vedirect-interface.ttyUSB0 once
@4000000062db0e8120028abc INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000062db0e8c3208e044 INFO: Start service gps-dbus.ttyUSB0 once
@4000000062db0e96068c5e64 INFO: Start service vedirect-interface.ttyUSB0 once
@4000000062db0e9b13f0c644 INFO: Start service dbus-serialbattery.ttyUSB0 once
@4000000062db0ea4244ea3bc INFO: Start service gps-dbus.ttyUSB0 once
@4000000062db0ead395f10ac INFO: Start service vedirect-interface.ttyUSB0 once
@4000000062db0eb31ec4da9c INFO: Start service dbus-serialbattery.ttyUSB0 once
```

Does anyone know what can cause the error, or anyone has the same hardware? especially the cable and bms...

dominikfe, 2022-07-24:
I am having the very same issue suddenly. Everything was working well with my JK BMS but now I am getting the same errors in the log.
Can this be related to some changes in Venus OS? I did not apply any updates afaik but am not sure if Venus OS did something automatically...

SamuelBrucksch, 2022-07-28:
@chavalaloco i had to press the BT button before i could communicate with the dongle. The BMS goes into standby on UART after a while and on start. So when the BT button is clicked, switch over to your UART cable that goes to Venus OS. 

Not sure why it does not report data alway...

closed 2023-04-20 by mr-manuel